VoodooI2C/VoodooI2C #369: Pinch with 3 fingers and thumb ELAN1200 problem

okram24, 2020-08-13:
**Describe the bug**
All gestures work fine, once I do the pinch with 3 fingers and thumb, the trackpad stops registering click and drag correctly, and each click is registered as multiple if held. A tap works fine. Sleep sometimes fixes the issue, not always, requiring a reboot or even more than 1 reboot. Sometimes the trackpad stops responding for about 2 minutes before re-connecting, no sleep required, but a sleep is preferable because it's faster to get it working again.

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
Yes

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
Yes

**System Environment**
Please answer:
1. What is your operating system? macOS Catalina and Big Sur
2. What version of the drivers (including drivers) Latest release, also specified in the Report.txt
3. What is your hardware configuration? ELAN1200 on an i7-8750H (ASUS ZenBook Pro 15 UX550GE)

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive. 
[VoodooI2C problem report 3.zip](https://github.com/VoodooI2C/VoodooI2C/files/5070110/VoodooI2C.problem.report.3.zip)

**Additional context**
My touchscreen that is also pinned, supports this gesture just fine. It doesn't stop responding or does this odd behavior.

okram24, 2020-08-13:
Experienced yet another issue, a ghost finger appears on the bottom right of the trackpad when I do multiple finger gestures... Not sure if it's a problem with my trackpad or with the kext.
[Ghost finger issue.mp4.zip](https://github.com/VoodooI2C/VoodooI2C/files/5071041/Ghost.finger.issue.mp4.zip)

ben9923, 2020-12-22:
Please try latest VoodooI2C release.

kprinssu, 2022-02-27:
Please test with 2.7, if there is no response within 3 days, I will be closing this issue.

kprinssu, 2022-03-05:
Please re-open if this is still an issue with 2.7.

closed 2022-03-05 by kprinssu