VoodooI2C/VoodooI2C #371: Touchpad stop working after wake from sleep.

the-eric-kwok, 2020-08-14:
**Describe the bug**
Touchpad stop working after wake from sleep. This issue occurs **both in latest code build and latest release [(v2.4.4)](https://github.com/VoodooI2C/VoodooI2C/releases/tag/2.4.4)**. 

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**
Please answer:
1. What is your operating system?  **macOS Catalina 10.15**
2. What version of the drivers (including drivers) are you using?  **VoodooI2CSynaptics**
3. What is your hardware configuration?
  Xiaomi Noteboot Air 12.5
  Intel m3-6Y30 Skylake
  Sunrise Point-LP Serial IO I2C Controller (pci8086,9d60)

**Troubleshooting Archive**
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5075246/Archive.zip)

Goshin, 2020-08-14:
Have you tried v2.4.3?

the-eric-kwok, 2020-08-14:
@Goshin confirm that reset to VoodooI2C/VoodooI2CSynaptics@e8923aa3 fix this.

Goshin, 2020-08-14:
So as @kprinssu said it is https://github.com/VoodooI2C/VoodooI2CSynaptics/commit/804db5c7da71365807555f94e77258f83e57d616 that causes the problem?

the-eric-kwok, 2020-08-14:
Yes, I think so.
And I didn't face that "getting stuck in a click/drag/highlight mode after waking from sleep" issue even if I revert it by the way.

the-eric-kwok, 2020-08-14:
[log.txt](https://github.com/VoodooI2C/VoodooI2C/files/5075880/log.txt)
Here is the new log.

TheRandMan, 2020-08-15:
Prior to removing `rmi_populate();` from the wake sequence,  I would only get errors in the following from the sequence when waking from sleep:
`rmi_populate_f01();`
`rmi_populate_f11();`
`rmi_populate_f30();`

If we drop those and only perform the the remaining actions from `rmi_populate();` during wake, might that resolve my click stick issue and @the-eric-kwok\'s wake issue?

`rmi_set_mode(RMI_MODE_ATTN_REPORTS);`
and
`rmi_scan_pdt();`

```
if (!awake){
            rmi_set_mode(RMI_MODE_ATTN_REPORTS);
            rmi_scan_pdt();
            
            awake = true;
            IOLog("%s::Woke up from Sleep!\n", getName());
        }

```

I've tested on my end and am not experiencing the click stick issue I had in previous versions (although it was intermittent
so I can’t say with certainty). 

@the-eric-kwok can you test on your end?

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5079285/Archive.zip)

the-eric-kwok, 2020-08-16:
@TheRandMan can confirm that on VoodooI2C/VoodooI2C@3527ec36 with your VoodooI2CSynaptics.kext I still have that "not functioning after sleep" issue.

the-eric-kwok, 2020-08-16:
By the way I had click stick issue on VoodooI2C/VoodooI2CSynaptics@e8923aa once yesterday.

1Revenger1, 2020-08-16:
With VoodooRMI, I needed to [scan the PDT table again](https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/RMIBus.cpp#L198) in order to get it to get the sensor to wake up after sleeping. I do further go in F01.cpp to put the sensor to [sleep/take it out of sleep](https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/Functions/F01.cpp#L460). I doubt putting the sensor to sleep/taking it out of sleep is *needed* though.

TheRandMan, 2020-08-16:
@1Revenger1 I actually saw that mentioned in VoodooRMI! I'm not a developer by any means so nice too see I read that right. 

@the-eric-kwok I noticed in your logs that your touchpad is affected by an F01 firmware bug that can cause the touchpad to stop responding.  Looks like there is a fix implemented in `rmi_populate_f01();`, so let's add that back to the wake sequence and see what happens.

Can you try this one? I've tested sleep/wake several times on my end and still no click stick issue.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5081148/Archive.zip)

If this doesn't do the trick, we'll have to leave it to somebody who actually knows what they're doing 😋

the-eric-kwok, 2020-08-17:
> @1Revenger1 I actually saw that mentioned in VoodooRMI! I'm not a developer by any means so nice too see I read that right.
>
> @the-eric-kwok I noticed in your logs that your touchpad is affected by [...]

@TheRandMan  It works!
Here are new logs. I've slept twice and it waked up both successfully! Need more time to test for the click stick issue.
[log.log](https://github.com/VoodooI2C/VoodooI2C/files/5081744/log.log)

the-eric-kwok, 2020-08-17:
Click stick issue occurs! Check the video and logs around 2020-08-17 12:16 in this archive. But I didn't see much different between normal wake up and this issue wake up. 
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5082179/Archive.zip)

wdubaiyu, 2020-08-17:
My xps 9380 has the same problem .
[device]([url]( #195 
[归档.zip](https://github.com/VoodooI2C/VoodooI2C/files/5084869/default.zip)
))  ELAN2930 and DELL08AF）

the-eric-kwok, 2020-10-10:
This issue occurs in release [v2.5.2](https://github.com/VoodooI2C/VoodooI2C/releases/tag/2.5.2)

profzei, 2020-10-13:
Same here with SYNA1D31 device using GPIO pinning mode and latest release VoodooI2C & VoodooI2CHID 2.5.2

kprinssu, 2020-10-13:
@TheRandMan can you create a PR in the upstream repo to fix the issue, please?

TheRandMan, 2020-10-15:
@kprinssu I'm not sure I have a good fix. @the-eric-kwok reports that my fix brings back the click-stick issue from here: https://github.com/VoodooI2C/VoodooI2CSynaptics/pull/8

I don't see click-stick on my end though.

@the-eric-kwok can you confirm you experience click-stick with the test from here: https://github.com/VoodooI2C/VoodooI2C/issues/371#issuecomment-674556895

@profzei's issue is likely different as the original issue from this thread seems specific to VoodooI2CSynaptics. If a fix could be implemented to the base VoodooI2C that resolves both issues that would probably be ideal. Unfortunately, that's well beyond my ability.

the-eric-kwok, 2020-10-21:
Sorry for the late reply, yes I can confirm that I experience click-stick with https://github.com/VoodooI2C/VoodooI2C/issues/371#issuecomment-674556895

When I occurs the click-stick issue, I press down my trackpad. Then the trackpad seems to be reset, and click-stick issue disappear, every thing resume to be normal.

giugrilli, 2020-12-14:
news on this?

ben9923, 2021-02-27:
@the-eric-kwok VoodooI2C + VoodooRMI work just fine, right?

I think we should just deprecate VoodooI2CSynaptics now and refer to VoodooRMI, that's actively maintained.
CC @kprinssu @1Revenger1

kprinssu, 2021-03-02:
@ben9923 I am in favour of deprecating the Synaptics satellite.

1Revenger1, 2021-04-06:
Sorry to hijack this issue a bit - I have a PR which significantly changes how sleep/wake works. I'd like to make sure that anyone using VoodooRMI doesn't have any issues with these changes: https://github.com/VoodooSMBus/VoodooRMI/pull/110

If you want build artifacts, click the green checkmark on the latest commit and go to the build action. Artifact should be on the right, with the 3 horizontal lines menu.

FreeJHack, 2021-04-11:
> if this bug doesn't come back with the version above

@1Revenger1 Confirmed here #371
Trackpad stop working after wake from sleep with version I2C 1.3.3. No issues with 1.3.2
Trackpad SYNA1202 using F12
VoodooI2C 2.6.5
VoodooPS2 2.2.2
Mojave 10.14.6 with OC 0.6.8
 
FYI
In addition, reset error during boot (as described in driver cpp file):
(VoodooRMI) VRMI - Error: Unable to reset
(VoodooRMI) VRMI - Error: RMI initial reset failed! Continuing in spite of this

closed 2021-09-21 by the-eric-kwok