VoodooI2C/VoodooI2C #372: VoodooI2C not loading if injected from OC

HQuest, 2020-08-17:
I understand this is not a support forum, but the documentation doesn't talk at all about this issue and I am looking for hints on this troubleshooting exercise.

This system is a Dell Latitude 7480 (Skylake i5-6300U), running Catalina 10.15.6.

I recently migrated this laptop from Clover to OpenCore. OC injects the two plugin kexts VoodooGPIO and VoodooI2CServices, but never loads its parent VoodooI2C nor the satellite VoodooI2CHID or the multitouch VoodooInput (configured to load on this order at OC's config.plist). By removing the injection and adding both VoodooI2C and VoodooI2CHID kext files into /L/E, all kexts loads on their own, with both touchscreen and trackpad working right away as expected.

Manually loading these extensions in the live OS from a folder other than /L/E can only be done with disabled CSR. Each extension load returns signature errors and personality warnings, however the extension loading is allowed and completed. Yet, they do not take control of the peripherals - I suspect this is from related Apple subsystems already been initialized.

I'm using the VoodooI2C official GitHub provided release files, combined with official OC release files or sources compiled as either Debug or Release, for the past week while doing a lot of trial and error attempts.

Since adding kexts into Big Sur is a bit more complicated than Catalina is, I'm looking at understanding why this fails before I make the jump into the beta OS.

Also since any other kexts can be loaded on OC, I suspect this is not on the OC end; I'm still however adding @vit9696 from the Acidanthera's team for awareness.

Happy for further guidance, and happy to provide further logging to assist on this exercise.

vit9696, 2020-08-17:
VoodooI2C team has a weird dependency order to my memory. So you may have added the entries to config.plist in the wrong order. Also, please check the HFS issue, as our team is pending changes due to that.

HQuest, 2020-08-17:
Thanks @vit9696 . The load order was one of the items I reviewed multiple times against other guides and documents, and confirmed later with the extensions on /L/E. They are, as mentioned, added in the correct order into config.plist and matching the Index order as seen from kextstat. Unless the injection order should not follow what I see during a working kextstat output.

hieplpvip, 2020-08-17:
@HQuest You should provide your OpenCore config.plist
The order that works for me is: `VoodooGPIO` -> `VoodooI2CServices` -> `VoodooInput` -> `VoodooI2C` -> `VoodooI2CHID`

kprinssu, 2020-08-17:
Please read and follow the Github issue template. You have not attached a troubleshooting archive and this seems to be a configuration issue, please use Gitter.

closed 2020-08-17 by kprinssu