VoodooI2C/VoodooI2C #375: Kaby Lake | ELAN1200 | ASUS (Polling or Interrupts not working on OpenCore)

Denisuu, 2020-09-05:
**Describe the bug**
Since I've migrated to OpenCore the trackpad isn't working anymore, it did work in polling mode on Clover 
(With the usual random freeze until sleep)

- When I don't add DSDT.aml to the OpenCore ACPI folder, the trackpad works but it doesn't show in settings. 
- Even if I ad the DSDT.aml with only AsusSMC patches applied, the trackpad doesn't do anything. 
- When I add the patches for Interrupts, the trackpad shows in settings but it's completely unresponsive.

I've read the Dortania guide multiple times. I tried the SSDT-GPI0.aml method, I tried the SSDT-XOSI.aml method.
Mald0n from Olarila have a look at my EFI and patched my DSDT, we just can't seem to get the trackpad to work.

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes, I did find a lot of information on GPIO pinning bug on Asus, mainly by @hieplpvip It seems he had a VoodooI2C repo for Asus but that link is broken. I don't find any final answer if anyone ever found a work-around for problematic Kaby Lake Asus machines.

https://github.com/hieplpvip/ASUS-ZENBOOK-HACKINTOSH/wiki/Notes-about-VoodooI2C
https://gitter.im/alexandred/VoodooI2C/archives/2018/07/20

**System Environment**
1. OS: MacOS 10.15.6 Catalina
2. Driver: VoodooI2C, VoodooI2CHID (2.4.4)
3. ASUS VivoBook 15 N580VD
CPU: i7-7700HQ (Kaby Lake FCBGA-1440)
GPU: Intel HD630 / GTX1050 (2GB)
RAM: 32GB
SSD: SanDisk x400 M.2 512GB 
SSHD: 2TB FireCuda (ST2000LX001)
Ethernet: Realtek RTL8168
WIFI: DW1820A
Audio: Conexant Audio CX8150
Built-in Display: B156HAN06.1 (1080p non touch)
Touchpad: ELAN1200

**Troubleshooting Archive**
**N580VD - Interrupts:** Contains DSDT with interrupts patches and IOreg > Trackpad not working.
**N580VD - Polling:** Contains DSDT in polling mode and IOreg > Trackpad not working.
**N580VD - NoDSDT** Contains IOreg when booted without DSDT > Trackpad working but not shown in settings. 
**N580VD - Original:** Original unpatched DSDT
**N580VD - OC0.6.0 FULL EFI:** Full OpenCore EFI
**N580VD - Clover_r5121 Debug:** EFI where VoodooI2C was working

[N580VD - Interrupts.zip](https://github.com/VoodooI2C/VoodooI2C/files/5178096/N580VD.-.Interrupts.zip)
[N580VD - Polling.zip](https://github.com/VoodooI2C/VoodooI2C/files/5178100/N580VD.-.Polling.zip)
[N580VD - NoDSDT.zip](https://github.com/VoodooI2C/VoodooI2C/files/5178097/N580VD.-.NoDSDT.zip)
[N580VD - Original.zip](https://github.com/VoodooI2C/VoodooI2C/files/5178099/N580VD.-.Original.zip)
[N580VD-OC0.6.0_FULL-EFI.zip](https://github.com/VoodooI2C/VoodooI2C/files/5178108/N580VD-OC0.6.0_FULL-EFI.zip)
[N580VD Clover_r5121 Debug.zip](https://github.com/VoodooI2C/VoodooI2C/files/5178107/N580VD.Clover_r5121.Debug.zip)

**Additional context**
Not sure if this works? I'd have have to get the trackpad working first though...

Potential fix for random freeze problem:
https://github.com/hieplpvip/ASUS-ZENBOOK-HACKINTOSH/blob/master/src/hotpatch/SSDT-I2CBUS.dsl

ben9923, 2020-09-05:
Statically patching ACPI is discouraged, especially by OpenCore devs. You're probably patching a DSDT that's out-of-sync with your native DSDT.
Do not use a patched DSDT. Instead combine SSDTs and config.plist patches. Rename the original `_CRS` (Not all `_CRS` methods!) to `XCRS` and insert a patched one via an SSDT.
[This guide](https://www.tonymacx86.com/threads/guide-patching-laptop-dsdt-ssdts.152573/) might be helpful. While it describes patching with Clover, the patching concepts are the same (i.e. using a few more bytes to only match the wanted code).

To see your trackpad in SysPrefs you must have functional battery status reporting (SMCBatteryManager.kext + proper ACPI patches).

As it seems like a user configuration error and not a VoodooI2C bug, I'll close this issue.
For further assistance, use the VoodooI2C Gitter chat. Thanks for the detailed report!

closed 2020-09-05 by ben9923

Denisuu, 2020-09-07:
Adding SMCBatteryManager.kext was indeed the solution to get the trackpad appear in settings! Thanks!