VoodooI2C/VoodooI2C #378: Worked on Catalina but not in Big Sur beta

seungjunProgramming, 2020-09-11:
**Describe the bug**
I use Liteon's touchpad and it worked on Catalina Using Clover.
But I installed VoodooI2C + VoodooI2CHID on Big Sur using Opencore, it doesn't work

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**
Please answer:
1. What is your operating system? Big Sur beta6
2. What version of the drivers (including drivers) are you using? 2.4.4
3. What is your hardware configuration?Liteon touchpad (TPD0) -> the maker of the laptop refuses to tell me the model

**Troubleshooting Archive**
[image]

**Additional context**
Add any other context about the problem here.

ben9923, 2020-09-11:
A screenshot of IOReg is not helpful. The issue template suggests what you should attach for debugging (ACPI, an actual copy of IOReg, kernel logs).

There's no need to block Apple's I2C kexts.

kprinssu, 2020-09-11:
As @ben9923 please fill out the issue template and attach a troubleshooting archive. Please re-open the Github issue once you have done so.

closed 2020-09-11 by kprinssu