VoodooI2C/VoodooI2C #379: I2C trapped not working on Big Sur with opencore(Worked on Catalina with Clover)

seungjunProgramming, 2020-09-12:
Describe the bug
I use Liteon's touchpad and it worked on Catalina Using Clover.
But I installed VoodooI2C + VoodooI2CHID on Big Sur using Opencore, it doesn't work

Did you read the common errors documentation?
yes

Did you read the troubleshooting documentation?
yes

Have you searched the issue on Github, Gitter, or Google?
yes

System Environment
Please answer:

What is your operating system? Big Sur beta6
What version of the drivers (including drivers) are you using? 2.4.4
What is your hardware configuration?Liteon touchpad (TPD0) -> the maker of the laptop refuses to tell me the model

**Troubleshooting Archive**
[BootLog.txt](https://github.com/VoodooI2C/VoodooI2C/files/5212623/BootLog.txt)
[MacBook Pro.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/5212624/MacBook.Pro.ioreg.zip)

**Additional context**
Add any other context about the problem here.

andreszerocross, 2020-09-13:
Clearly missing configuration between your Clover config vs OpenCore config.

Can you use your current OpenCore Config and boot to Catalina system?? Does your touchpad work??

kprinssu, 2020-09-13:
Big Sur is currently not supported, but multiple users have gotten VoodooI2C work. 

This sounds more of an environment/config issue. Please use Gitter for support.

closed 2020-09-13 by kprinssu