VoodooI2C/VoodooI2C #382: SYNA3602

tmbt78, 2020-09-18:
**Describe the bug**
SYNA3602 touchpad doesn't work. 
Kexts are loaded correctly (VoodooI2C and HID) and VoodooGPIOSunrisePointLP is present in Ioreg under GPI0

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes. Read a lot about SYNA3602 on Gitter chat but without  finding a solution

**System Environment**
Please answer:
1. What is your operating system? 
Catalina
2. What version of the drivers (including drivers) are you using?
The latest 2.4.4
3. What is your hardware configuration?
Chuwi Aerobook PLUS
Intel i5-6287U
8GB RAM
SSD 250GB

**Troubleshooting Archive**
I've attached an archive with my ioREG, Kernel Log, DSDT untouched and OC folder without the Resource folder so you can check also my hotpached SSDTs.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5246766/Archive.zip)

**Additional context**
Battery indicator is working (i've patched it according to RehabMan guide) but there is a delay in notify when connecting or disconnecting the power plug. The icon needs around 30 seconds to change.

kprinssu, 2020-09-18:
This is most likely a configuration issue. Please wait for support on Gitter.

Please only re-open if someone (such as Ben or myself) has directed you to do so on Gitter.

closed 2020-09-18 by kprinssu

tmbt78, 2020-09-23:
@kprinssu I've waited as instructed after reporting on Gitter without anyone answering my questions. I'm not asking for a solution but in these days I've checked everywhere and haven't found a single case of SYNA3602 Hackintosh with trackpad working. I've patched my DSDT in any way possible without luck. I've read all the Gitter room from July 2019 when the first SYNA3602 touchpad came out ... all the people reporting problems with this touchpad didn't solve their problem. My DSDT is identical 99% to other with my problem. I'm quite sure it's an ACPI poor implementation on my side but I need some advice to where to look. Right now my ioREG is like this everything is up and running as it should ...
Please don't tell me is just a configuration issue ...
[image]

CrisHotpatch, 2023-06-07:
This is the solution:

https://www.tonymacx86.com/threads/voodooi2c-issues-syna3602.304029/post-2177566

fmgdias, 2024-01-26:
SYNA3602 Trackpad - Solution

Show contents pkg on VoodooI2CHID.kext

Open Info.plist with PlistEditor Pro

In IOKitPersonalities: Rename VoodooI2CHIDSYNA3602Device to VoodooI2CHIDSYNA3602&Col2Device

IOClass: Rename VoodooI2CHIDSYNA3602Device to VoodooI2CHIDSYNA3602&Col2Device

IOPropertyMatch: Rename SYNA3602 To SYNA3602&Col2

If "&Col2" not work, try "&Col1"

You must turn off your notebook after setup, it can't be rebooted, because this trackpad is intermittent

Lorys89, 2024-01-31:
> SYNA3602 Trackpad - Solution
>
> Show contents pkg on VoodooI2CHID.kext
>
> Open Info.plist with PlistEditor Pro
>
> In IOKitPersonalities: Rename VoodooI2CHIDSYNA3602Device to [...]

i have same problem and this fix not work

kprinssu, 2024-01-31:
Please use Gitter for extended support. This issue has been resolved for months.