VoodooI2C/VoodooI2C #383: Switch to Acidanthera MacKernelSDK

vit9696, 2020-09-21:
Hello, we made a new product designed to help all the kext developers to easily target different macOS versions — [MacKernelSDK](https://github.com/acidanthera/MacKernelSDK). This is a modified Kernel.framework SDK, which contains private headers, removed deprecations, availability checks in many places, and several other fixes.

In Acidanthera we migrated all our kexts to use MacKernelSDK including VoodooInput. Perhaps it will be useful for VoodooI2C as well. Feel free to adopt or even contribute.

kprinssu, 2020-09-22:
I am looking at the changes in VoodooInput (https://github.com/acidanthera/VoodooInput/commit/c3f635221ffbeda788b28568dce733d91d0a8a10) and it looks like a drop-in replacement for Apple provided SDK,

If so, this should be a realtively easy change on my end, however I do not exactly have a I2C device that I can test with.

kprinssu, 2020-12-20:
Implemented with https://github.com/VoodooI2C/VoodooI2C/pull/420.

closed 2020-12-20 by kprinssu