VoodooI2C/VoodooI2C #384: SYNA3602 & Timeout waiting ..

tmbt78, 2020-09-28:
**Describe the bug**
Trackpad is not always working after boot

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**
Please answer:
1. Catalina 
2. 2.4.4
3. Intel i5 6th gen 8GB RAM SYNA3602 as touchpad

After the closing of my previous bug report I've done some more research and finally got SYNA3602 working modding VoodooI2CHID info.plist changing IOClass to VoodooI2CHIDSYNA3602&Col02Device and name to SYNA3602&Col02

The problem is that the trackpad works only once or twice every 10 reboot or so. 
When it works everything works (pref panel, gesture, Interrupt mode).
When it doesn't work this is the error : 
VoodooI2CControllerDriver::pci8086,9d61 Timeout waiting for bus to accept transfer request
VoodooI2CControllerDriver::pci8086,9d61 Timed out waiting for bus to change state
VoodooI2CHIDDevice::SYNA3602 Could not get report descriptor

I will attach kernel log and ioReg for both scenarios.
These are the facts.

What I've noticed but this is more empirical is that if during boot I keep at least one finger on the touchpad usually the trackpad works.. don't know why.

 [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5292756/Archive.zip)

mishurov, 2020-10-03:
I had the same problem with ELAN1200, I manually applied the patch from this https://github.com/VoodooI2C/VoodooI2C/issues/319#issuecomment-646437623 which increases timeouts, compiled the extension and now it works on every boot.

Kmbeatz, 2020-10-03:
Can you Upload please your Compiled Kext? Thank you so much

mishurov, 2020-10-03:
@Kmbeatz here you go [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5321408/Archive.zip)

Kmbeatz, 2020-10-03:
Thank you soo much Alex 👍

kprinssu, 2020-10-03:
@Kmbeatz please only comment on issues that contribute to help solve the problem.

mishurov, 2020-10-03:
@kprinssu it probably might make sense to include the extended timeouts into the main code, in my empirical experience even 10x increase in `VoodooI2CControllerDriver` didn't contribute a lot into overall boot time, I haven't noticed any difference. Yet it can make your life easier by preventing these issues from popping up from time to time.

kprinssu, 2020-10-03:
Thanks @mishurov, the timeout increase has been pushed to master and within 2.5.

@tmbt78 has gotten it working with the fix above.

closed 2020-10-03 by kprinssu