VoodooI2C/VoodooI2C #392: XPS 9570 Touchscreen doesn't work with 2.5.2 but work with 2.4.4

M69W, 2020-10-09:
XPS 9570: WCOM488F + SYNA2393
OC: [xxxzc/xps15-9570-macos](https://github.com/xxxzc/xps15-9570-macos)
It's seems to like same issues @/issues/282
[Voodoo.log](https://github.com/VoodooI2C/VoodooI2C/files/5356284/Voodoo.log)

kprinssu, 2020-10-09:
Please fill out the Github issue template.

ben9923, 2020-10-09:
After editing the issue according to the template, you'll also need to attach an ACPI dump for further debugging.
You can automate the whole troubleshooting archive generation using something like [RunMe.app](https://www.olarila.com/files/Utils/RunMe.app.zip).
Preferably run on both 2.4.4 and 2.5.2.

zhen-zen, 2020-10-09:
Your patching method involves certain hacks according to `https://github.com/xxxzc/xps15-9570-macos/blob/master/ACPI/SSDT-TPDX.dsl`. There's nothing could be inferred from files you uploaded and the repo as well.

M69W, 2020-10-10:
@kprinssu @ben9923 @zhen-zen 
Updated files both i2c_2.4.4 and i2c_2.5.2.

zhen-zen, 2020-10-10:
Can you try remove the `TPDM = Zero` or the whole TPDX ssdt with 2.5.2?
`SHPO (GPDI, One)` shouldn't be executed since that's intended for unsupported OS, i.e. `OSYS < 0x07DC`. I'm not sure if it will interfere with following initialization. Because `pin 36 cannot be used as IRQ` is from https://github.com/coolstar/VoodooGPIO/blob/master/VoodooGPIO/VoodooGPIO.cpp#L342. The pin might be configured in ACPI mode wrongly.

M69W, 2020-10-10:
> Can you try remove the `TPDM = Zero` or the whole TPDX ssdt with 2.5.2? Also please enable the `_OSI` to `XOSI` patch.
> 
> `SHPO (GPDI, One)` shouldn't be executed since that's intended for unsupported OS, i.e. `OSYS < 0x07DC`. I'm not sure if it will interfere with following initialization. Because `pin 36 cannot be used as IRQ` is from https://github.com/coolstar/VoodooGPIO/blob/master/VoodooGPIO/VoodooGPIO.cpp#L342. The pin might be configured in ACPI mode wrongly.

I remove the `TPDM = Zero` it still doesn't work.
[i2c_without_TPDX.log](https://github.com/VoodooI2C/VoodooI2C/files/5358495/i2c_without_TPDX.log)
[SSDT-TPDX.aml 2.zip](https://github.com/VoodooI2C/VoodooI2C/files/5358496/SSDT-TPDX.aml.2.zip)

zhen-zen, 2020-10-10:
https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/prebuilt.html

M69W, 2020-10-10:
> https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/prebuilt.html

thanks, I will try it

ben9923, 2020-10-10:
The touchscreen isn't working as 2.5+ is trying to use GPIO interrupts for it, extracted from _DSM, but they aren't configured correctly.

Keep `TPDM = Zero`.
Try adding this too:
```
TPLM = Zero
```

Of course, add an external definition for `TPLM` (And remove the one of the unused `TPLT`).

M69W, 2020-10-10:
@zhen-zen  
use i2c 2.5.2, remove the `TPDM = Zero` and enable [the _OSI to XOSI](https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/prebuilt.html)  that only trackpad works.

[Send me z_without_TPDM_and_with_SSDT-XOSI.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359076/Send.me.z_without_TPDM_and_with_SSDT-XOSI.zip)
[Voodoo_without_TPDM_and_with_SSDT-XOSI.log](https://github.com/VoodooI2C/VoodooI2C/files/5359078/Voodoo_without_TPDM_and_with_SSDT-XOSI.log)
[IOREG_z_without_TPDM_and_with_SSDT-XOSI.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359080/IOREG_z_without_TPDM_and_with_SSDT-XOSI.zip)

M69W, 2020-10-10:
> The touchscreen isn't working as 2.5+ is trying to use GPIO interrupts for it, extracted from _DSM, but they aren't configured correctly.
> 
> Keep `TPDM = Zero`.
> Try adding this too:
> 
> ```
> TPLM = Zero
> ```
> 
> Of course, add an external definition for `TPLM` (And remove the one of the unused `TPLT`).

use i2c 2.5.2 and use both `TPDM = Zero` and `TPLM = Zero`.
reboot the system: there are two situations.
- once the Touchscreen was used and it works but trackpad die.
- if  first time  trackpad was used  and it works. then Touchscreen was used and trackpad die only Touchscreen works.

use Touchscreen before:
[IOREG_z_before_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359143/IOREG_z_before_Touchscreen_used.zip)
[i2c_with__before_Touchscreen_used.log](https://github.com/VoodooI2C/VoodooI2C/files/5359147/i2c_with__before_Touchscreen_used.log)
[Send me z_before_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359148/Send.me.z_before_Touchscreen_used.zip)

use Touchscreen after:
[IOREG_z_afer_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359149/IOREG_z_afer_Touchscreen_used.zip)
[Voodoo_with_afer_Touchscreen_used.log](https://github.com/VoodooI2C/VoodooI2C/files/5359152/Voodoo_with_afer_Touchscreen_used.log)
[Send me z_afer_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359150/Send.me.z_afer_Touchscreen_used.zip)

ben9923, 2020-10-10:
> use i2c 2.5.2 and use both `TPDM = Zero` and `TPLM = Zero`.
> reboot the system,
> **there are two situations**:
> 
> * once the Touchscreen was used and it works but trackpad die.
> * if  first time  trackpad was used  and it works. then Touchscreen was used and trackpad die only Touchscreen works.
> 
> above the two situations after only Touchscreen works.
> 
> Video:
> [1602338564997069.mp4.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359159/1602338564997069.mp4.zip)
> 
> use Touchscreen before:
> [IOREG_z_before_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359143/IOREG_z_before_Touchscreen_used.zip)
> [i2c_with__before_Touchscreen_used.log](https://github.com/VoodooI2C/VoodooI2C/files/5359147/i2c_with__before_Touchscreen_used.log)
> [Send me z_before_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359148/Send.me.z_before_Touchscreen_used.zip)
> 
> use Touchscreen after:
> [IOREG_z_afer_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359149/IOREG_z_afer_Touchscreen_used.zip)
> [Voodoo_with_afer_Touchscreen_used.log](https://github.com/VoodooI2C/VoodooI2C/files/5359152/Voodoo_with_afer_Touchscreen_used.log)
> [Send me z_afer_Touchscreen_used.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359150/Send.me.z_afer_Touchscreen_used.zip)

Try adding an OS patch too. Notice the one you previously used seems to be emulating an old version of Windows. Add `Windows 2015`.

Note you're injecting VoodooPS2 mouse & trackpad plugins you don't actually need. Add those to your config.plist and disable them.

M69W, 2020-10-10:
@ben9923 Thanks for help.

add `Windows 2015` in `SSDT-XOSI.aml` and [set XOSI Rename](https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/prebuilt.html)(add this under config.plist -> ACPI -> Patch)
disable VoodooPS2 mouse & trackpad plugins (add them under config.plist -> Kernel -> Add `set Enabled false`)

if I did right thing and this same result without luck that only  Touchscreen works.
[Send me z.zip](https://github.com/VoodooI2C/VoodooI2C/files/5359319/Send.me.z.zip)

zhen-zen, 2020-10-10:
I suppose the simplest way based on your old config is to break `_DSM` mechanism and get it in polling mode.
```

    Scope (_SB.PCI0.I2C0.TPL1)
    {
        Name (TP7G, ToUUID ("ef87eb82-0000-0000-0000-14871ac6f84b"))
    }

```

However, your touchscreen should be able to run in GPIO interrupt mode, which is more efficient. And there's no `cannot be used as IRQ` prompt after applying the `XOSI` patch. So we seem to get it back on track. It's weird that `VoodooInput` is not attached to `VoodooI2CNativeEngine` for touchscreen though.

M69W, 2020-10-11:
> I suppose the simplest way based on your old config is to break `_DSM` mechanism and get it in polling mode.
> 
> ```
> 
>     Scope (_SB.PCI0.I2C0.TPL1)
>     {
>         Name (TP7G, ToUUID ("ef87eb82-0000-0000-0000-14871ac6f84b"))
>     }
> ```
> 
> However, your touchscreen should be able to run in GPIO interrupt mode, which is more efficient. And there's no `cannot be used as IRQ` prompt after applying the `XOSI` patch and `TPLM` & `TPDM`. So we seem to get it back on track. It's weird that `VoodooInput` is not attached to `VoodooI2CNativeEngine` for touchscreen though.
> 
> How about just keep `TPLM = 0`?

After reboot system both touchscreen and trackpad can work, but once touchscreen used again.
just left only touchscreen works and trackpad can't work any more.
[Send me z_only_TPLM.zip](https://github.com/VoodooI2C/VoodooI2C/files/5360539/Send.me.z_only_TPLM.zip)

zhen-zen, 2020-10-11:
Could that be some kind of interference?
Can you try insert the code above to either touchpad or touchscreen scope? This will force one of them into pulling mode. I wonder if either will be good?

M69W, 2020-10-11:
> Could that be some kind of interference?
> Can you try insert the code above to either touchpad or touchscreen scope? This will force one of them into polling mode. I wonder if either is good?

I'm sorry.
insert the Code in SSDT-TPDX.aml like this?:
```
DefinitionBlock ("", "SSDT", 2, "hack", "TPDX", 0x00000000)
{
    External (_SB_.PCI0.I2C0, DeviceObj)
    External (_SB_.PCI0.I2C0.TPL1, DeviceObj)
    External (TPLM, FieldUnitObj)

    Scope (\)
    {
        If (_OSI ("Darwin"))
        {
            TPLM = Zero
        }
    }

    Scope (_SB.PCI0.I2C0.TPL1)
    {
        Name (TP7G, Buffer (0x10)
        {
            /* 0000 */  0x82, 0xEB, 0x87, 0xEF, 0x00, 0x00, 0x00, 0x00,  // ........
            /* 0008 */  0x00, 0x00, 0x14, 0x87, 0x1A, 0xC6, 0xF8, 0x4B   // .......K
        })
    }
}
```
[SSDT-TPDX.aml 2.zip](https://github.com/VoodooI2C/VoodooI2C/files/5360783/SSDT-TPDX.aml.2.zip)

zhen-zen, 2020-10-11:
Yes. And change the scope to `I2C1.TPD0` for another attempt.

M69W, 2020-10-12:
> Yes. And change the scope to `I2C1.TPD0` for another attempt.

same  before.

```
DefinitionBlock ("", "SSDT", 2, "hack", "TPDX", 0x00000000)
{
    External (_SB_.PCI0.I2C0.TPL1, DeviceObj)
    External (_SB_.PCI0.I2C0.TPL1.TP7G, BuffObj)
    External (_SB_.PCI0.I2C1.TPD0, DeviceObj)
    External (_SB_.PCI0.I2C1.TPD0.HIDG, BuffObj)
    External (TPLM, FieldUnitObj)

    Scope (\)
    {
        If (_OSI ("Darwin"))
        {
            TPLM = Zero
        }
    }

    Scope (_SB.PCI0.I2C0.TPL1)
    {
        Name (TP7G, ToUUID ("ef87eb82-f951-46da-84ec-14871ac6f84b"))
    }

    Scope (_SB.PCI0.I2C1.TPD0)
    {
        Name (TP7G, ToUUID ("ef87eb82-f951-46da-84ec-14871ac6f84b"))
    }
}
```
[Send me z.zip](https://github.com/VoodooI2C/VoodooI2C/files/5362361/Send.me.z.zip)

zhen-zen, 2020-10-15:
Oops, don't apply them both at the same time. The patch intends to keep one in polling mode and another with GPIO interrupt. According to your previous experience, `TPL1` one will force your touchscreen into polling mode, while the touchpad should be good with GPIO. However, the workaround is not optimal as I mentioned earlier.

M69W, 2020-10-25:
Sorry for the past days, we got some good news.
WCOM488F(touchscreen) and SYNA2393(trackpad) both supported by GPIO interrupt.
Has anyone tried they are both work well on the same time with I2C?

M69W, 2020-10-26:
@ben9923 @zhen-zen 
With the help of your and `宪武`(Sorry, I don't his github's account ). 
宪武 said my xps's OSYS(DSDT) is supported by Windows 2015.
so we don't need `_OSI to XOSI`([OC-little/04-操作系统补丁](https://github.com/daliansky/OC-little/tree/master/04-%E6%93%8D%E4%BD%9C%E7%B3%BB%E7%BB%9F%E8%A1%A5%E4%B8%81)) or  [OC-little/19-I2C专用部件](https://github.com/daliansky/OC-little/tree/master/19-I2C%E4%B8%93%E7%94%A8%E9%83%A8%E4%BB%B6)
[image]
[image]

### SSDT-GPI0-disable.aml
```
// Disable GPI0
//
// In config ACPI, GPI0:_STA to XSTA
// Count:          1
// Find:           5F 53 54 41 00
// Replace:        58 53 54 41 00
// Skip:           3
// TableSignature: 44 53 44 54
//
DefinitionBlock("", "SSDT", 2, "OCLT", "GPI0no", 0)
{
    External(_SB.PCI0.GPI0, DeviceObj)
    External(_SB.PCI0.GPI0.XSTA, MethodObj)
    
    Scope (_SB.PCI0.GPI0)
    {
        Method (_STA, 0, NotSerialized)
        {
            If (_OSI ("Darwin"))
            {
                Return (0)
            }
            Else
            {
                Return (\_SB.PCI0.GPI0.XSTA())
            }
        }
    }
}
```

A: when use `SSDT-GPI0-disable.aml` and `I2C(2.4.4 version)`, 
Both trackpad and touchscreen works together.
[image]
IOREG: [z-SSDT-GPI0-disable-I2C(2.4.4).zip](https://github.com/VoodooI2C/VoodooI2C/files/5440086/z-SSDT-GPI0-disable-I2C.2.4.4.zip)
Send me(RunMe): [Send me z-SSDT-GPI0-disable-I2C(2.4.4).zip](https://github.com/VoodooI2C/VoodooI2C/files/5440092/Send.me.z-SSDT-GPI0-disable-I2C.2.4.4.zip)

B: when use `SSDT-GPI0-disable.aml` and `I2C(2.5.2 version)`, 
neither trackpad and touchscreen don't work.
[image]
IOREG: [z-SSDT-GPI0-disable-I2C(2.5.2).zip](https://github.com/VoodooI2C/VoodooI2C/files/5440111/z-SSDT-GPI0-disable-I2C.2.5.2.zip)
Send me(RunMe): [Send me z-SSDT-GPI0-disable-I2C(2.5.2).zip](https://github.com/VoodooI2C/VoodooI2C/files/5440112/Send.me.z-SSDT-GPI0-disable-I2C.2.5.2.zip)

### SSDT-TPD0_TPL1-APIC.aml
```
// I2C1_TPD0 APIC enable
// I2C0_TPL1 APIC enable
DefinitionBlock ("", "SSDT", 2, "ACDT", "TPxxAPIC", 0x00000000)
{
    External (TPDM, FieldUnitObj)
    External (TPLM, FieldUnitObj)

    Scope (\)
    {
        If (_OSI ("Darwin"))
        {
            TPDM = One
            TPLM = One
        }
    }
}
[... 168 lines cut ...]
                }
            }

            Method (_CRS, 0, NotSerialized)  // _CRS: Current Resource Settings
            {
                Return (ConcatenateResTemplate (I2CM (I2CX, BADR, SPED), SBFI))
            }
        }
    }
}
```

C: when use `SSDT-TPD0_TPL1-APIC.aml` and `SSDT-OCI2C1-TP01-xps9570.aml` with `I2C(2.4.4 version)`, 
Both trackpad and touchscreen works together.
[image]
IOREG: [z-I2C(2.4.4)-SSDT-TPD0_TPL1-APIC-SSDT-OCI2C1-TP01-xps9570.zip](https://github.com/VoodooI2C/VoodooI2C/files/5440147/z-I2C.2.4.4.-SSDT-TPD0_TPL1-APIC-SSDT-OCI2C1-TP01-xps9570.zip)
Send me(RunMe): [Send me z-I2C(2.4.4)-SSDT-TPD0_TPL1-APIC-SSDT-OCI2C1-TP01-xps9570.zip](https://github.com/VoodooI2C/VoodooI2C/files/5440151/Send.me.z-I2C.2.4.4.-SSDT-TPD0_TPL1-APIC-SSDT-OCI2C1-TP01-xps9570.zip)

D: when use `SSDT-TPD0_TPL1-APIC.aml` and `SSDT-OCI2C1-TP01-xps9570.aml` with `I2C(2.5.2 version)`, 
Only trackpad works.
[image]
IOREG: [z-I2C(2.5.2)-SSDT-TPD0_TPL1-APIC-and-SSDT-OCI2C1-TP01-xps9570.zip](https://github.com/VoodooI2C/VoodooI2C/files/5440169/z-I2C.2.5.2.-SSDT-TPD0_TPL1-APIC-and-SSDT-OCI2C1-TP01-xps9570.zip)
Send me(RunMe): [Send me z-I2C(2.5.2)-SSDT-TPD0_TPL1-APIC-and-SSDT-OCI2C1-TP01-xps9570.zip](https://github.com/VoodooI2C/VoodooI2C/files/5440172/Send.me.z-I2C.2.5.2.-SSDT-TPD0_TPL1-APIC-and-SSDT-OCI2C1-TP01-xps9570.zip)

I could offer what you want and just let me know to get them? (Sorry，the first time I got in touch with ACPI)

M69W, 2020-10-26:
> @ben9923 @zhen-zen
> With the help of your and `宪武`(Sorry, I don't his github's account ).
> 宪武 said my xps's OSYS(DSDT) is supported by Windows 2015.
> so we don't need `_OSI to [...]

more test before the above (my self): 
they are works In I2C(2.4.4 and 2.5.2)

[I2C-2.5.2-enable-trackpad-and-enable-touchscreen.zip](https://github.com/VoodooI2C/VoodooI2C/files/5440256/I2C-2.5.2-enable-trackpad-and-enable-touchscreen.zip)

[I2C-2.4.4-enable-trackpad-and-enable-touchscreen.zip](https://github.com/VoodooI2C/VoodooI2C/files/5440250/I2C-2.4.4-enable-trackpad-and-enable-touchscreen.zip)

[only-use-(SSDT-OCI2C0-TP02-touchscreen.aml)--enable-trackpad-and-enable-touchscreen).zip](https://github.com/VoodooI2C/VoodooI2C/files/5440259/only-use-.SSDT-OCI2C0-TP02-touchscreen.aml.--enable-trackpad-and-enable-touchscreen.zip)

M69W, 2020-10-26:
https://github.com/williambj1/GenI2C
https://github.com/williambj1/GenI2C/releases

fewtarius, 2020-10-28:
This looks like the same bug or a similar bug to #325

M69W, 2020-11-18:
@kprinssu @ben9923 @fewtarius @zhen-zen @宪武 @xxxzc

Thanks a lot for your more more help about this issues.
Just like @zhen-zen said 
>  This will force one of them into polling mode. 

With the help of @@xxxzc, force touchscreen  device into polling mode and trackpad with GPIO interrupt.
So, they works.

```
DefinitionBlock ("", "SSDT", 2, "ACDT", "TPD0_IRQ", 0x00000000)
{
    External (_SB_.PCI0.I2C0.TPL1, DeviceObj)
    External (TPDM, FieldUnitObj)

    Scope (\)
    {
        If (_OSI ("Darwin"))
        {
            TPDM = Zero
        }
    }

    Scope (_SB.PCI0.I2C0.TPL1)
    {
        If (_OSI ("Darwin"))
        {
            Name (TP7G, Buffer (0x10)
            {
                /* 0000 */  0x82, 0xEB, 0x87, 0xEF, 0x00, 0x00, 0x00, 0x00,  // ........
                /* 0008 */  0x00, 0x00, 0x14, 0x87, 0x1A, 0xC6, 0xF8, 0x4B   // .......K
            })
        }
    }
}
```

come back this issues, I still don't know What is the difference between 2.4.4 and 2.5.2. It beyond my knowledge.

Thanks you all.

thefiredragon, 2020-11-18:
Dear all I have the same issue, with the newest version on my dell xps 9500 Version 2020 Touchscreen is working and Touchpad not, mouse is haning and unusable. With version 2.4.4 Touchpad works fine, Touchscreen Id like to test now, how I could help to solve this with version 2.5.2?

thefiredragon, 2020-11-18:
> @kprinssu @ben9923 @fewtarius @zhen-zen @宪武 @xxxzc
>
> Thanks a lot for your more more help about this issues.
> Just like @zhen-zen said
>
>
> With the help of @@xxxzc, force touchscreen device into polling [...]

Thank you, this solves to work with Dell XPS 9500, Touchscreen and Touchpad, but only at version 2.4.4
2.5.2 still break my elan device attached my ioreg and patched SSDT-i2c.aml

[Uploading Archive.zip…]()

zhen-zen, 2020-11-18:
Theoretically this fix will only work for 2.5.2 since previous version don't read `_DSM` method. Maybe you can try https://github.com/VoodooI2C/VoodooI2C/pull/398 and force polling mode.

thefiredragon, 2020-11-18:
I only patched TPL0 like in the example that only the touchpad is in polling mode.

With this patch the touchpad do not stop working After an minute.

But under version 2.5.2 touchpad is hanging and unusable, I can try 
Thanks ;)

M69W, 2020-11-19:
> Theoretically this fix will only work for 2.5.2 since previous version don't read `_DSM` method. Maybe you can try #398 and force polling mode.

And I how to try it with #398 (kexts)?

kprinssu, 2020-11-19:
@thefiredragon please don't spam other Github issues, please use this one to help troubleshoot/fix the problem

fewtarius, 2020-11-19:
> @kprinssu @ben9923 @fewtarius @zhen-zen @宪武 @xxxzc
>
> Thanks a lot for your more more help about this issues.
> Just like @zhen-zen said
>
>
> With the help of @@xxxzc, force touchscreen device into polling [...]

This is a hack, not a fix.  Running the touch display in polling mode will cause significant interrupts drastically reducing your battery life.  It's better to just disable and live without it.

M69W, 2020-11-19:
@fewtarius 
If two touch device support GPIO interrupt same time, I2C will make it work？
It seems not to support it with i2c(2.5.2).

fewtarius, 2020-11-20:
> @fewtarius
> If two touch device support GPIO interrupt same time, I2C will make it work？
> It seems not to support it with i2c(2.5.2).

It should, but VoodooI2C doesn't work properly with multiple GPIO devices.  Best to disable the touch display and forget about it.

M69W, 2020-11-20:
> > @fewtarius
> > If two touch device support GPIO interrupt same time, I2C will make it work？
> > It seems not to support it with i2c(2.5.2).
> 
> It should, but VoodooI2C doesn't work properly with multiple GPIO devices. Best to disable the touch display and forget about it.

Sorry about that, touch display is useful for us on Big Sur like iPad.
It will be more and more laptop with touch display in the future.

fewtarius, 2020-11-20:
> Sorry about that, touch display is useful for us on Big Sur like iPad.
> It will be more and more laptop with touch display in the future.

I'm very familiar with how useful it could be.  It would be nice for it to be fixed. :)

M69W, 2020-12-27:
Thanks a lot for focus on this issues.

update the issue: [The touchscreen can work great in polling mode! You should re-enable it.](https://github.com/xxxzc/xps15-9570-macos/issues/74#issuecomment-751479285).

Now there two ways to make them work.

**touchscreen in Polling Mode, touchpad in GPIO Pinning**
use the [release 2.6.1](https://github.com/VoodooI2C/VoodooI2C/releases/tag/2.6.1) and let touchscreen in Polling Mode and USTP to one patch. they are both work together.
[image]
[image]

**touchscreen and touchpad botn in GPIO Pinning**
anther way is just use this SSDT and [release 2.6.1](https://github.com/VoodooI2C/VoodooI2C/releases/tag/2.6.1).
[SSDT-OCI2C0-TP02-touchscreen.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/5744716/SSDT-OCI2C0-TP02-touchscreen.aml.zip).

### the problem is show three device, one of them is fake device.
[image]
[image]
[image]

`log show --last boot | grep -i i2c`
[i2c-2020-12-27.log](https://github.com/VoodooI2C/VoodooI2C/files/5744721/i2c-2020-12-27.log)

Thanks again!
Happy New Year!

ben9923, 2020-12-27:
> Thanks a lot for focus on this issues.
>
> update the issue: [The touchscreen can work great in polling mode! You should re-enable [...]

Happy new year!

I don't understand how adding a fake device made it work. What's happening when you don't add it exactly?
Please attach another log, but `grep -I Voodoo` rather than `I2C` so VoodooGPIO logs will show up as well.
Preferably, attach a full troubleshooting archive. You may add `[SSDT-I2CD`](https://github.com/VoodooI2C/VoodooI2C/blob/master/Documentation/SSDT-I2CD.dsl) too to help us debugging.

M69W, 2020-12-28:
@ben9923 
thanks for your reply.

Only touchPad can work with `SSDT-I2CD.aml`.
use SSDT-I2CD.dsl (SSDT-I2CD.aml ):
[SSDT-I2CD.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/5747526/SSDT-I2CD.aml.zip)

`log show --last boot | grep -i Voodoo > /Users/z/Desktop/Voodo-20201228.log`
[Voodo-20201228.log](https://github.com/VoodooI2C/VoodooI2C/files/5747524/Voodo-20201228.log)

[image]
[image]

ben9923, 2020-12-28:
@M69W Thanks for the files, however I asked for something like the output of [OC-gen-debug](https://github.com/hieplpvip/OC-gen-debug) in addition to the log. Make sure IOReg is included.

Can you also attach logs + archive made when adding the fake device that somehow makes it work? :P

SSDT-I2CD makes no functional difference, it only shows us a couple of values in IOReg for troubleshooting.

thefiredragon, 2020-12-29:
@M69W could you help me to generate same patch for ups 9500?
XPS 9500 uses other device names under acpi then the 9570 one. 
Touchscreen acpi name is called "TPL0" 
For some help I would be really happy at this point. best Regards

hameerabbasi, 2021-01-04:
Hello, I can reproduce the issue, which in short, is the following:

- Initially, touchpad works.
- Touchpad dies after using touchscreen but touchscreen keeps working.

You can find the output of `OC-gen-debug` here: https://1drv.ms/u/s!Alzah7pKkOJVsXTPZEpu2M3yEMyy?e=HVW623

hameerabbasi, 2021-01-04:
Here's a version of `OC-gen-debug` with `SSDT-OCI2C0-TP02-touchscreen.aml` added and `SSDT-TPDX.aml` disabled: https://1drv.ms/u/s!Alzah7pKkOJVsXV9aYRn-PvbD8bG?e=2BttBc

ben9923, 2021-01-09:
@hameerabbasi Please install `DebugEnhancer.kext` and attach the output of the `dmesg` in each case.
Sorry for the hassle, but logs just don't seem to appear normally with the `log` command.

M69W, 2021-02-02:
#435

M69W, 2021-03-30:
updated：
use this SSDT from @WildBeastRouen the touchScreen works well.
change `pin` to `1b`.
[SSDT-TPDX.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/6230292/SSDT-TPDX.aml.zip)
[image]

https://user-images.githubusercontent.com/15809201/113019814-91baa080-91b4-11eb-8969-1d3968386c46.mp4

hameerabbasi, 2021-04-09:
I can confirm that the approach in https://github.com/VoodooI2C/VoodooI2C/issues/392#issuecomment-810386204 works well for me on my 9570.

ben9923, 2021-04-10:
@M69W @hameerabbasi So changing `GPLI` changed the GPIO pin accordingly?
Can you add SSDT-I2CD and attach an IOReg both when setting `GPLI` and keeping it native?

closed 2021-06-19 by M69W