VoodooI2C/VoodooI2C #393: Trackpad stops to work after wake from sleep

alessiomatricardi, 2020-10-11:
**Describe the bug**
When computer wake from sleep, trackpad doesn't work. Before the sleep, it works very well
**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
I asked on Gitter but nobody replyied

**System Environment**
Please answer:
1. What is your operating system? macOS Catalina 10.15.7
2. What version of the drivers (including drivers) are you using? 2.5.2
3. What is your hardware configuration?

**Troubleshooting Archive**
[debug_31287.zip](https://github.com/VoodooI2C/VoodooI2C/files/5362047/debug_31287.zip)

**Additional context**
No additional context

kprinssu, 2020-10-12:
Please provide more information. Have you used VoodooI2C before?

alessiomatricardi, 2020-10-12:
Yes, with VoodooI2C 2.4.4 it worked. But I was using fake device method (TPXX) via SSDT, so it is no reliable. With 2.5.2 and same fake device I got KP so developers helped me on Gitter same days ago to solve that problem. Now I haven't KP but when I stops PC, Trackpad doesn't work on wake.

alessiomatricardi, 2020-10-13:
`2020-10-13 13:11:57.515608+0200  localhost kernel[0]: (VoodooGPIO) VoodooGPIOSunrisePointLP::Going to Sleep!
2020-10-13 13:11:57.518619+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Going to sleep
2020-10-13 13:11:57.519116+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,9d60 Going to sleep
2020-10-13 13:11:57.622736+0200  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::SYNA7DB5 Going to sleep
2020-10-13 13:11:57.622766+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Going to sleep
2020-10-13 13:11:57.622789+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,9d61 Going to sleep
2020-10-13 13:12:01.986182+0200  localhost kernel[0]: (VoodooGPIO) VoodooGPIOSunrisePointLP::Woke up from Sleep!
2020-10-13 13:12:02.002896+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
2020-10-13 13:12:02.004728+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,9d60 Woke up
2020-10-13 13:12:02.005319+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
2020-10-13 13:12:02.006180+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CPCIController::pci8086,9d61 Woke up
2020-10-13 13:12:02.006243+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d60 Woke up
2020-10-13 13:12:02.006830+0200  localhost kernel[0]: (VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Woke up
2020-10-13 13:12:02.118176+0200  localhost kernel[0]: (VoodooGPIO) VoodooGPIOSunrisePointLP::Registering hardware pin 0x55 for GPIO IRQ pin 0x55
2020-10-13 13:12:02.118188+0200  localhost kernel[0]: (VoodooGPIO) VoodooGPIOSunrisePointLP::Successfully registered hardware pin 0x55 for GPIO IRQ pin 0x55
2020-10-13 13:12:02.118209+0200  localhost kernel[0]: (VoodooGPIO) VoodooGPIOSunrisePointLP:: pin 85 cannot be used as IRQ
2020-10-13 13:12:02.118229+0200  localhost kernel[0]: (VoodooI2CHID) VoodooI2CHIDDevice::SYNA7DB5 Woke up`

This is the Voodoo output logs when pc wake up

profzei, 2020-10-13:
Same here with SYNA1D31 device using GPIO pinning mode and latest release VoodooI2C & VoodooI2CHID 2.5.2

mishurov, 2020-10-13:
@alessiomatricardi `pin 85 cannot be used as IRQ`, I have this on boot on Asus UX310UQ, nothing helps, I have to use polling mode.

zhen-zen, 2020-10-15:
Can you provide some log for 2.4.4 after sleep? Is it in polling mode?

alessiomatricardi, 2020-10-15:
> Can you provide some log for 2.4.4 after sleep? Is it in polling mode?

Also with 2.4.4 trackpad doesn't work after sleep. Logs are identical. So I think it is a problem with DSDT, i don't know if it is in polling mode... how I can view it?
2.4.4 was working with fake device via DSDT, but this device caused some KP with 2.5.2 and you helped me on gitter

alessiomatricardi, 2020-10-19:
After some tries, the problem seems fixed
Steps to do:

- Remove all SSDT related to trackpad from ACPI folder

- Add SSDT-XOSI.aml (add "Windows 2015" to the list to support Windows 10) and apply the patch "_OSI to XOSI" in config.plist 
You can find here https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/prebuilt.html

- With MaciASL create new file
Drop this into editor
`DefinitionBlock ("", "SSDT", 2, "hack", "_TPL1", 0x00000000)
{
    External (_SB_.PCI0.I2C1, DeviceObj)
    External (_SB_.PCI0.I2C1.TPL1, DeviceObj)
    External (TPLM, FieldUnitObj)

    Scope (\)
    {
        If (_OSI ("Darwin"))
        {
            TPLM = Zero
        }
    }

    Scope (_SB.PCI0.I2C1.TPL1)
    {
        Name (TP7G, Buffer (0x10)
        {
            /* 0000 */  0x82, 0xEB, 0x87, 0xEF, 0x00, 0x00, 0x00, 0x00,  // ........
            /* 0008 */  0x00, 0x00, 0x14, 0x87, 0x1A, 0xC6, 0xF8, 0x4B   // .......K
        })
    }
}`
**ALERT** my trackpad was the device I2C1.TPL1, replace with yours (usually it has the form I2Cx.TPyz where x = {0,1}, y = {D,L} and z = {0,1})

This should enable polling mode. I will update this guide if there will be the need

For now, i can close the issue. Special thanks to @zhen-zen

closed 2020-10-19 by alessiomatricardi