VoodooI2C/VoodooI2C #399: [Ice Lake] ALPS touchpad is not working after sleep

m0d16l14n1, 2020-11-05:
**Describe the bug**
After boot and *sometimes (very rare case)* after sleep touchpad is working with GPIO interrupts by default (btw, it's working *waaaay*  better than it's working in polling mode, so big thx for developers for that feature). In most cases, after sleep touchpad is unresponsive at all. Tested with FingerMgmt - no input showing. 

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes, we have actually troubleshoot a lot with that trackpad with @ben9923 (he helped a lot with ACPI patches - FCMN/SSCN + **SHPO (GDPI, One) stuff (without that patch touchpad didn't work at all even on first boot)** and finally made my touchpad work), and I have already made lots of logs/archives. Ben told me that logs looks good and normal, nothing suspicious

**Have you searched the issue on Github, Gitter, or Google?**
Yes, I check all the issues here, that didn't help me to fix mine

**System Environment**

1. What is your operating system? Catalina / **Big Sur 11.0.1 beta**
2. What version of the drivers (including drivers) are you using? Latest, VoodooI2C, VoodooI2CHID
3. What is your hardware configuration? 

[CPU] - Intel Core I7-1065G7
[Graphics] - Iris Graphics G7 (Ice Lake)
[RAM] - 16 GB RAM (Crucial 2466)
[STORAGE] - 1T SSD (NVME, Crucial)
[OTHER] 1xTB3 port, BCM94360NG m.2, IPS 15,6

**Troubleshooting Archive**
[before sleep > touchpad working.zip](https://github.com/VoodooI2C/VoodooI2C/files/5494167/before.sleep.touchpad.working.zip)

[after sleep > touchpad not working.zip](https://github.com/VoodooI2C/VoodooI2C/files/5494165/after.sleep.touchpad.not.working.zip)

[DSDT.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/5494201/DSDT.aml.zip)

P.S. I have unlocked BIOS (many settings are possible to change in it). Here is some settings related to touchpad - https://github.com/Ardentwheel/OpenCore-Hasee-X57S1/issues/2

*also in repository you can find all the screenshots of the BIOS settings, if that somehow can help to debug and fix that*

m0d16l14n1, 2020-11-13:
@zhen-zen @kprinssu any thoughts guys? Sorry for ping :(

m0d16l14n1, 2020-11-15:
Tested with the polling mode on previous version (it was the test build with ice lake support) - same problem here, after sleep touchpad is not working at all.

m0d16l14n1, 2020-11-19:
Idk, but maybe it's somehow related to that new Idle Sleep feature of Intel on new platforms? AFAIK - it's appeared on Ice Lake stuff. I have it disabled in BIOS, tried to enable it / disabled-enabled I2C controller in mitigation list, but it's still not working after sleep-wake cycle. 
P.S Also i have HID platfrom enabled-disabled in BIOS - that setting didn't help too, still not working after sleep. Logs are the same

zhen-zen, 2020-11-20:
I'm not sure if it's this case: https://patchwork.kernel.org/project/linux-input/patch/20171106150041.13933-1-hdegoede@redhat.com/
The VID and PID matches yours (0911:5288), but they state it as `SYNA3602`. However, the vid for synaptic should be `06cb`. Maybe you can try VoorooRMI after changing this line https://github.com/VoodooSMBus/VoodooRMI/blob/master/VoodooRMI/Transports/I2C/RMII2C.cpp#L45

m0d16l14n1, 2020-11-21:
@zhen-zen Hm, interesting, i thought VoodooRMI is support only SMBus/Synaptics touchpads. I will check it today, thank you so much for your response, at least now i have something to try and check. Thanks again!

m0d16l14n1, 2020-12-25:
@zhen-zen, hello! I wasn't able to to get touchpad working with VoodooRMI, i don't why, maybe i did something wrong with building kext, but it's just not loading (injecting) to the system. 

Updated to the latest version of VoodooI2C with bus-config expose in IOreg. I will attach it here too. Logs are the same, as previous ones, nothing suspicious. 

[ioreg-bus-config.zip](https://github.com/VoodooI2C/VoodooI2C/files/5742160/ioreg-bus-config.zip)

zhen-zen, 2020-12-25:
Have you modified that line I mentioned above? It will print details in the log. 
Also please add https://github.com/VoodooI2C/VoodooI2C/blob/master/Documentation/SSDT-I2CD.dsl and send a new ioreg dump.

m0d16l14n1, 2020-12-26:
> Have you modified that line I mentioned above? It will print details in the log.

Yep, but maybe i did it wrong, heh. I will redo it today. 
Here is the IOreg with that I2CD - actually it's not worked as should, and i don't know why. I2CD entry in IOreg is empty.
[b0ltun’s MacBook Pro.zip](https://github.com/VoodooI2C/VoodooI2C/files/5742733/b0ltun.s.MacBook.Pro.zip)

zhen-zen, 2020-12-26:
You need to switch to IOACPIPlane (⌘+4) and check the value under it.

zhen-zen, 2020-12-26:
Sorry I should have double check your SSDT first. Can you try this one instead? https://github.com/zhen-zen/VoodooI2C/blob/patch-4/Documentation/SSDT-I2CD.dsl

m0d16l14n1, 2020-12-26:
> Sorry I should have double check your SSDT first. Can you try this one instead? https://github.com/zhen-zen/VoodooI2C/blob/patch-4/Documentation/SSDT-I2CD.dsl

Okay, that's it.

m0d16l14n1, 2021-01-07:
After digging a lot (thx to @zhen-zen and @ben9923 for inspiring me), i found out that my touchpad is not ALPS at all (idk why vendor Hasee used ALPS name for it), it's actually Hantick touchpad (HTIX5288), according to some linux patches i found with my device and vendor ids (thanks to zhen-zhen), also i found some Synaptic touchpads, but i think that's just same "fake" name, because vendor and device ids are equial to my.  (https://github.com/torvalds/linux/blob/71c061d2443814de15e177489d5cc00a4a253ef3/drivers/hid/hid-ids.h#L561)

So, i found that in linux previously there were issues after sleep/reboot with touchpads.
Here is the links with details about that. 

https://patchwork.kernel.org/project/linux-input/patch/20171106150041.13933-1-hdegoede@redhat.com/ (@zhen-zen already mentioned that "Synaptic" touchpad issue above)
https://patchwork.kernel.org/project/linux-input/patch/20180912130705.17431-1-anisse@astier.eu/
https://github.com/torvalds/linux/blob/71c061d2443814de15e177489d5cc00a4a253ef3/drivers/hid/i2c-hid/i2c-hid-core.c#L268

Is it possible to fix that?

m0d16l14n1, 2021-01-08:
Okay, @ben9923 actually helped me a lot with a trick to make touchpad working after sleep. It's unloading and reloading kext after sleep. 

```
sudo kextunload -b com.alexandred.VoodooI2CHID
sudo kextunload -b com.alexandred.VoodooI2C

# Navigate to the directory with the kexts
sudo chown -R root:wheel VoodooI2C.kext VoodooI2CHID.kext
sudo kextutil -d VoodooI2C.kext VoodooI2CHID.kext
```

Tested already, seems to be working trick for that moment. Need to disable SIP actually to make it work. Will try to automate that trick after sleep process. 
Will keep issue open, maybe someday it will be fixed without that "dirty" trick, heh. Thanks for all one more time!

m0d16l14n1, 2021-01-08:
Big update on that issue

@zhen-zen made a test build for me, but it won't work. I did some changes to his patch (just deleted that [line](https://github.com/zhen-zen/VoodooI2CHID/blob/e5400d6a9999775dded26d1f1969a6bf8f5bea6d/VoodooI2CHID/VoodooI2CHIDDevice.cpp#L416) from file), now touchpad seems to be working after sleep. I still testing, if it's actually works stable.

ben9923, 2021-01-09:
@m0d16l14n1 VoodooI2C v2.6.3 was released with a proper fix, feel free to to close after verifying the release :)

m0d16l14n1, 2021-01-10:
Seems to be working as it should now. Thanks!

closed 2021-01-10 by m0d16l14n1