VoodooI2C/VoodooI2C #401: Kernel panic on boot or resume from sleep with a USB HID orientation sensor

usr-sse2, 2020-11-13:
On Lenovo Yoga 13 both the touchscreen and orientation sensor work, but on boot or resuming from sleep it often crashes. Here is the panic log from boot with keepsyms:
```
panic(cpu 0 caller 0xffffff80010449ea): Kernel trap at 0xffffff7f842191a3, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000009485000, CR4: 0x00000000001626e0
RAX: 0xffffff8017e3a780, RBX: 0x0000000000000000, RCX: 0x04ee235c5a9a003d, RDX: 0xffffff8017c90a70
RSP: 0xffffff9112ee3da0, RBP: 0xffffff9112ee3e10, RSI: 0x0000000000000010, RDI: 0x0000000000000001
R8:  0x0000000000000000, R9:  0xffffff801accb880, R10: 0x0000000000000000, R11: 0xffffff801a92fb00
R12: 0xffffff800180ce00, R13: 0xffffff8017b0d370, R14: 0xffffff801a92fb00, R15: 0xffffff8017e3a780
RFL: 0x0000000000010202, RIP: 0xffffff7f842191a3, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff9112ee3800 : 0xffffff8000f1868d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff9112ee3850 : 0xffffff8001052ab5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff9112ee3890 : 0xffffff800104463e mach_kernel : _kernel_trap + 0x4ee
0xffffff9112ee38e0 : 0xffffff8000ebea40 mach_kernel : _return_from_trap + 0xe0
0xffffff9112ee3900 : 0xffffff8000f17d57 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff9112ee3a00 : 0xffffff8000f18147 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff9112ee3a50 : 0xffffff80016bf2bc mach_kernel : _panic + 0x54
0xffffff9112ee3ac0 : 0xffffff80010449ea mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff9112ee3c40 : 0xffffff80010446e8 mach_kernel : _kernel_trap + 0x598
0xffffff9112ee3c90 : 0xffffff8000ebea40 mach_kernel : _return_from_trap + 0xe0
0xffffff9112ee3cb0 : 0xffffff7f842191a3 com.alexandred.VoodooI2CHID : __ZN15VoodooI2CSensor15setElementValueEP12IOHIDElementj + 0x35
0xffffff9112ee3e10 : 0xffffff7f84213f2b com.alexandred.VoodooI2CHID : __ZN28VoodooI2CAccelerometerSensor13setPowerStateEmP9IOService + 0x43
0xffffff9112ee3e30 : 0xffffff8001610fe4 mach_kernel : __ZN9IOService19driverSetPowerStateEv + 0x184
0xffffff9112ee3ea0 : 0xffffff8001610dea mach_kernel : __ZN9IOService15pmDriverCalloutEPS_ + 0x2a
0xffffff9112ee3ec0 : 0xffffff8000f5a645 mach_kernel : _thread_call_delayed_timer + 0xec5
0xffffff9112ee3f40 : 0xffffff8000f5a171 mach_kernel : _thread_call_delayed_timer + 0x9f1
0xffffff9112ee3fa0 : 0xffffff8000ebe13e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.alexandred.VoodooI2CHID(1.0)[D892AED5-6F7F-3496-B3DA-E014475EB2CB]@0xffffff7f84211000->0xffffff7f84239fff
            dependency: com.alexandred.VoodooI2C(2.5.2)[46356335-D26B-38BF-ACE3-43565C70A2C8]@0xffffff7f841d7000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2956198D-24F2-3790-A9B2-1EAB9434B906]@0xffffff7f8199a000
            dependency: com.apple.iokit.IOHIDFamily(2.0.0)[9643D692-8F23-37B2-9B89-4AA85D94BEE3]@0xffffff7f8237e000

BSD process name corresponding to current thread: kernel_task
Boot args: keepsyms=1 -cdfon msgbuf=1048576 

Mac OS version:
19H15

Kernel version:
Darwin Kernel Version 19.6.0: Thu Oct 29 22:56:45 PDT 2020; root:xnu-6153.141.2.2~1/RELEASE_X86_64
Kernel UUID: 9B5A7191-5B84-3990-8710-D9BD9273A8E5
Kernel slide:     0x0000000000c00000
Kernel text base: 0xffffff8000e00000
__HIB  text base: 0xffffff8000d00000
System model name: MacBookAir5,2 (Mac-2E6FAB96566FE58C)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 12588808282
```

kprinssu, 2020-11-13:
This is a duplicate of https://github.com/VoodooI2C/VoodooI2C/issues/357.

Ideally, I would like to remove the Sensors portion of VoodooI2CHID. @alexandred has previously mentioned it was never fully supported by macOS and designed to only work with his hardware.

closed 2020-11-13 by kprinssu

usr-sse2, 2020-11-14:
> This is a duplicate of #357.
> 
> Ideally, I would like to remove the Sensors portion of VoodooI2CHID. @alexandred has previously mentioned it was never fully supported by macOS and designed to only work with his hardware.

Why, if it works? I wanted to make it myself since 2013 but HID seemed to be too complex to understand with all these reports and descriptors...

kprinssu, 2020-11-16:
As previously mentioned, it was only tested by @alexandred and there really is no maintainer for it. Issues like this crop up and I really don't have the hardware to test with it.

@ben9923 Your thoughts on the above?