VoodooI2C/VoodooI2C #406: Voodooi2c problems with big sur

Aldo97, 2020-11-16:
Hi,
I'm writing here because it seems that voodooi2c and voodoops2(last version) can not work together because of voodooinput.
Trying to boot the last version with 2.1.8 voodoops2 the system can't boot.
Using an older version of voodoops2 (2.0.4) it boots but has some side problems (like ACPI_SMC_PLATFORMPLUGIN stall etc).
I solved using the last version of both the kext and removing the plugin voodooinput from voodooi2c.
I'm personally using clover as bootloader but I know people with the same problem and opencore and removing one of the two voodooinput solves the issue.
Don't know what the devs want to do, in case this is a walk-around to the problem.
I'll leave this issue open unless the devs say otherwise and/or fix the problem.
Good day.

marianopela, 2020-11-16:
What if you remove VoodooInput from VoodooPS2 and keep the one bundled with VoodooI2C? It is supposed to be the better way, as the code in each VoodooI2C version behaves with respect to its specific VoodooInput version

kprinssu, 2020-11-16:
Please use Gitter for support with configuration.

closed 2020-11-16 by kprinssu