VoodooI2C/VoodooI2C #414: Sluggish i2c after wake

TaiPhamD, 2020-11-30:
**Describe the bug**
Waking from sleep causes trackpad to have sluggish performance. Kernel log shows:
```
  kernel: (VoodooI2CServices) <compose failure [UUID]>
```

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**
Please answer:
1. What is your operating system? 
BigSur 11.0.1 (20B50)
2. What version of the drivers (including drivers) are you using?
```
   % kextstat -a | grep Voodoo                            
Executing: /usr/bin/kmutil showloaded --arch-info
No variant specified, falling back to release
   53    1 0                  0xb000     0xb000     x86_64             com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   61    1 0                  0x26000    0x26000    x86_64             org.coolstar.VoodooGPIO (1.1) FB3BE4D0-E5F6-30B4-9797-DE6ADDEC2F90 <13 6 5 3>
   69    1 0                  0x3a000    0x3a000    x86_64             com.alexandred.VoodooI2C (2.5.2) 46356335-D26B-38BF-ACE3-43565C70A2C8 <63 61 53 14 13 6 5 3 1>
   83    0 0                  0x34000    0x34000    x86_64             com.alexandred.VoodooI2CHID (1) D892AED5-6F7F-3496-B3DA-E014475EB2CB <69 63 13 6 5 3>
   89    0 0                  0x13000    0x13000    x86_64             me.kishorprins.VoodooInput (1.0.8) 0FB2D588-4249-3466-9315-D5448ADBF279 <63 6 5 3>
```
3. What is your hardware configuration?

 - Your machine model
    - Dell Latitude 7400
 - Your CPU model
    - Intel i5-8365U

**Troubleshooting Archive**
https://files.gitter.im/553d220a15522ed4b3df8b02/1bXi/i2cservices_wake_bug.zip

**Additional context**
Add any other context about the problem here.

kprinssu, 2020-11-30:
That "error" on its own is hard to debug and not related to the sluggish trackpad performance. You may want to try following Vit9696's suggestion here, https://github.com/acidanthera/bugtracker/issues/1148#issuecomment-690744648

I am not sure what could be the issue here, can you add printf with timestamps and coordinates of the touch event every time the I2C stack handles the input interrupt?

TaiPhamD, 2020-11-30:
I will add the printf with timestamps after I get back. However, I found a few more logs from dmesg after wake:
Before wake:

```
% sudo dmesg | grep Voodoo <---Returns nothing before sleep
% sudo dmesg | grep Voodoo <---After sleeps
[  439.034775]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  439.045175]: VoodooGPIOCannonLakeLP::Registering hardware pin 0xCC for GPIO IRQ pin 0x117
[  439.045201]: VoodooGPIOCannonLakeLP::Successfully registered hardware pin 0xCC for GPIO IRQ pin 0x117
[  439.045239]: VoodooI2CHIDDevice::WCOM48C9 Woke up
[  439.288428]: VoodooInputSimulatorDevice:0x100000334 new user client
[  439.288551]: VoodooInputSimulatorDevice:0x100000334 open by IOHIDLibUserClient 0x10000085a (0x0)
[  439.288602]: VoodooInputSimulatorDevice:0x100000334 close by IOHIDLibUserClient 0x10000085a (0x0)
[  440.046398]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  440.632719]: tcp_timers: tcp_output() returned 0 with retransmission timer disabled for 49309 > 59932 in state 4, reset timer to 450VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  442.048626]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  443.049977]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  444.051288]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  445.052554]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
```

Then if I try dmesg again I am flooded with:

```
[  557.147340]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  558.148258]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  559.148638]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  560.148959]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  561.150190]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  562.151511]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  563.152000]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  564.153241]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  565.153745]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  566.155060]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  567.156396]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  568.157625]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  569.157974]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  570.158167]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  571.158478]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  572.158872]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  573.160147]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  574.160448]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  575.161820]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  576.163016]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  577.163602]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  578.163948]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  579.165221]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  580.166540]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  581.167669]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  582.167921]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  583.168629]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  584.169271]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  585.170537]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  586.171863]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  587.173185]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  588.174432]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  589.175683]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  590.176945]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  591.177805]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[... cut ...]
[  628.206308]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  629.206585]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  630.206857]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  631.207725]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
[  632.208984]: VoodooI2CControllerDriver::pci8086,9de8 Timeout waiting for bus to accept transfer request
```

kprinssu, 2020-11-30:
It looks like your hardware is not responding to the read/write requests. 

The relevant code, https://github.com/VoodooI2C/VoodooI2C/blob/e0cd64d6e20c4693e332ffdb09251af16e1bfcef/VoodooI2C/VoodooI2C/VoodooI2CController/VoodooI2CControllerDriver.cpp#L178

Your device may need non-standard patch, which is currently not supported. I suggest you look at the Linux kernel modules for your device and see if they have some kind of reset fix for wake/sleep.

TaiPhamD, 2020-11-30:
Thank you For the suggestion. Just for me to understand the problem do you think my device is not responding to read/write request sometimes ? Or did the driver just auto switch to polling mode? Because the trackpad still works but just timeout periodically.

TaiPhamD, 2020-12-01:
FYI I was able to resolve my problem. The touchpad i2c works perfectly. The issue with the bus pci8086,9de8  errors are from the touch screen i2c device. I think after sleep the driver can't talk to the touch screen so it slows down the other i2c processing which affected my touchpad. I just updated my touchscreen to return zero so voodoo doesn't try to attach to it.

```
  Scope (_SB.PCI0.I2C0)
    {
        Device (TPL0)
        {
            Name (SBFB, ResourceTemplate ()
            {
                I2cSerialBusV2 (0x004C, ControllerInitiated, 0x00061A80,
                    AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                    0x00, ResourceConsumer, _Y38, Exclusive,
                    )
            })
......
            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                If (_OSI ("Darwin"))
                {
                    Return (Zero)
                }

                If (LEqual (SDS0, 0x08))
                {
                    Return (0x0F)
                }
                Else
                {
                    Return (Zero)
                }
            }
...
}
```

kprinssu, 2020-12-01:
Glad you got it working. Marking it solved.

closed 2020-12-01 by kprinssu

ben9923, 2020-12-17:
@TaiPhamD While it's resolved by disabling the device, that might be a better suggestion:

First read [here](https://openintelwireless.github.io/itlwm/Troubleshooting.html#using-dmesg) on how to get the complete kext logs.
If you see something like `Warning: Error getting bus config, using defaults where necessary` for `9de8`, it might mean the problem has to do with incorrect bus configuration.
As you're using a patched DSDT, you may change the `If (USTP)` condition in it to `If (One)`, that'll expose the bus configuration via ACPI.

b1t0y, 2022-09-13:
> read [here](https://openintelwireless.github.io/itlwm/Troubleshooting.html#using-dmesg) on how to get the complete kext logs.
> If you see something like `Warning: Error getting bus config, using [...]

Hi @ben9923 @TaiPhamD .  I have the same problem with the same laptop.  I tried the latest kext release 2.7 and its sluggish when it wakes up from sleep.  I wanted to know if you could help me to solve the touchscreen to work when waking up from sleep.  I can disable it by doing the _STA to 0, but wanted to use touchscreen if possible.  

@ben9923 patched DSDT, you may change the `If (USTP)` condition in it to `If (One)`, that'll expose the bus configuration via ACPI. will this help?   Sorry im new to modification with voodooi2c stuff.  Thanks in advance.