VoodooI2C/VoodooI2C #417: Self-built kexts does not load

po-stulate, 2020-12-11:
OS: Catalina (1.15.7)
Xcode: 12.2 (12B45b)
OpenCore

I am following [the documentation site](https://voodooi2c.github.io/#Build%20Environment/Build%20Environment)' guide, Build Environment section trying to build VoodooI2c.kext and VoodooI2CHID.kext with some of my patches.

The problem is, everything looks fine until the kext doesn't load even without any patches (I've tried building 2.4.4 tag, 2.5.2 tag and latest master), while the official builds do work. I noticed that VoodooI2c.kext I built had size of 1.3 MB while the official builds are only about 800 KB. Also, if I load it manually with command line, it says link error.

Any idea of what's happening?

kprinssu, 2020-12-12:
Without looking code, no I cannot say what is wrong. I suggest you look at how IOKit personality matching documentation and troubleshoot on your own.

Due to the nature of custom builds, we officially do not provide support for third builds. Please use the Dev channel on Gitter.

closed 2020-12-12 by kprinssu

ben9923, 2020-12-15:
You're building with an Xcode version that builds for Big Sur, while running Catalina.
It can commonly cause link issues...

When we build the release kexts, while we use new Xcode versions, we use old (10.12) macOS SDK, to make it compatible for 10.12+.
Either use an older version of Xcode, or use an older macOS SDK.
Check out the [CI build script](https://github.com/VoodooI2C/VoodooI2C/blob/master/.circleci/config.yml) to see how it's done.