VoodooI2C/VoodooI2C #418: Can it run on AMD devices?

hikaruchang, 2020-12-15:
I tried to use VoodooI2C and VoodooCHID to run on AMD laptop. Unfortunately, the TPAD device was not started, nor did I find that VoodooI2C tried to initialize it.

[DSDT.zip](https://github.com/VoodooI2C/VoodooI2C/files/5693488/DSDT.zip)

m0d16l14n1, 2020-12-15:
@hikaruchang VoodooI2C working only for Intel controllers, not AMD. Once AMD laptops will be fully supported (if ever will be) i think it might be possible to get it working. But for now - it's kinda useless, AMD APU is not supported and not working. I have already mentioned that, when i had AMD laptop and was trying to get touchpad working too, heh

hikaruchang, 2020-12-15:
> @hikaruchang VoodooI2C working only for Intel controllers, not AMD. Once AMD laptops will be fully supported (if ever will be) i think it might be possible to get it working. But for now - it's [...]

I'm using the Mechrevo S2 air (4800H) for testing. At present, only the touchpad and graphics card are not driven, and the others are running normally.

m0d16l14n1, 2020-12-15:
> I'm using the Mechrevo S2 air (4800H) for testing. At present, only the touchpad and graphics card are not driven, and the others are running normally.

I know. Main problem here is that nobody gonna support AMD laptops, until there will be any support for graphics part. Also, power management is not natively supported, so battery life is awful and it's not good for laptop, heh, then AMD APU have some glitches and stuff with broken audio playback. Try to keep that OS as your daily driver and you will see how many things not working or working badly. Touchpad is not the first thing to fix, believe me :)

hikaruchang, 2020-12-15:
> > 我正在使用Mechrevo S2 air（4800H）进行测试。目前，只有触摸板和图形卡没有被驱动，其他都在正常运行。
> 
> 我知道。这里的主要问题是，除非有对图形部分的任何支持，否则没人会支持AMD笔记本电脑。此外，电源管理本身不受支持，因此电池寿命很长，并且对笔记本电脑不利，呵呵，然后AMD APU会出现故障，并且音频播放中断。尝试将该操作系统作为日常驱动程序，您将看到多少事情不正常或运行不正常。触摸板不是要修复的第一件事，相信我:)

AMD support has a long way to go (TAT

ben9923, 2020-12-15:
As @m0d16l14n1 said, VoodooI2C doesn't support AMD hardware, and no such support is planned...
You'll need an Intel-based laptop to achieve an (almost?) fully working setup, I believe.

closed 2020-12-15 by ben9923