VoodooI2C/VoodooI2C #421: ELAN469 stopped working correctly with update 2.6

kasti0, 2020-12-21:
Multitouch gestures don´t work correctly. It seems like finger detection is false, for example scrolling is now 3 instead of 4 fingers

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**

1. What is your operating system? MacOS Bi Sur 11.1
2. What version of the drivers (including drivers) are you using? VoodooI2C + VoodooI2CHID from latest release 2.6
3. What is your hardware configuration? touchpad: ELAN469 as "I2C-HID" device (via windows device manager)

kprinssu, 2020-12-21:
Please reopen after reading the troubleshooting docs. Especially, the part where it explains how to generate and upload troubleshooting files.

closed 2020-12-21 by kprinssu

kasti0, 2020-12-22:
@kprinssu 
I don´t use a patched DSDT so I included my unpatched DSDT and the SSDT-GPIO and SSDT-XOSI
[I2C trouble shooting archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/5728703/I2C.trouble.shooting.archive.zip)

kasti0, 2020-12-22:
as mentioned here https://github.com/VoodooI2C/VoodooI2C/issues/422#issuecomment-749236709 using VoodooI2CHID from VoodooI2C release version 2.5.2 works fine