VoodooI2C/VoodooI2C #422: With version 2.6 I lose complete the control of the trackpad

anonymous-writer, 2020-12-21:
With version 2.6 I lose complete the control of the trackpad. Only the trackpad mouse can be moved. Clicking not longer possible.

Also my Apple's Magic Mouse will no longer work. Means that only a right-click is possible. I don't know why because the mouse has nothing to do with I2C.

Attach my IOREG. Maybe someone has an idea.

[neuer_i2c.zip](https://github.com/VoodooI2C/VoodooI2C/files/5726797/neuer_i2c.zip)

anonymous-writer, 2020-12-21:
Addition: Problem comes from the VoodooI2CHID.kext. When I use the old VoodooI2CHID.kext from 2.5.2 everything works fine.

217heidai, 2020-12-22:
I have the same problem as you

kasti0, 2020-12-22:
I have the same problem and opened an issue with all problem reporting files attached.
https://github.com/VoodooI2C/VoodooI2C/issues/421

rockavoldy, 2020-12-22:
I had same problem too, reported on gitter but no one notice.
Recompiled with change on [line 585](https://github.com/VoodooI2C/VoodooI2CHID/commit/640c0c1c7866f130693b92435671eaf07c7af768#diff-a17b259a35228d1fa6e960ea2b4eb5931aec9585cee01403b5e89a8da9d54493R585) to
```
if (element->conformsTo(kHIDPage_Button, kHIDUsage_Button_1)) {
```

[VoodooI2C v2.6-rockavoldy.zip](https://github.com/VoodooI2C/VoodooI2C/files/5731201/VoodooI2C.v2.6-rockavoldy.zip)

anonymous-writer, 2020-12-22:
Hello @rockavoldy , top work. Helps also on my Trackpad. Works perfekt now.

ben9923, 2020-12-22:
Just to confirm, @anonymous-writer @rockavoldy @kasti0 
You don't have two buttons on your trackpads, right?

kasti0, 2020-12-22:
No buttons for me, just the trackpad.

ben9923, 2020-12-22:
> No buttons for me, just the trackpad.

Yeah, I mean you can just use the whole trackpad as a clickable (Don't mean tap) area.

rockavoldy, 2020-12-22:
@ben9923 yeah, just multitouch trackpad with single center button located below the trackpad.

ben9923, 2020-12-22:
Please try commit https://github.com/VoodooI2C/VoodooI2C/commit/3cdccc954927a69d26c226f8de13f075e5be9932 build artifacts:
https://app.circleci.com/pipelines/github/VoodooI2C/VoodooI2C/183/workflows/94ae948a-be85-4c23-bfb4-2570bacd0a0d/jobs/691/artifacts

It should keep two-button compatibility without breaking some single-button machines.

anonymous-writer, 2020-12-22:
Hello @ben9923 , works perfekt for me. Thank you for your work.

ben9923, 2020-12-22:
Released [2.6.1](https://github.com/VoodooI2C/VoodooI2C/releases/tag/2.6.1) with a proper fix :)

closed 2020-12-22 by ben9923