VoodooI2C/VoodooI2C #426: v.2.6.x not working at all in Mojave: "Trackpad not found" with ELAN1300 - fine in Catalina

LeeBinder, 2021-01-01:
Hi dev team. We had the exact same issue before with some older release. Back then it was because you did not compile with sdk for 10.14+ but 10.15+

Back to 2.5.2 in Mojave -> all fine again.

ELAN1300 with VoodooI2C + VoodooI2CHID

Let me know if you need any more data.

kprinssu, 2021-01-01:
We are using MacKernelSDK to build versions 2.6. There should be a linking error if you tried to load the kexts manually.
Can you try loading the kexts manually?

LeeBinder, 2021-01-01:
you mean, attach a mouse, then
`sudo kextload /Library/Extensions/VoodooI2C.kext 
sudo kextload /Library/Extensions/VoodooI2CHID.kext
`
->
`/Library/Extensions/VoodooI2CHID.kext failed to load - (libkern/kext) link error;`
meaning VoodooI2C.kext loads fine?

kprinssu, 2021-01-01:
Looks like we messed up the MacKernelSDK switch somewhere within VoodooI2C. Looks like the linker is failing due to out of date headers?

LeeBinder, 2021-01-01:
Anything else I could contribute error-report wise, or are you rather asking the other devs?

vit9696, 2021-01-02:
This error usually happens when the headers are actually targeting a too recent version of macOS despite the confusing message. Are you sure you properly set the macOS target when building VoodooI2C? IOHIDEventService does have checks for `__MAC_OS_X_VERSION_MIN_REQUIRED`.

vit9696, 2021-01-02:
Fixed in https://github.com/acidanthera/MacKernelSDK/commit/565b875076c27ce3386c20e74663235fb964eee4. It happened that some headers were duplicated, and VoodooI2C was including an incorrect version. I removed the duplicates, but VoodooI2C might need to be updated to fix the wrong include paths that are now gone.

ben9923, 2021-01-02:
> Fixed in [acidanthera/MacKernelSDK@565b875](https://github.com/acidanthera/MacKernelSDK/commit/565b875076c27ce3386c20e74663235fb964eee4). It happened that some headers were duplicated, and VoodooI2C was including an incorrect version. I removed the duplicates, but VoodooI2C might need to be updated to fix the wrong include paths that are now gone.

Thanks @vit9696 !
Seems like build is now failing for VoodooI2CHID, after changing include paths.

Including `IOKit/hid/IOHIDEventService.h` seems to include `HIDDriverKit/IOHIDEventService.h`.
This one includes `IOKit/hidevent/IOHIDEventService.h`, which doesn't exist now.
Sure the IOKit `IOHIDEventService.h ` should be in the `hid` directory rather than `hidevent`?

Also, should it include HIDDriverKit headers even when targeting 10.12?

vit9696, 2021-01-02:
Sorry, I did not commit the fixed header. Yes, DriverKit headers are fine.

ben9923, 2021-01-02:
@LeeBinder Please try 194833216e7631010278d45ea821c4be7ccee20c build artifacts :)

LeeBinder, 2021-01-03:
@ben9923 thanks. If you test-compiled to verify it's working: can you please zip & attach your compile?

ben9923, 2021-01-03:
@LeeBinder you can find the archive by going to the commit's page and heading to the build job.

https://app.circleci.com/pipelines/github/VoodooI2C/VoodooI2C/196/workflows/3eb730e2-f19e-4d72-ad4c-d2e16091feef/jobs/705/artifacts

LeeBinder, 2021-01-03:
@ben9923 wow, how cool - I did not know about CircleCI till now, thanks for the pointer. Does it compile on-the-fly/ on demand on first access of a build?

re. build 1948332: YES, working fine in Mojave - big thank you to all involved devs. Please take the rest of Sunday off and grab your skis/ snowboards/ surfboards whatever :) 👍 🥇

closed 2021-01-03 by LeeBinder

ben9923, 2021-01-03:
@LeeBinder it builds every new pushed commit to our repo, on all branches. You can click that small green checkmark next to each commit to see more information.
On `master` it also updates the documentation site, and if the version number was bumped - a new release is automatically triggered.

LeeBinder, 2021-01-03:
very cool. 2.6.2 confirmed working fine in Mojave & Catalina here (Big Brother not installed until Apple's spy-spree is undercut/ circumnavigated/ bypassed..).

I also created a screenshot aided click-flow chart for CircleCI :)