VoodooI2C/VoodooI2C #433: Timeout waiting for bus not to be busy

kdmitrienko, 2021-01-31:
**Describe the bug**
I'am using gpio mode.
If you use 5 fingers on the touchpad for 1-2 minutes (this is the easiest and fastest way to reproduce the problem
) the touchpad stops working (that moment at video in archive) and will not turn on again until the restart.

After touchpad begin to not working i got in log (before problem log is clear):
```
(VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Timeout waiting for bus to accept transfer request
(VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Timed out waiting for bus to change state
(VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
(VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
(VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
…………………………
(VoodooI2C) VoodooI2CControllerDriver::pci8086,9d61 Warning: Timeout waiting for bus not to be busy
```

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**
Please answer:
1. 10.15.7 (19H2) 
2. 2.6.3
3. Asus Zenbook UX410UFR (Core i5-8250u, uhd 620)

**Troubleshooting Archive**
[gpio.zip](https://github.com/VoodooI2C/VoodooI2C/files/5900062/gpio.zip)

**Additional context**
This bug is not reproduced on Ubuntu and Windows 10.

kdmitrienko, 2021-02-01:
[polling.zip](https://github.com/VoodooI2C/VoodooI2C/files/5903841/polling.zip)

This bug is repeated in polling mode too. But for now there is no messages in log.

kprinssu, 2021-02-01:
After speaking with @kdmitrienko, this is probably related to https://github.com/VoodooI2C/VoodooI2C/issues/321

closed 2021-02-01 by kprinssu