VoodooI2C/VoodooI2C #436: Provide a way to disable the reset on resume feature

michyprima, 2021-02-02:
Hey!

My touchpad never needed this, even though it is a SYNA2393. Please provide a way to disable this fix. I'm back to an older version for now.

Thank you

kprinssu, 2021-02-02:
The reset quirk only applies to devices which uses VoodooI2CHID, which I suspect that you do. Can you check if your device id and vendor id matches the following: https://github.com/VoodooI2C/VoodooI2CHID/commit/e8e2f90b2f233147b90491e6bfaaec29a10bb518#diff-a47f2568efc61b16f1d5d6411b686bb662433c920345bfb009a9302e5e0752b7R43

If so, I would recommend that we set up a boot arg to explicitly enable this.

In regards to the pointer being sluggish, I cannot help you with out a troubleshooting archive.

michyprima, 2021-02-02:
That's correct. I checked my device id and it's [image]
I never had any problem with VoodooI2CHID including resume from sleep. Don't worry about the pointer issue, I mentioned it just to tell you one more reason why it would be preferable to be able to disable this quirk.

kprinssu, 2021-02-02:
@ben9923 what do you think of a boot-arg enabling/disabling the quirk? I.e. have `voodooi2c-disable-resume` disable the auto enable of the quirk?

ben9923, 2021-02-02:
@michyprima Thanks for the report!
When running a recent version of Linux, is the driver matched for your device the i2c-hid one? Is it working properly?

Actually, until 2.6.3 *all* devices were reset before wake, so really there shouldn't be any change regarding that one for your device...
Please do attach a troubleshooting archive so we can properly troubleshoot that.

@kprinssu Those quirks are a straight Linux port, I kinda doubt we need a feature allowing to disable those. Issue probably lays somewhere else.
Do you see any other relevant change on the VoodooI2CHID pull request? :/

kprinssu, 2021-02-02:
@ben9923 I really don't see why it will not work without looking at the Linux kernel modules and comparing it to what we have. From what I can see, that PR looks good... 

We just have to wait until @michyprima confirms if the issue exists within Linux. I say we should not make it not auto enable it and have some kind of feature flag (i.e boot-arg) to explicitly enable the quirk.

michyprima, 2021-02-06:
Sorry for the delay guys. I'll do my best to describe the issue. Until a few days ago I ran VoodooI2C (2.5.x) then decided to update to 2.6.3. On 2.6.3 my touchpad is unresponsive for several seconds when resuming from sleep then goes back to normal. I'm currently running VoodooI2C 2.6.3 with VoodooI2CHID from 2.6.1 and this does **not** happen.
Ubuntu 20.10 also does not have this issue. I can confirm it is using i2c_hid:
`hid                   135168  3 i2c_hid,hid_multitouch,hid_generic`
I'll attach a troubleshooting archive later.

ben9923, 2021-02-06:
@michyprima You didn't mention 2.6.2, which also had resume-related changes.

Note that on recent macOS versions (Not sure if bootloader-related) you'll need `DebugEnhancer.kext` installed to get the kernel logs, using the `dmesg` command.

usr-sse2, 2021-02-19:
I have this issue too. In 2.6.1 it's OK, but in 2.6.4 the touchpad doesn't work for 12 seconds after waking from sleep. According to the logs, it always takes from 12100 to 12103 milliseconds.
Here is the output of `pmset -g log | grep "VoodooI2CHIDDevice driver is slow"`:
```
2021-02-15 15:45:29 +0300 Kernel Client Acks  	Delays to Wake notifications: [ApplePS2Controller driver is slow(msg: SetState to 2)(607 ms)] [BroadcomBluetoothHostControllerUSBTransport driver is slow(msg: SetState to 2)(371 ms)] [BNBMouseDevice driver is slow(msg: SetState to 1)(1082 ms)] [pci8086,15d2 driver is slow(msg: SetState to 2)(1000 ms)] [TBL2 driver is slow(msg: SetState to 2)(1000 ms)] [pci-bridge driver is slow(msg: SetState to 2)(1000 ms)] [TBL3 driver is slow(msg: SetState to 2)(1000 ms)] [IOThunderboltController driver is slow(msg: SetState to 2)(304 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(2304 ms)] [AppleUSB30HubPort driver is slow(msg: SetState to 3)(404 ms)] [AppleUSB30HubPort driver is slow(msg: SetState to 3)(417 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 2)(416 ms)] [AppleHDAHDMI_DPDriver driver is slow(msg: SetState to 1)(448 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 2)(406 ms)] [AppleHDADriver driver is slow(msg: SetState to 1)(1852 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12100 ms)]           
2021-02-16 10:02:13 +0300 Kernel Client Acks  	Delays to Wake notifications: [ApplePS2Controller driver is slow(msg: SetState to 2)(580 ms)] [TBL2 driver is slow(msg: SetState to 2)(556 ms)] [pci-bridge driver is slow(msg: SetState to 2)(467 ms)] [IGPU driver is slow(msg: SetState to 2)(572 ms)] [HDEF driver is slow(msg: SetState to 2)(573 ms)] [IOThunderboltController driver is slow(msg: SetState to 2)(303 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 2)(412 ms)] [AppleHDAHDMI_DPDriver driver is slow(msg: SetState to 1)(458 ms)] [AppleHDADriver driver is slow(msg: SetState to 1)(1838 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(3284 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12102 ms)]           
2021-02-16 15:33:36 +0300 Kernel Client Acks  	Delays to Wake notifications: [ApplePS2Controller driver is slow(msg: SetState to 2)(552 ms)] [TBL2 driver is slow(msg: SetState to 2)(309 ms)] [IOThunderboltController driver is slow(msg: SetState to 2)(304 ms)] [BNBMouseDevice driver is slow(msg: SetState to 1)(1378 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(2695 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12102 ms)]           
2021-02-16 15:47:20 +0300 Kernel Client Acks  	Delays to Wake notifications: [ApplePS2Controller driver is slow(msg: SetState to 2)(552 ms)] [TBL2 driver is slow(msg: SetState to 2)(309 ms)] [IOThunderboltController driver is slow(msg: SetState to 2)(304 ms)] [BNBMouseDevice driver is slow(msg: SetState to 1)(1378 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(2695 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12102 ms)] [IOThunderboltController driver is slow(msg: SetState to 2)(304 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 2)(420 ms)] [AppleHDADriver driver is slow(msg: SetState to 1)(493 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(3355 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12103 ms)]           
2021-02-16 21:44:04 +0300 Kernel Client Acks  	Delays to Wake notifications: [ApplePS2Controller driver is slow(msg: SetState to 2)(603 ms)] [pci8086,15d2 driver is slow(msg: SetState to 2)(1000 ms)] [IGPU driver is slow(msg: SetState to 2)(1009 ms)] [TBL3 driver is slow(msg: SetState to 2)(1001 ms)] [TBL2 driver is slow(msg: SetState to 2)(1002 ms)] [pci-bridge driver is slow(msg: SetState to 2)(1001 ms)] [HDEF driver is slow(msg: SetState to 2)(1008 ms)] [IOThunderboltController driver is slow(msg: SetState to 2)(303 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 2)(407 ms)] [AppleHDAHDMI_DPDriver driver is slow(msg: SetState to 1)(429 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 2)(359 ms)] [AppleUSB30HubPort driver is slow(msg: SetState to 3)(411 ms)] [AppleUSB30HubPort driver is slow(msg: SetState to 3)(409 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(2166 ms)] [AppleHDADriver driver is slow(msg: SetState to 1)(1870 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12101 ms)] [VoodooI2CPrecisionTouchpadHIDEventDriver driver is slow(msg: SetState to 1)(13106 ms)]           
2021-02-16 23:10:06 +0300 Kernel Client Acks  	Delays to Sleep notifications: [AppleIntelFramebuffer driver is slow(msg: SetState to 1)(1491 ms)] [AppleIntelFramebuffer driver is slow(msg: SetState to 1)(1534 ms)] [AppleHDADriver driver is slow(msg: SetState to 0)(1319 ms)] [powerd is slow(28007 ms)] [BroadcomBluetoothHostControllerUSBTransport driver is slow(msg: SetState to 1)(461 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 0)(7790 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 0)(27462 ms)] [VoodooI2CControllerDriver driver is slow(msg: SetState to 0)(358 ms)] 
2021-02-16 23:10:06 +0300 Kernel Client Acks  	Delays to Wake notifications: [ApplePS2Controller driver is slow(msg: SetState to 2)(602 ms)] [TBL2 driver is slow(msg: SetState to 2)(546 ms)] [TBL3 driver is slow(msg: SetState to 2)(489 ms)] [IGPU driver is slow(msg: SetState to 2)(594 ms)] [HDEF driver is slow(msg: SetState to 2)(595 ms)] [IOThunderboltController driver is slow(msg [... cut ...] 1)(509 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(3349 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12100 ms)]           
2021-02-19 01:41:16 +0300 Kernel Client Acks  	Delays to Wake notifications: [IOThunderboltController driver is slow(msg: SetState to 2)(303 ms)] [BNBMouseDevice driver is slow(msg: SetState to 1)(1002 ms)] [AirPort_BrcmNIC driver is slow(msg: SetState to 2)(3478 ms)] [VoodooI2CHIDDevice driver is slow(msg: SetState to 1)(12101 ms)]           
```

ben9923, 2021-02-19:
@usr-sse2 Indeed it times out after 12 seconds in VoodooI2C. We may want to change it to 5 like Linux does.
Can you please attach a full troubleshooting archive? IOReg, ACPI, dmesg logs, etc.

As there were several releases with sleep/wake related changes, can you please verify which version broke it exactly for you?

@michyprima As I noted before, you didn't address a specific version nor attached a troubleshooting archive.

ben9923, 2021-02-19:
@michyprima @usr-sse2 Please try the attached build. CI isn't working so I built locally.
[wake-irq-fix.zip](https://github.com/VoodooI2C/VoodooI2C/files/6012026/wake-irq-fix.zip)

Changes can be seen here: https://github.com/VoodooI2C/VoodooI2CHID/commit/b5b699f9981745c80656b944b56cc7219bc8be42

usr-sse2, 2021-02-19:
After first wake works immediately without delay, after second and all following wakes doesn't work at all.
```
[   42.343094]: VoodooI2CPrecisionTouchpadHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
[   42.343285]: VoodooI2CTouchscreenHIDEventDriver::setProperties USBMouseStopsTrackpad = 0
[  220.872634]: VoodooI2CControllerDriver::pci8086,9d62 Going to sleep
[  220.873783]: VoodooI2CPCIController::pci8086,9d62 Going to sleep
[  220.985009]: VoodooI2CHIDDevice::WCOM4840 Going to sleep
[  220.985045]: VoodooI2CHIDDevice::DELL0823 Going to sleep
[  220.985081]: VoodooI2CControllerDriver::pci8086,9d60 Going to sleep
[  220.985102]: VoodooI2CControllerDriver::pci8086,9d61 Going to sleep
[  220.985124]: VoodooI2CPCIController::pci8086,9d60 Going to sleep
[  220.985127]: VoodooI2CPCIController::pci8086,9d61 Going to sleep
[  227.692294]: VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
[  227.692325]: VoodooI2CPCIController::pci8086,9d60 Woke up
[  227.692345]: VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
[  227.692362]: VoodooI2CPCIController::pci8086,9d61 Woke up
[  227.692377]: VoodooI2CPCIController::pci8086,9d62 Set PCI power state D0
[  227.692393]: VoodooI2CPCIController::pci8086,9d62 Woke up
[  227.693725]: VoodooI2CControllerDriver::pci8086,9d60 Woke up
[  227.693794]: VoodooI2CControllerDriver::pci8086,9d61 Woke up
[  227.694088]: VoodooI2CControllerDriver::pci8086,9d62 Woke up
[  227.797752]: VoodooI2CHIDDevice::WCOM4840 Woke up
[  227.909336]: VoodooI2CHIDDevice::DELL0823 Device initiated reset accomplished
[  227.909344]: VoodooI2CHIDDevice::DELL0823 Woke up
[  232.365574]: VoodooI2CControllerDriver::pci8086,9d60 Timeout waiting for bus to accept transfer request
[  243.783155]: VoodooI2CControllerDriver::pci8086,9d62 Going to sleep
[  243.783531]: VoodooI2CPCIController::pci8086,9d62 Going to sleep
[  243.884409]: VoodooI2CHIDDevice::DELL0823 Going to sleep
[  243.884449]: VoodooI2CControllerDriver::pci8086,9d61 Going to sleep
[  243.884467]: VoodooI2CPCIController::pci8086,9d61 Going to sleep
[  243.885024]: VoodooI2CHIDDevice::WCOM4840 Going to sleep
[  243.885401]: VoodooI2CControllerDriver::pci8086,9d60 Going to sleep
[  243.885423]: VoodooI2CPCIController::pci8086,9d60 Going to sleep
[  249.717069]: VoodooI2CPCIController::pci8086,9d60 Set PCI power state D0
[  249.717103]: VoodooI2CPCIController::pci8086,9d60 Woke up
[  249.717111]: VoodooI2CPCIController::pci8086,9d61 Set PCI power state D0
[  249.717128]: VoodooI2CPCIController::pci8086,9d61 Woke up
[  249.717135]: VoodooI2CPCIController::pci8086,9d62 Set PCI power state D0
[  249.717156]: VoodooI2CPCIController::pci8086,9d62 Woke up
[  249.718373]: VoodooI2CControllerDriver::pci8086,9d62 Woke up
[  249.718402]: VoodooI2CControllerDriver::pci8086,9d60 Woke up
[  249.718663]: VoodooI2CControllerDriver::pci8086,9d61 Woke up
[  249.819496]: VoodooI2CHIDDevice::WCOM4840 Woke up
[  249.931055]: VoodooI2CHIDDevice::DELL0823 Device initiated reset accomplished
[  249.931065]: VoodooI2CHIDDevice::DELL0823 Woke up
[  254.400716]: VoodooI2CControllerDriver::pci8086,9d60 Timeout waiting for bus to accept transfer request
[  254.735121]: VoodooI2CControllerDriver::pci8086,9d60 Timed out waiting for bus to change state
[  257.363084]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  261.786357]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  265.187663]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  268.558749]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  272.076176]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  275.662058]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  279.288540]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  282.424593]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  286.456474]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  290.077388]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  293.653285]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  294.086316]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  300.926056]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  304.579441]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  308.218844]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  311.825649]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  315.472182]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  319.098185]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  322.717806]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  326.415408]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  329.984551]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
[  333.603129]: VoodooI2CControllerDriver::pci8086,9d60 Warning: Timeout waiting for bus not to be busy
```

usr-sse2, 2021-02-19:
Tested 2.6.2, works OK without timeouts, always works after wake.

ben9923, 2021-02-19:
@usr-sse2 Thanks for testing. Does it only reproduce in the build I attached? Constantly in a couple of separate boots?
The controller code wasn't touched for a while... No idea what happened there.

Anyway it seems like a proper fix to merge in.

usr-sse2, 2021-02-20:
I booted with wake-irq-fix.zip twice, and it happened this way (works after first wake, doesn't work after second, third, fourth) both times. On the first attempt I touched the touchpad right after waking and thought that it may be the reason, but on the second time I waited before touching, and it still didn't work after second wake.

ben9923, 2021-02-20:
@usr-sse2 To me it seems like a bug in the controller code. My understanding of that code isn't too good, but I'm not sure how the changes made to VoodooI2CHID caused it (On your device we pretty much added a wait-for-IRQ step on wake, which shouldn't really change anything related to the controller behavior). On other devices - it behaves just the same.

That said, I'd still want to look at a troubleshooting archive and a dmesg log since boot.

Anyway, we'll probably release 2.6.5 with a couple of additional bug fixes, including the one I attached earlier. I wish we could know if @michyprima's machine behaves the same.

michyprima, 2021-02-21:
@ben9923 I'm testing the wake-irq-fix build right now. My touchpad seems fine - no freezes and works immediately after resume from sleep or reboot. To be clear, I'm not experiencing what @usr-sse2 is. I'll keep using this build and let you know if something goes wrong.

I'm sorry for the delay in my response, sadly I'm having quite a bit of stuff going on with my life. I've been holding on the troubleshooting archive because it will inevitably contain sensitive data unless I censor it which is quite a bit of work. I know I'll sound paranoid but this is my main machine.

ben9923, 2021-02-27:
@usr-sse2 Let's follow up on your second problem on a new issue, as this one appears to be resolved :)

kprinssu, 2021-02-28:
Merging the PR auto closed the issue, I'll reopen it.

ben9923, 2021-03-06:
The issue should've been closed once the PR was merged, as it resolved the issue mentioned by @michyprima.

Discussion on the second, unresolved problem should probably happen on another issue.
@usr-sse2 Please do open one once you're free to do so :)

PR #446 (2021-02-27): Update VoodooI2C, Release v2.6.5
Fixes #436
Fixes #438
closed 2021-03-06 by ben9923