VoodooI2C/VoodooI2C #438: ATML1000 not working with VoodooI2CAtmelMXT

XXCoreRangerX, 2021-02-07:
**Describe the bug**
ATML1000 touchscreen doesn't work with VoodooI2CAtmelMXT on HP EliteBook 840 G4.

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
I did find a few people having the same issue, but it wasn't solved anywhere.

**System Environment**
macOS Big Sur 11.2
VoodooI2C 2.6.4
[HP EliteBook 840 G4](https://support.hp.com/lv-en/document/c05349510) [OpenCore 0.6.6]: i7-7600u, HD620, ATML1000 - more info in debug archive

**Troubleshooting Archive**
[debug.zip](https://github.com/VoodooI2C/VoodooI2C/files/5939894/debug.zip)

**Additional context**
The touchscreen works with VoodooI2CHID, but without any gestures (only one finger supported).
In verbose boot it says that FMCN and SSCN are missing, I don't know how to fix it, tried with SSDT-I2C with no success.

ben9923, 2021-02-12:
VoodooI2CAtmelMXT should not be used for `ATML1000`, the correct one should be `VoodooI2CHID`.

Remove that unnecessary SSDT-I2C as a start.
Provided logs are incomplete. Install `DebugEnhancer.kext` and attach the output of the `dmesg` command.

Try running FingerMgmt.app, is the input properly shown there?

XXCoreRangerX, 2021-02-12:
Hey! Thank you so much for commenting.

`VoodooI2CHID` works for me, but only with **single touch**. I thought `VoodooI2CAtmelMXT` is the one I should use, as my device is named **ATML**.

I made `SSDT-I2C` for **SSCN** and **FMCN** values, that are shown in the kernel log as not implemented. I'll disable it for now since it doesn't work anyway...

I injected `DebugEnhancer.kext` and ran `dmesg`, this is the [output](https://github.com/VoodooI2C/VoodooI2C/files/5973706/dmesg.rtf.zip).

FingerMgmt doesn't detect fingers with `VoodooI2CAtmelMXT`, nor with `VoodooI2CHID`. It does work for my touchpad though.

kprinssu, 2021-02-12:
@XXCoreRangerX Those ACPI errors are related to your battery and are out of the scope of VoodooI2C. Please fix them on your own. The `SSCN` and `FMCN` are warnings and can be ignored.

XXCoreRangerX, 2021-02-12:
Is there anything I can do with the touchscreen multitouch though?

ben9923, 2021-02-12:
@XXCoreRangerX Vendor doesn't always matter. The same applies for the ELAN satellite. Only relatively old ELAN devices use it, newer ones use HID - A common protocol to rule them all (And the only decent one that works well in Windows...).

Another issue in your setup is your TPL0 SSDT - It doesn't load (conflicting methods, obviously), hence does nothing. You can just remove it... OEM setup already uses GPIO interrupts apparently.

That said, it still looks like a bug in VoodooI2CHID with your device. It's not working in mouse emulation mode (where you will expect to only see minimal functionality), but rather using the multitouch event driver.

Can you remove the `VoodooI2CHIDDevice Touchscreen HID Event Driver` personality from VoodooI2CHID's Info.plist and see how it goes? Check FingerMgmt too in that case.

XXCoreRangerX, 2021-02-13:
Thank you so much! The touchscreen now works almost perfectly with all gestures! FingerMgmt now detects 10 fingers. However, when I try to move the cursor with touch, it doesn't really follow my finger. It "runs away". Is it normal behavior?

ben9923, 2021-02-13:
@XXCoreRangerX Cool!
Removing the personality basically made it work as a normal trackpad, so you can't expect it to work like a touchscreen.
It was just to isolate the problem, which now indeed seems to lay in the touchscreen event driver code.

It seems that currently the touchscreen driver will drop any event with more than 5 fingers, we should probably undo that.

Can you try this build, instead?
It's this commit's build: https://github.com/VoodooI2C/VoodooI2C/commit/36308bd181c4c32b7c8ac3c4c32d545db3d5adf9
CI just broke due to some Python package updates, attached a local build.
[Touchscreen-Fix-Attempt-36308bd.zip](https://github.com/VoodooI2C/VoodooI2C/files/5975724/Touchscreen-Fix-Attempt-36308bd.zip)

XXCoreRangerX, 2021-02-13:
The touchscreen now fully works, it is pretty much perfect. However my SMBus touchpad broke after using this version of the kext. It works, but it's buttons aren't working at all. Could you help me with that? I know it's a little bit out of the scope of VoodooI2C, but it will also be helpful for other people having similar issues.

ben9923, 2021-02-13:
@XXCoreRangerX Neat! Thanks for testing :)
I think I just have an old source of VoodooInput locally... Just get the one from VoodooI2C 2.6.4 and put it inside the kext I've uploaded.

I'll open a pull request in VoodooI2CHID. Once we sort our CI we can release a new version with that fix included.
Let's keep this issue opened until it happens.

XXCoreRangerX, 2021-02-13:
Just replaced the old VoodooInput with the one I downloaded from the releases tab. It is working flawlessly now. Both the touchpad and the touchscreen are working amazing.  I'll keep the issue opened, as you told me to. Thank you so much for your time and patience.

blankmac, 2021-02-14:
> It seems that currently the touchscreen driver will drop any event with more than 5 fingers, we should probably undo that.
> 

There was a reason @alexandred wanted it to drop more than 5 waaaay back when but for the life of me I can't remember why, lol.

alexandred, 2021-02-14:
> > It seems that currently the touchscreen driver will drop any event with more than 5 fingers, we should probably undo that.
> 
> There was a reason @alexandred wanted it to drop more than 5 waaaay back when but for the life of me I can't remember why, lol.

If I recall correctly it was because macOS didn’t support more than 5 touch points at a time and would start ignoring certain fingers as a result. Things could have since changed but I have no been up to date with these things to be honest.

PR #446 (2021-02-27): Update VoodooI2C, Release v2.6.5
Fixes #436
Fixes #438
closed 2021-02-27 by kprinssu