VoodooI2C/VoodooI2C #442: Request for change in VoodooI2CHIDSYNA3602Device

m0d16l14n1, 2021-02-14:
**Describe the bug**
The bug is touchpad not working after hot reboot from Windows

```
[ 3.990763]: VoodooI2CHIDDevice::ALPS0001 Unexpected size of HID descriptor: 0
[ 4.007027]: VoodooI2CHIDDevice::ALPS0001 Could not get HID descriptor
```

So, i was trying to debug why it's not working. Actually we did it together with @ben9923, so he did receive all my debug files/logs and etc

After digging a lot, i find out that in Linux it's overriding descriptor [here](https://github.com/torvalds/linux/blob/28a17338738839494752c5da3e12c889a55219db/drivers/hid/i2c-hid/i2c-hid-dmi-quirks.c#L50)

I made teardown and find sticker on touchpad with same name mentioned in the link above (SIPODEV SP1064)

It's same vendor and device id which i have in my fake "ALPS" touchpad actually. But afaik it's matching in Linux in same way like it did in VoodooI2C: via SYNA3602 name. (I still don't know how it's working in Linux, because i wasn't able to get debug files from Linux to understand how that matching working there > afaik it's matching via name of touchpad: in that case SYNA3602 or DMI info, but my touchpad fake name is ALPS and there is no DMI info in that list about my laptop)

I have already mentioned [there](https://github.com/VoodooI2C/VoodooI2C/issues/399#issuecomment-756231001), that's there are lots of reports of issues with same touchpad having different names, but have same vendor and device id (0911/5228). 

So, i just change SYNA3602 in plist file (under VoodooI2CHIDSYNA3602) to ALPS0001 to match my touchpad name. Rebooted to Windows > rebooted to macOS > touchpad is now working like it should!

[image]

Here is the part of log: 

```
[ 3.990763]: VoodooI2CHIDDevice::ALPS0001 Unexpected size of HID descriptor
[ 4.007027]: VoodooI2CHIDDevice::ALPS0001 Could not get HID descriptor
[ 4.009644]: VoodooI2CHIDSYNA3602Device:0x100000319 start
[ 4.009655]: VoodooGPIOIceLakeLP::Registering hardware pin 0xA1 for GPIO IRQ pin 0xE8
[ 4.026944]: VoodooGPIOIceLakeLP::Successfully registered hardware pin 0xA1 for GPIO IRQ pin 0xE8
[ 4.348694]: VoodooI2CHIDSYNA3602Device::ALPS0001 Overriding report descriptor
[ 4.349164]: VoodooI2CHIDSYNA3602Device:0x100000319 creating interfaces
```

So, i want to know if it's possible to change the system of matching? I mean to match via vendor and device id, not the name of touchpad, because it can be different and totally relies on vendors. 

I know, i can just change that SYNA3602 > ALPS0001 in plist file after every update of VoodooI2C kexts, but it can fix issues with some of next and unknown issues with same touchpads around the world, heh :)

kprinssu, 2021-02-15:
This would be best to be done with SSDT patch. You should be able to rename the ACPI device, though I forget which field does that.

m0d16l14n1, 2021-02-15:
Sure, i don't mind to do in that way. Just strange that name is SYNA3602, but vendor and device id are not Synaptics at all (according to VoodooRMI info for example), so i though it would okay to do matching via device and vendor id. 

[dsdt.zip](https://github.com/VoodooI2C/VoodooI2C/files/5982860/dsdt.zip)

ben9923, 2021-02-16:
AFAIK, Linux is doing both DMI & device name matching.
As we don't have any DMI framework to rely on (And DMI data under macOS is probably a fake Mac-like data), we chose to only match via device name.
Feel free to suggest an alternative, or we may just ask users to inject the device's DMI system name as a device property somehow.

It is impossible to do matching via `Vendor ID:Device ID`. We get those from the HID descriptor which we get, well, from the override data.

I believe Linux doesn't use descriptor overrides for your device (Please prove otherwise), and you should be seeing the same behavior under it.
To prove it's using an override you may look for `Using a HID descriptor override`, otherwise `Fetching the HID descriptor` in the I2C HID logs.
If that's really the case, you may suggest adding the DMI+ALPS001 data to the Linux devs so we can keep a source matching to the Linux one.

BTW, until we have a proper fix, you can also try an injector, code-less kext that just adds the same personality matching, with your device name.
That way VoodooI2CHID is kept unmodified for ease of update.

m0d16l14n1, 2021-02-17:
Okay, thx! So i think i will just use rename patch ALPS0001 > SYNA3602 to get that matching working

```
Find: 414C5053 30303031
Replace: 53594E41 33363032
```

Once i find out how to get those debug logs in Linux properly and will get them - i will just post new info here

closed 2021-09-30 by m0d16l14n1