VoodooI2C/VoodooI2C #447: Unable to use keyboard on Surface Pro's type cover.

Xiashangning, 2021-03-10:
- Situation: The Surface Type Cover is connected using a proprietary USB connector. With the **official version of VoodooI2C**, the trackpad works fine and has no problem detaching or reattaching the type cover to the computer. All gestures are recognized.  However, the keyboard does not work (no response at all, just like does not exist). 

- BUT, there is a custom version which supports both the trackpad and the keyboard. The problem is that there is no source code released and it is limited to 2.4.2. Although it does have a version of 2.5.2, it will cause a kernel panic whenever reattaching the type cover again(e.g close the lid to put it into sleep and open to wake it up). The kernel reports that it `Trying to change a collection in the registry"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-7195.81.3/libkern/c++/OSCollection.cpp`.

So I took a look at the IORegistry and compared the official version with the custom one. The photos are below and I find that for the official version, there is something called `TrackpointDevice` attached along with two VoodooDevices. This device seems to be an apple's driver for usb trackpad?

[image]
[image]

I know that I can just use the custom one of 2.4.2 but 
1. It is from an unknown developer
2. It is outdated, the new version has tons of performances improvement and bug fixes.
3. I would like to contribute to the development of VoodooI2C and help make it more universal.

By opening this issue, I hopes that we could fix this bug for SP users with the official version. I am very glad to do whatever I can to help the debug process. Maybe we could start by disabling this `TrackpointDevice` from attaching to the type cover?

The custom one is attached in case you need it.
[custom I2C.zip](https://github.com/VoodooI2C/VoodooI2C/files/6113631/custom.I2C.zip)

kprinssu, 2021-03-10:
Typically, VoodooI2C does not support keyboards. That being said, We would love to support it, however I am not aware of what changes were made to the kext.

I am quite reluctant to disassemble the kext and try to figure out what the differences are. I also will point out that we don't provide support for custom built changes for VoodooI2C.

If someone would be able to create a PR with changes supporting your hardware, we will be able to merge it in.

1Revenger1, 2021-03-10:
TrackpointDevice doesn't have anything to do with your issue. It's just an extra device which takes trackpoint/button packets. It wouldn't prohibit your keyboard working at all.

ben9923, 2021-03-12:
It's possible you're referring to https://github.com/shdkpr2008/VoodooI2CHID

Try building it with a VoodooI2C version from the time the fork's parent commit dates to verify it really is this one (Assuming you have an OS version that's capable of running it).
Someone will need to update the fork with upstream VoodooI2CHID, assuming it's this one anyway.

Xiashangning, 2021-03-12:
@ben9923 Thank you so much! I have compared his code to your official version(2.6.5) and the only thing I have done is to copy his keyboard implementation into the official one. You guess what? It not only works perfectly but also gets ride of the KP I ran into before. I have tested it for a while and nothing unexpected happened.

So yes, maybe the one I got from some random website is his version. (But outdated) Thanks again.

Now the only question remains is that I am not sure if it is proper that I submit a PR, since it is not code... Or I can make some clarification comments in the code?

kprinssu, 2021-03-12:
@Xiashangning The code is released under the GPL V3. I am not a lawyer but the original author most likely released under the same license (as it is considered a derivative work). If you wish to create a PR, please do so, there shouldn't be any issues.

Xiashangning, 2021-03-13:
PR submitted, so far no KP has occurred. It works pretty fine, hot plug has no issues, after sleeping and waking up Mac still recognizes it,  backlight and volume shortcut works.

kprinssu, 2021-03-16:
Thanks for submitting https://github.com/VoodooI2C/VoodooI2CHID/pull/52, I am hoping we can merge this in time for the next release (sometime in April).

MrHomebrew, 2021-06-21:
Any chance we could get a copy of this updated kext by any chance?

ben9923, 2021-09-08:
Hey, does the keyboard work natively when VoodooI2CHID is not installed?

Xiashangning, 2021-09-10:
Yes, when i did not use VoodooI2C kext, the keyboard works well out of box (except that the touchpad is not responsive in this case)

ben9923, 2021-09-10:
@Xiashangning Huh, damn.
CC @1Revenger1 @kprinssu 

What if you add something like this to the Info.plist of current release's VoodooI2CHID?
```
<key>IOMatchCategory</key>
<string>VoodooI2CMultitouchHIDEventDriver</string>
```
Add it to the `VoodooI2CHIDDevice Multitouch HID Event Driver (VoodooI2CMultitouchHIDEventDriver)` dict.

It might allow both to work :/

1Revenger1, 2021-09-10:
[VoodooI2CHID.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/7145699/VoodooI2CHID.kext.zip)
Here's a version with the above modification (I just applied it to the latest release)
I don't see why it wouldn't work if the keyboard works out of the box without VI2C HID

ben9923, 2021-09-10:
@1Revenger1 I believe you are right, both will match. HID trackpads have both pairs.
All event driver classes should have a `Trackpad` category for the sake of the test, I guess.

1Revenger1, 2021-09-10:
> 
> 
> @1Revenger1 I believe you are right, both will match. HID trackpads have both pairs.
> All event driver classes should have a `Trackpad` category for the sake of the test (And for a proper implementation), I guess.
> 
> BTW @Xiashangning I would be happy if you attach an IOReg using that kext (with proper changes as we discussed), either if it works or not. Thanks :)

Which personalities should share the same iomatchcategory?

From what I see, there is:
Touchscreen HID
Stylus HID
Precision Touchpad HID
Multitouch HID
Generic Mouse HID

Should the touchscreen and the last three have their own category, while the Stylus has it's own category?
Kinda confused, especially since Multitouch HID's usage pages overlap with both the Touchscreen HID and Precision Touchpad HID.

ben9923, 2021-11-13:
I'm planning to merge VoodooI2C/VoodooI2CHID#56 regardless of the confirmation.

I would still be happy if you test it @Xiashangning

Xiashangning, 2021-11-14:
Well, as for the very specific Surface Pro case, I decided to provide a custom version of VoodooHID as you can see in my repo (other machines don't have this issue), so I am totally ok with that reversion and sorry for the problem I brought in. I have tried other ways, they all fails to drive both the keyboard and touchpad of Surface's Typecover, so it seems to me providing a custom version without disturbing others might be the best solution

Xiashangning, 2021-11-14:
Oh, by the way I have tested that build and sadly it does not work for me :(

closed 2021-11-14 by Xiashangning

1Revenger1, 2022-12-01:
Sorry to bring up an old issue, but I found a HID property that may allow multiple interfaces and allow the keyboard and touchpad to work. `kIOHIDMultipleInterfaceEnabledKey` changes the behavior in IOHIDInterface matching and allows multiple interfaces. Have you tried working with this property?