VoodooI2C/VoodooI2C #449: Touchpad needs aka force activation (sleep and wake) to work

Amin-0-1, 2021-03-25:
**Describe the bug**

My touchpad is working with gestures in interrupt gpio pinning with SSDT_TPDX (i2c1_TPD0), interrupt specifier is 0x33,
The problem is sometimes when startup the OS the touchpad doesn't work or doesn't move but it reads in the system preference touchpad tab hence that if I put the laptop into sleep for some seconds and wake it up the touchpad works as if it needs a force activation to move

**Did you read the common errors documentation?**
yes
**Did you read the troubleshooting documentation?**
yes
**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**

1. BigSur 11.2.3
2. VoodooI2C (2.6.5) & voodooI2CHID (1.0)
3. Processor : 9th Generation Intel®️ Core™️ i7 -9750H (12MB Cache, up to 4.5 GHz,6 Cores)

[I2C HID.zip](https://github.com/VoodooI2C/VoodooI2C/files/6207001/I2C.HID.zip)

ben9923, 2021-04-03:
Thanks for reporting :)
VoodooI2C kernel logs are missing due to some weird macOS stuff.
To get those, please inject `DebugEnhancer.kext` and attach the output of `dmesg` command. Preferably attach 3 sets of kernel logs+IOReg:
1. When it's working after boot.
2. When it's not working after boot.
3. When it's working after sleep/wake cycle (but not before it).

Does it have any relation to cold boot vs. reboot? Rebooting from macOS vs. Windows?
Also, just to confirm - When it's not working, FingerMgmt.app doesn't show any input too, right?

Amin-0-1, 2021-04-08:
@ben9923 
sorry for being late, and thanks for responding appreciate your efforts
here are the attached files with booting with DebugEnhancer.kext , FingerMgmt works with 5 fingers (img attached too)
, don't know what 3 sets of kernel that you've mentioned, about your questions it is not obvious at all , it sometimes doesn't work and that is not related to anything like if I reboot or normal boot, it just I boot and try to move touchpad I found it is not working or move for some cm and stuck, putting it into sleep and awaking it , seems to make it reactivate 
[I2C HID.zip](https://github.com/VoodooI2C/VoodooI2C/files/6279289/I2C.HID.zip)

ben9923, 2021-04-10:
@Amin-0-1 You didn't attach `dmesg` output for kernel logs. Running the `log` command produces no useful information.

Did you test FingerMgmt while trackpad is working? I asked you to check it while it isn't. Same for logs.

By 3 sets I meant collecting troubleshooting information (IOReg + dmesg) in each of the 3 situations I mentioned...

Amin-0-1, 2021-04-18:
@ben9923  can you please tell me how exactly to run dmesg ?

Amin-0-1, 2021-04-18:
just typing in terminal `sudo dmesg`

ben9923, 2021-05-01:
@Amin-0-1 Well, just run `dmesg` in the shell.

Amin-0-1, 2021-07-20:
@ben9923 
[debug.zip](https://github.com/VoodooI2C/VoodooI2C/files/6851007/debug.zip)
here are the files with ior+dmesg for every state 
1- touchpad works after booting
2- touchpad not working after booting
3- touchpad works after awaking from sleep 'put it to sleep when it is not working , reactivate it and make it works'

Amin-0-1, 2021-08-13:
@ben9923 did you figured anything in those files?

kprinssu, 2021-08-13:
@Amin-0-1 Please avoid cluttering this issue with unnecessary pings. Ben will get around to looking into it whenever he has the chance.

kprinssu, 2022-07-25:
@Amin-0-1 Please see this guide, https://dortania.github.io/Getting-Started-With-ACPI/Laptops/trackpad-methods/manual.html#checking-gpi0.  I will close this issue for now, please comment if it does not work.

closed 2022-07-25 by kprinssu