VoodooI2C/VoodooI2C #450: [VoodooI2C 2.6.5] MSFT0001 touchpad is not working

mysticfact, 2021-03-30:
**Describe the bug**
MSFT0001 touchpad is not working

**Did you read the common errors documentation?**
Yes.

**Did you read the troubleshooting documentation?**
Yes.

**Have you searched the issue on Github, Gitter, or Google?**
Yes.

**System Environment**
Please answer:
1. What is your operating system?
macOS 10.15.7
2. What version of the drivers (including drivers) are you using?
VoodooI2C 2.6.5
3. What is your hardware configuration?
Lenovo Yoga S730-13IML, i7-10510U, MSFT0001 trackpad (ACPI name: TPD0 under I2C0)

**Troubleshooting Archive**
[Send me YogaS730.zip](https://github.com/VoodooI2C/VoodooI2C/files/6229954/Send.me.YogaS730.zip)

**Additional context**
I already read this issue 
https://github.com/VoodooI2C/VoodooI2C/issues/388

So, I set GPHD to Zero and TPDM to Zero.
But, It is not worked.

This is GenI2C screenshot.
[image]

zilchulant, 2021-03-31:
Make sure your kexts are loaded in the right order:  VoodooInput, VoodooI2CServices, VoodooGPIO, VoodooI2C, VoodooI2CHID (also VoodooPS2Controller, VoodooPS2Controller-VoodooPS2Keyboard if you have a PS2 connected Keyboard). I would try `-vi2c-force-polling` as boot argument to validate if it is possible to get it working.

mysticfact, 2021-04-01:
> Make sure your kexts are loaded in the right order: VoodooInput, VoodooI2CServices, VoodooGPIO, VoodooI2C, VoodooI2CHID (also VoodooPS2Controller, VoodooPS2Controller-VoodooPS2Keyboard if you have a [...]

It's working by -vi2c-force-polling as boot argument with polling mode.
Why isn't it working with GPIO mode?
Have I any mistake?

ben9923, 2021-04-01:
Probably a user error, but we'll keep it open because it's similar to the one you mentioned.

Kernel logs provided are not full due to some macOS bug/some weird stuff with the logging system.
Please inject `DebugEnhancer.kext` and attach the output of the `dmesg` command. Without using polling mode, of course.

mysticfact, 2021-04-01:
> Probably a user error, but we'll keep it open because it's similar to the one you mentioned.
>
> Kernel logs provided are not full due to some macOS bug/some weird stuff with the logging system.
> Please [...]

This is my log without polling mode.
[log.zip](https://github.com/VoodooI2C/VoodooI2C/files/6242938/log.zip)

Thanks for your help.

ben9923, 2021-04-01:
@mysticfact It doesn't look like proper `dmesg` log :/
Missing the same logs, and has no time since startup for each log.

mysticfact, 2021-04-01:
> @mysticfact It doesn't look like proper `dmesg` log :/
> Missing the same logs, and has no time since startup for each log.

This is the command I ran.
"sudo dmesg > dmesg-log.log"
Is it wrong?

ben9923, 2021-04-01:
> > @mysticfact It doesn't look like proper `dmesg` log :/
> > Missing the same logs, and has no time since startup for each log.
> 
> This is the command I ran.
> "sudo dmesg > dmesg-log.log"
> Is it wrong?
> 
> This is new RunMe result file with DebugEnhancer.kext.
> [Send me YogaS730.zip](https://github.com/VoodooI2C/VoodooI2C/files/6243298/Send.me.YogaS730.zip)

It looks fine... Try rebooting and get another log, or add the `msgbuf=1048576` boot arg which should be a sufficient alternative to `DebugEnhancer`.

BTW, there's no reason to have the other VoodooI2C satellites disabled there, you won't use them. Just delete those altogether from config.plist and Kexts directory.

mysticfact, 2021-04-01:
@ben9923 
This is new RunMe result file with DebugEnhancer.kext and msgbuf=1048576 boot arg.
[Send me YogaS730.zip](https://github.com/VoodooI2C/VoodooI2C/files/6243476/Send.me.YogaS730.zip)

And dmesg log
[dmesg-log.txt](https://github.com/VoodooI2C/VoodooI2C/files/6243483/dmesg-log.txt)

ben9923, 2021-04-01:
@mysticfact Finally logs fully appear. I believe `msgbuf` has no effect with `DebugEnhancer.kext` injected.

Logs look good too. You can see `Device initiated reset accomplished` meaning an interrupt was fired successfully after reset.
Do you see any input in `FingerMgmt.app`?

If there's a BIOS update available, apply it.

And there's also a possibility your device just can't work in GPIO interrupts mode.
Please do cleanup your setup, those many unused ACPI files + kexts make it hard to troubleshoot.

mysticfact, 2021-04-01:
@ben9923 
Do you see any input in FingerMgmt.app?
-> There is no input or event at FingerMgmt.app.

If there's a BIOS update available, apply it.
-> It has latest version.

These are cleaned files.
[Send me YogaS730.zip](https://github.com/VoodooI2C/VoodooI2C/files/6244125/Send.me.YogaS730.zip)
[dmesg-log.txt](https://github.com/VoodooI2C/VoodooI2C/files/6244129/dmesg-log.txt)

mysticfact, 2021-04-01:
This is the file after it worked about 1 second.
[Send me YogaS730.zip](https://github.com/VoodooI2C/VoodooI2C/files/6244191/Send.me.YogaS730.zip)
[dmesg-log.txt](https://github.com/VoodooI2C/VoodooI2C/files/6244192/dmesg-log.txt)

ben9923, 2021-04-02:
@mysticfact Maybe the GPIO pin provided by ACPI is wrong. Manual pinning might work...
Undo that `TPDM = Zero` patch to get your APIC pin, then follow the [GPIO pinning guide](https://voodooi2c.github.io/#GPIO%20Pinning/GPIO%20Pinning) to find matching GPIO pin(s).

mysticfact, 2021-04-02:
@ben9923 
I think my laptop is root pinned and well-pinned.
[image]

So, I set GPHD to Zero and TPDM to Zero.
But, It is not worked.

I try GPIO pinning.
[image]

#define GPP_B3_IRQ				0x33
#define GPP_B3			28
CNL_GPP(1, 25, 50, 32),             /* GPP_B */
In this case, GPIO pin number is 0x23

#define GPP_F3_IRQ				0x33
#define GPP_F3			96
CNL_GPP(1, 93, 116, 128),           /* GPP_F */
In this case, GPIO pin number is 0x83

So, I try 0x23, 0x83, 0x17, 0x1B, 0x34, and 0x55.
But, It is not worked.

for all GPIO pin number, I2C is as below.
[image]

Have I any mistake?

rockguy40, 2021-05-25:
Same problem here with the same touchpad on a Lenovo yoga s940. Someone actually managed to fix for the s740 (same touchpad, very similar model)  at this [Github](https://github.com/frozenzero123/YOGA-S740/blob/master/README-EN.md). The GPi0 file, I tried to replicate what he did but failed miserably. Maybe someone more experienced could take a look.

ricardocosta43, 2021-05-30:
Acer Aspire 3 a315-56-569f (ice lake) not working to.

nurdiny13, 2021-06-16:
Lenovo ideapad gaming 3i not working, Os 10.15.7 same case

kprinssu, 2021-06-18:
@nurdiny13 Please use Gitter for support for general questions.

WY-WY-W, 2021-09-18:
same problem here with same touchpad on Lenovo s340 14iwl whiskey lake.

ben9923, 2021-09-18:
@WY-WY-W You'll need to attach a troubleshooting archive.
Like OP who set `TPDM = Zero`, you might also need to do something similar.

---

@mysticfact It appears that there were additional BIOS updates since I asked you about that. Could you try another update?
Setting `TPDM = Zero` is probably still necessary (Removal + manual pinning was only for a test).
BTW, sure you need to set `GPHD`?

Zero-zer0, 2022-03-17:
I have encountered with similar issue with my Lenovo Yoga S740-14IIL as well, and I could confirm that issue is related to firmware.

Yoga S740 is an IceLake-U laptop with 1035G1, and a touchpad with `_HID=MSFT0001` and `_SUB=SYNA0001`, and it works with polling mode with `VoodooI2CHID` and `RMII2C`. The default interrupt mode for the touch pad is `APIC` interrupt with `IRQ 48`, namely `0x30` for Windows and Linux. With some modification on BIOS settings, the touchpad could obtain the GPIO resource correctly and move but broke the lid sensor on the laptop.

The issue may related to `Intel Integrated Sensor Hub`, which handles some bus resources (e.g.. I2C, UART or SPI) and GPIO resources to ME, and turns everything related into a blackbox and user have no access from ACPI. It may also related to the debugging serial port setting as well.

In macOS, my attempt was to change return of `_CRS` into `SBFB,SBFG` for macOS and GPIO Pin was obtained from `_DSM` correctly, which is `0x30` as well. Also VoodooI2C logging shows `Hardware pin 0x18 is successfully assigned to GPIO pin 0x30` and there is no timeout for resetting the touchpad. The information in IOReg shows success as well. Also, with some tiny luck, I could move the pointer with a small distance and then the touchpad is dead. 

Such situation reminded me of some experience on playing with a laptop with touchpad and touchscreen, where both devices are forced to run at GPIO mode and the touchscreen usually dead after a small move. Someone suggest that the situation may related to "GPIO Resource Conflict", for the touchscreen actually works at APIC interrupt for windows while touchpad works at GPIO by default.

I have attempt to change a BIOS setting, `Force Unlock on All GPIO Pads` to make it enable, which is inspired by the configuration for Lenovo Legion Y530 (Y7000) [Link](https://github.com/chilledHamza/Hackintosh_Legion_Y530). Then in macOS my touchpad works successfully with GPIO interrupt. However, I could notice that some sensors on my laptop was broken, for example, the sensor for detecting lid open is not working anymore for macOS. After I closed the option in BIOS, things went back again while the lid sensor came back. By checking my BIOS firmware with `H2OUVE`, I found `I2C0` was configured with `ISH Configuration` for certain sensor though `I2C0` is an empty device in DSDT. However, voodooi2c could find and attach to this controller.

I have also read [Intel ® 495 Series Chipset Family OnPackage Platform Controller Hub (PCH) Datasheet](https://www.intel.com/content/www/us/en/products/docs/chipsets/495-series-chipset-on-package-pch-datasheet-vol-2.html) and found that some `GPPs` could be assigned to ISH and not presented as well. However the datasheet is full of registers and finding related BIOS settings is hard for a locked BIOS.

Here are just some quick information for what I have found and was written in a hurry. Please tell me if further information is need. Thanks!