VoodooI2C/VoodooI2C #453: Synaptics SYNA2393 touchpad may need some patch.

L-Ryland, 2021-04-21:
[According to this page](https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg2159960.html), Synaptics SYNA2393 touchpad may need some patch. 
My laptop use this model as well and is experiencing the same issue, it randomly stops working after suspend or some inputs, before that everything works fine. This issue has been staying long since I started to use this laptop since late 2019, and seems to have been fixed tested on my ArchLinux with the latest system update, but on my Hackintosh the problem remains. 
Can you have a look on this? @VoodooI2C

kprinssu, 2021-04-21:
Next time, **_please fill in the issue template_**.

We already have the quirk set up here:

https://github.com/VoodooI2C/VoodooI2CHID/blob/e3328f179c0ac2362fabd45dc42988d9540a9b9f/VoodooI2CHID/VoodooI2CHIDDevice.hpp#L31

In fact, the quirk should be applied to your trackpad as seen here https://github.com/VoodooI2C/VoodooI2CHID/blob/e3328f179c0ac2362fabd45dc42988d9540a9b9f/VoodooI2CHID/VoodooI2CHIDDevice.hpp#L86

It's set by checking the vendor and product id of the hardware, https://github.com/VoodooI2C/VoodooI2CHID/blob/e3328f179c0ac2362fabd45dc42988d9540a9b9f/VoodooI2CHID/VoodooI2CHIDDevice.hpp#L43

If those don't match, then you may need to add your product id to the above to apply the quirk (which you can easily do and PRs are welcome).

L-Ryland, 2021-04-22:
Here's my `logshow` trace of the last 10 minutes during which time my touchpad stops working and resumes after a close&open lid operation.  There seems to be  a bunch of VoodooI2C errors suddenly. 
[templog.log](https://github.com/VoodooI2C/VoodooI2C/files/6358148/templog.log)
[DSDT_Patch_with_configPlist.zip](https://github.com/VoodooI2C/VoodooI2C/files/6358167/DSDT_Patch_with_configPlist.zip)

kprinssu, 2021-04-22:
An IORegistry file would more useful than some random ACPI files (please use the runme app, it generates a zip file with everything you need to upload). I would also suggest you upload the kexts you are using.

I looked into those logs and it seems to be a communication failure between the kext and hardware. It seems like the hardware is failing to respond the data it's receiving. The logs are not particularly useful since I cannot determine if the quirks are being applied, that is why the IORegistry dump would be useful. You can do this yourself by verifying your trackpad's vendor and product ID match the one's here, https://github.com/VoodooI2C/VoodooI2CHID/blob/e3328f179c0ac2362fabd45dc42988d9540a9b9f/VoodooI2CHID/VoodooI2CHIDDevice.hpp#L43

L-Ryland, 2021-04-24:
From what I check in my IORegistry.app, vendor ID is 0x6cb and product ID says 0x7a13 which match [VoodooI2CHIDDevice_hpp](https://github.com/VoodooI2C/VoodooI2CHID/blob/e3328f179c0ac2362fabd45dc42988d9540a9b9f/VoodooI2CHID/VoodooI2CHIDDevice.hpp#L43) here. 
What else should I look into, or should I upload my IORegistry dump as well? 
[image]
[image]

kprinssu, 2021-05-05:
I would next check if you are using the latest available version of Voodooi2C. if that does not work, then add a `IOLog` or an IORegistry Entry to verify the quirk is being applied.

The appropriate line to do so is here:
https://github.com/VoodooI2C/VoodooI2CHID/blob/e3328f179c0ac2362fabd45dc42988d9540a9b9f/VoodooI2CHID/VoodooI2CHIDDevice.cpp#L403

kprinssu, 2022-02-27:
Please test with 2.7, if there is no response within 3 days, I will be closing this issue.

kprinssu, 2022-03-02:
I am closing this as this issue is now stale and I am considering it fixed within 2.7.

closed 2022-03-02 by kprinssu