VoodooI2C/VoodooI2C #455: QuietTimeAfterTyping never working because VoodooI2CMultitouchHIDEventDriver::message never get called

kingo132, 2021-05-07:
**System Environment**
1. operating system : Catalina 10.15.7
2. version of the drivers: 2.6.5
3. hardware configuration: Surface go2 with surface type cover

First I found QuietTimeAfterTyping not working with the trackpad of the type cover of surface go2.

Then I added some trace log, and found VoodooI2CMultitouchHIDEventDriver::message never get called. So key press time is always zero. This cause QuietTimeAfterTyping to have no effect.

[image]

[image]

And here is my log

[image]

And here is my ioreg info.

[image]

[myspgo.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/6441733/myspgo.ioreg.zip)

kingo132, 2021-05-07:
> This functionality only works if the keyboard is being set up by VoodooPS2 and your IORegistry file suggests that is the case.
> 
> This feature has not been very thoroughly tested of as of late and I advise you to test it on your own. Can you please check if the PS2 kext is emitting the events? I would start there and try to troubleshoot why VoodooI2C may not be receiving the messages.

Do you mean those kexts in acidanthera/VoodooPS2? I did not include them in EFI. And without them, the keyboard of type cover works fine. I need to enable them to check if they emitting the events right?

I just tested to enable them in EFI, but still no event going into VoodooI2CMultitouchHIDEventDriver::message. And also there are no VoodooPS2 kexts attached to any device in ioreg.

By the way, how do I check if VoodooPS2 is emitting the event? Which function should I start to add some trace log to ?

kprinssu, 2021-05-10:
If VoodooPS2 is not loading you most definitely don't have a device that is supported by them. That is,  `QuietTimeAfterTyping` only works alongside the VoodooPS2 and PS2 based keyboards. That would explain why the message is never sent to VoodooI2C as you most likely have a USB keyboard. In fact, I confirmed this by viewing your attached IORegistry file.

See:
[image]

In this case, this is not definitely not an issue with VoodooI2C.

closed 2021-05-10 by kprinssu

kingo132, 2021-05-15:
> If VoodooPS2 is not loading you most definitely don't have a device that is supported by them. That is, `QuietTimeAfterTyping` only works alongside the VoodooPS2 and PS2 based keyboards. That would explain why the message is never sent to VoodooI2C as you most likely have a USB keyboard.
> 
> In fact, I confirmed this by viewing your attached IORegistry file.
> See:
> [image]
> 
> In this case, this is not definitely not an issue with VoodooI2C. The functionality should be added a USB based driver, perhaps adding the functionality [VoodooI2C/VoodooI2CHID#52](https://github.com/VoodooI2C/VoodooI2CHID/pull/52) can be a solution.

Hello, thanks for your reply. I applied the usb keyboard patch and ported the key event emit code from VoodooPS2. Now ti totaly worked. Here my type cover keyboard in ioreg is like this

[image]

Thank you!

kprinssu, 2021-05-17:
No problem, I'm glad you were able to get it working.