VoodooI2C/VoodooI2C #456: Timeout or laggy input

tmbt78, 2021-05-09:
**Describe the bug**
Timeout waiting for device to complete host initiated reset OR laggy input on touchpad and perfectly working touchscreen

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes. Could be something similar to https://github.com/VoodooI2C/VoodooI2C/issues/319 AND/OR https://github.com/VoodooI2C/VoodooI2C/issues/435

**System Environment**
Please answer:
1. Catalina
2. Latest AND self compiled. VoodooI2C + VoodooI2CHID
3. Intel i5-10210 - 16GB RAM - 512GB SSD

I'm researching this issue for some days now and after reading a lot of the threads and forum and voodooi2c gitter chat logs i don't know what else do.
My laptop has a touchpad and a touchscreen. On I2C0 in ACPI table i see also a device called (NFC)
The problem is i always get Timeout waiting for device to complete host initiated reset on both TPDE AND TPL1 device if i use the latest voodooi2c from repository. I tried rebooting 20 times with always the same result.
After reading some opened and closed issues i self compiled VoodooI2C increasing timeout as saw here https://github.com/VoodooI2C/VoodooI2C/issues/319 with even higher values.
After that i have a 25% chance of timeout as before but when this is not the case i got a fully working touchscreen (gesture included) and a really laggy but working touchpad at the point to be useless (the arrow moves really slow BUT left click and right click are recognized immediately).

I followed istruction on issues 319 and modded my files accordingly but until i increased timeout i wasn't able to make things improve.

Attached you will find ioreg & kernel log for both the situations.

[Timeout.zip](https://github.com/VoodooI2C/VoodooI2C/files/6447569/Timeout.zip)
[NO_Timeout.zip](https://github.com/VoodooI2C/VoodooI2C/files/6447570/NO_Timeout.zip)

Goshin, 2021-06-04:
Can you get your touchpad working well in GPIO mode with your touchscreen disabled or in polling mode? If so it could be a problem in VoodooGPIO. You can try out the commit I mentioned in https://github.com/VoodooI2C/VoodooI2C/issues/435#issuecomment-785810972 with the latest VoodooI2C&GPIO.

tmbt78, 2021-06-04:
@Goshin I've tried that already. If i disable the touchscreen doesn't change anything. If i try to enforce polling mode to the touchscreen it seems succesfull according to the logs but the touchscreen doesn't work at all. 
So the only solution i've right now is to enforce polling mode on the touchpad and then i can use interrupt mode on the touchscreen o disable it doesn't matter.
My problem is that i use the laptop mostly on battery and i when in polling mode the laptop consume 1.4W while on interrupt i'm at around 0.8W .. so it's a huge difference for me.
Is there something else i can try ?

tmbt78, 2021-06-11:
I've finally solved this issue !
The problem for the laggy pointer was here: Name (SBFG, ResourceTemplate ()
            {
                GpioInt (**Edge**, ActiveLow, ExclusiveAndWake, PullDefault, 0x0000,
                    "\\_SB.PCI0.GPI0", 0x00, ResourceConsumer, ,
                    )
                    {   // Pin list
                        0x0000
                    }
            })
I hotpatched the SBFG method changing **Edge to Level** and now the cursor isn't laggy anymore !
@Goshin @kprinssu Not sure if there is something to check on VoodooI2C side or if it is simply a problem with my DSDT but thanks again for your time

Mattia

Goshin, 2021-06-11:
Congrats. It should be level interrupt according to Microsoft's I2C HID spec.

kprinssu, 2021-06-11:
Thanks for update @tmbt78. Glad you got it working.

closed 2021-06-11 by kprinssu