VoodooI2C/VoodooI2C #458: VoodooPS2, VoodooI2C and VoodooI2CHID race condition?

xMaximillianous, 2021-06-23:
**Describe the bug**

Okay so I've got VoodooPS2Controller, VoodooI2C and VoodooI2CHID installed on my laptop. Trackpad (HID) works fine, however the keyboard doesn't always work on startup and I have to force it by going into sleep mode then waking it up (works every time with that method). The laptop also has touchscreen (which sometimes loads and works but thats about 5% chance of it working), also if the touchscreen works and I go into sleep mode, it stops working. Any ideas on how to fix that? I've been told that it could be a race condition.

PS: It's very rare that all 3 of these devices work at the same time, but it happens every now and then (out of the blue).
ALSO: VoodooI2C makes the touchscreen work and trackpad doesn't
and VoodooI2CHID makes trackpad work and touchscreen doesn't

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
[image]

Yes

**System Environment**
Please answer:
1. What is your operating system? Big Sur
2. What version of the drivers (including drivers) are you using? Latest 2.6.5
3. What is your hardware configuration? 
System Specs:
Laptop: Acer Swift 7 SF714-52T
CPU: Intel i7-8500Y (Amber Lake) - with iGPU (UHD 615)
RAM: 16GB LPDDR3 - Soldered into MOBO
SSD: 512GB - Soldered into MOBO

kprinssu, 2021-06-24:
This is mostly not a race condition. Please upload more than just a screenshot such as a dump of your IO Registry and the kext a you are using.

For starters, I would suggest you remove VoodooInput from the PS2 kext.

xMaximillianous, 2021-06-24:
Okay I've uploaded both the EFI and IOReg dump here: https://drive.google.com/drive/folders/1eP9v_sApIWaXrek74Z4XXdtbpk2f_Nez?usp=sharing

I use the 3 main kexts: VoodooPS2Controller, VoodooI2C and VoodooI2CHID; and also their subkexts such as VoodooPS2Keyboard, VoodooPS2Mouse, VoodooPS2Trackpad, VoodooGPIO, VoodooI2CServices and VoodooInput (the one in VoodooI2C).
I've also removed VoodooInput from the PS2 kext, and still nothing changed. :/

kprinssu, 2021-06-28:
Next time please upload any archive to Github. You can simply drag and drop files into the editor and Github will handle the markup and upload.

From what I can see, the I2C kexts are loading fine. Can you please clarify if the touchpad and touchscreen work consistently? Is it just the PS2 keyboard that is an issue?

xMaximillianous, 2021-06-28:
Okay I apologize for that.

The touchpad always works, however the touchscreen doesn't work most of the time (works 1 (once) in 30 boots). While the PS2 Keyboard has ~25% chance of working on boot, however even if it doesn't I can just force it by going into sleep and then waking up the laptop (this method works everytime).

ben9923, 2021-11-13:
Can you attach an IOReg without VoodooI2CHID (Which means touchscreen is working in this case)?

Does removing/disabling the PS2 mouse/trackpad drivers help? You don't need those anyway.

kprinssu, 2022-03-02:
Please try with 2.7, otherwise I will be closing this issue in 3 days.

kprinssu, 2022-03-05:
Please re-open if this is still an issue with 2.7.

closed 2022-03-05 by kprinssu