VoodooI2C/VoodooI2C #460: Dell ALPS I2C: Non-working physical buttons and trackpoint

lulujyc, 2021-06-28:
**Describe the bug**
A clear and concise description of what the bug is.
Dell ALPS I2C Dual-Pointing is not fully supported; Physical buttons and TrackPoint (buttons) don't work well. On I2C Dual-Pointing integrated Dell Precision and Latitude platforms, only the TouchPad and physical TouchPad Left button work. TouchPad Middle, Right buttons, all three TrackPoint buttons, and TrackPoint don't work.

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
Yes

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading .
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
Yes -- I'm using a Precision 7530, but I've found Precision 7540 users, as well as Latitude users, experiencing this problem. It might not be necessarily a bug -- rather a not fully unsupported I2C device.

**System Environment**
Please answer:
1. What is your operating system?  Big Sur
2. What version of the drivers (including drivers) are you using? Latest to date.
3. What is your hardware configuration? Precision 7530: Coffee Lake-H CPU

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting
[Uploading Archive.zip…]()
) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

**Additional context**
Add any other context about the problem here.

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/6728467/Archive.zip)

kprinssu, 2021-06-28:
Physical buttons nor are track points supported in VoodooI2C. There are open PRs to support it (https://github.com/VoodooI2C/VoodooI2C/pull/445) but nothing has been merged in yet.

lulujyc, 2021-06-28:
> Physical buttons nor are track points supported in VoodooI2C. There are open PRs to support it (#445) but nothing has been merged in yet.

Thank you! Is there a build of the PR branch available so that I can test?

kprinssu, 2021-06-28:
There aren't any official dev or test builds. Not will support be given if you do get your hands on one.

I suggest you do a build yourself.

lulujyc, 2021-06-29:
Ok, thanks :)

Lorys89, 2021-10-08:
> Ok, thanks :)

try this https://github.com/juico/AlpsT4USB/releases/tag/1.5.2

lulu-gh, 2021-10-10:
> > Ok, thanks :)
> 
> try this https://github.com/juico/AlpsT4USB/releases/tag/1.5.2

Ah thanks! I am currently not available to test, but I wonder if you happen to achieve a working trackpoint as well? Thanks for your huge contribution again!