VoodooI2C/VoodooI2C #463: Could not get ACPI device for PCI provider

lulujyc, 2021-07-11:
**Describe the bug**
The trackpad stopped working after I added an extra M.2 drive, but it still didn't work when I removed the drive. It works in Windows always nevertheless. When in verbose boot, there are two lines, one is VoodooI2CPCIController::pci8086,a369 Could not get ACPI device for PCI provider, and the other one is the same but just for 8086,a368. The machine has two I2C controllers, one is I2C0 which is for the touchscreen optional assembly that I don't have, and I2C1 is for the dual pointing device. Many thanks!

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
yes

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading .
yes

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
yes

**System Environment**
Please answer:
1. What is your operating system? Big Sur
2. What version of the drivers (including drivers) are you using? Latest
3. What is your hardware configuration? CFL-H

**Troubleshooting Archive**

[mac.zip](https://github.com/VoodooI2C/VoodooI2C/files/6797366/mac.zip)
[DSDT](https://github.com/lulujyc/Dell-Precision-E7530-Prototype-OpenCore/blob/main/EFI/OC/ACPI/DSDT.aml)

**Additional context**
Add any other context about the problem here.

kprinssu, 2021-07-12:
This really looks like a specific issue with your ACPI as from what I can from your IO Registry dump, VoodooI2C is not finding any I2C controllers.  As I do not have your EFI here, I can only speculate as to what the problem is, please upload a copy of your EFI folder.

Additionally, how are you patching in the GPIO pin?

lulujyc, 2021-07-12:
> This really looks like a specific issue with your ACPI as from what I can from your IO Registry dump, VoodooI2C is not finding any I2C controllers. As I do not have your EFI here, I can only [...]

Hi,

Thank you so much for your reply! My EFI folder can be downloaded from [here](https://github.com/lulujyc/Dell-Precision-E7530-Prototype-OpenCore/archive/refs/heads/main.zip).

I've patched it via 0x1b, which i believe should be the correct calculated value. Both GPIO and polling mode worked very well before I installed that cursed M.2 drive into my system. Thank you again!

kprinssu, 2021-07-12:
That would be your problem. By adding a new device, your ACPI device paths most likely changed. That's why you should  hotpatch. For now, you will need to take a new dump of your DSDT and re-patch it (I highly advise against this).

As this is a configuration issue, please use Gitter for further support.

closed 2021-07-12 by kprinssu

lulujyc, 2021-07-12:
THANK YOU!!!!!!

You're absolutely correct. The memory address in DSDT changed after I installed the drive. After I compared the changes and modified the patched DSDT, everything backs to work. Thank you sooooo much and fixing this -- and finally realizing the cause of this -- definitely made my day. Can't appreciate it more!!!

kprinssu, 2021-07-12:
No worries @lulujyc, enjoy your hack :)