VoodooI2C/VoodooI2C #464: (VoodooI2CELAN) <compose failure [UUID]> error floods the kernel logs upon each boot

mackonsti, 2021-07-18:
**Describe the bug**
On my Lenovo laptop, with OpenCore 0.7.1 and recent VoodooI2C v2.6.5 and plugins, I get this error flooding kernel logs despite using all RELEASE versions and nowhere any "debugging" being enabled:

`2021-07-18 11:15:37.10 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>`

**Did you read the common errors documentation?**
Yes, this is not mentioned.

**Did you read the troubleshooting documentation?**
Yes.

**Have you searched the issue on Github, Gitter, or Google?**
Yes. I was prompted from the chat to read the ticket https://github.com/acidanthera/bugtracker/issues/1148 but it was rather irrelevant as I have no DEBUG versions installed.

Note: There are no other I2C devices attached that may interfere with the kext; according to the IORegistry Explorer only I2C0 is present, the laptop's firmware is not providing other I2C devices (despite present in DSDT).

**System Environment**
Please answer:
1. What is your operating system? MacOS Big Sur 11.3.1
2. What version of the drivers (including drivers) are you using? Latest VoodooI2C 2.6.5 incl. plugins
3. What is your hardware configuration? Lenovo ELAN 0629 touchpad
```
% kextstat -a | grep Voodoo
Executing: /usr/bin/kmutil showloaded --arch-info
No variant specified, falling back to release
   54    1 0                  0xb000     0xb000     x86_64             com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   66    1 0                  0x27000    0x27000    x86_64             org.coolstar.VoodooGPIO (1.1) 7ED39A4D-2EFD-35AE-9BC6-6215C17D93F1 <13 6 5 3>
   72    1 0                  0x3b000    0x3b000    x86_64             com.alexandred.VoodooI2C (2.6.5) 748C2CC2-07FD-3FD5-8E86-874C73317356 <66 61 54 14 13 6 5 3 1>
   83    0 0                  0xc000     0xc000     x86_64             me.kishorprins.VoodooI2CELAN (1) D3EE7E6D-F3FB-3A39-BD36-7C4BCF972E9B <72 61 13 6 5 3>
   90    0 0                  0x19000    0x19000    x86_64             me.kishorprins.VoodooInput (1.1.1) 2DBFF028-3A8F-37C9-A5D7-B435F5AA516F <61 6 5 3>
```

**Troubleshooting Archive**
Here is a quick visual sample of the log:
```
2021-07-18 21:13:42.98 kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f7 start
2021-07-18 21:13:42.98 kernel[0]: (IOHIDFamily) VoodooInputActuatorDevice:0x1000002fa start
2021-07-18 21:13:42.98 kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f7 creating interfaces
2021-07-18 21:13:42.98 kernel[0]: (IOHIDFamily) VoodooInputActuatorDevice:0x1000002fa creating interfaces
2021-07-18 21:13:42.99 kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f7 open by AppleMultitouchTrackpadHIDEventDriver 0x100000302 (0x0)
2021-07-18 21:13:42.99 kernel[0]: (IOHIDFamily) VoodooInputActuatorDevice:0x1000002fa open by AppleActuatorHIDEventDriver 0x100000305 (0x0)
2021-07-18 21:13:58.45 kernel[0]: (VoodooI2CELAN) <private>
2021-07-18 21:14:00.05 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.05 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.05 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.07 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.07 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.07 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.07 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.07 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.08 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.08 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:00.08 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:05.87 kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f7 new user client
2021-07-18 21:14:05.87 kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f7 open by IOHIDLibUserClient 0x100000515 (0x0)
2021-07-18 21:14:05.87 kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f7 close by IOHIDLibUserClient 0x100000515 (0x0)
2021-07-18 21:14:06.37 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:06.37 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:06.37 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:06.37 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2021-07-18 21:14:06.37 kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
[...]
```
Here are the files requested:
* IORegistry
* The log itself from `log show --style syslog -debug --info --last boot --predicate 'process == "kernel"'`
* Injected SSDT for I2C device (only) & extracted DSDT (untouched)

[Compose.log](https://github.com/VoodooI2C/VoodooI2C/files/6836834/Compose.log)
[Lenovo.IORegistry.zip](https://github.com/VoodooI2C/VoodooI2C/files/6836837/Lenovo.IORegistry.zip)
[DSDT+SSDT.zip](https://github.com/VoodooI2C/VoodooI2C/files/6836840/DSDT%2BSSDT.zip)

**Additional context**
Was just checking boot for errors after successfully making my ELAN0629 run with polling mode, due to the sequence in which OpenCore config.plist includes the VoodooI2C kexts. Namely, the following order is found:
1. VoodooGPIO.kext
2. VoodooI2CServices.kext
3. VoodooInput.kext
4. VoodooI2C.kext
5. VoodooI2CELAN.kext

Thank you in advance.

kprinssu, 2021-08-29:
> 相同的问题,不知道怎么办

Please only rely on English.

778280151, 2021-09-03:
I have the same problem, how to solve it

ben9923, 2021-11-13:
@mackonsti Not sure I found something useful in that article :(
Thanks for putting the effort, you (or someone else) might find the cause eventually...

Anyway it's more of a cosmetic issue with just a couple of extra logs, so it's not that bad.

kprinssu, 2022-02-27:
@mackonsti Can you please test with 2.7?

mackonsti, 2022-03-01:
Hello @kprinssu thank you for taking time to look into this and updating the code, here are my results.

* MacOS **Big Sur 11.6.4** (latest)
* Latest **VoodooI2C 2.7.0** incl. plugins (RELEASE version)
* Lenovo **ELAN 0629** touchpad
* Parameter `force-polling` still used in **OpenCore 0.7.8** configuration (injected)
* Boot arguments do NOT include `msgbuf=1048576`

```
% kextstat -a | grep Voodoo
Executing: /usr/bin/kmutil showloaded --arch-info
No variant specified, falling back to release
   54    1 0                  0xb000     0xb000     x86_64             com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   66    1 0                  0x27000    0x27000    x86_64             org.coolstar.VoodooGPIO (1.1) 7ED39A4D-2EFD-35AE-9BC6-6215C17D93F1 <13 6 5 3>
   75    1 0                  0x3b000    0x3b000    x86_64             com.alexandred.VoodooI2C (2.7) F536F1B4-E44D-3A19-86B2-8B0DF4F2C3C5 <66 61 54 14 13 6 5 3 1>
   83    0 0                  0xc000     0xc000     x86_64             me.kishorprins.VoodooI2CELAN (1) D3EE7E6D-F3FB-3A39-BD36-7C4BCF972E9B <75 61 13 6 5 3>
   90    0 0                  0x19000    0x19000    x86_64             me.kishorprins.VoodooInput (1.1.3) F3F0AEF0-FA53-3E76-823F-C9151C2B4A2F <61 6 5 3>
```
Please note, ELAN is reporting v1 without other minor or build numbers, perhaps consider changing to 1.0.0 or correct it?

Next, the logs show this:

```
% log show --style syslog -debug --info --last boot --predicate 'process == "kernel"' | grep -I voodoo

2022-03-01 21:17:52.548048+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:17:52.741986+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:17:52.741994+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:17:52.741996+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:17:52.742040+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:17:52.742071+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:17:53.035870+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 start
2022-03-01 21:17:53.038335+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputActuatorDevice:0x1000002fb start
2022-03-01 21:17:53.041121+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 creating interfaces
2022-03-01 21:17:53.041186+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputActuatorDevice:0x1000002fb creating interfaces
2022-03-01 21:17:53.053050+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 open by AppleMultitouchTrackpadHIDEventDriver 0x100000316 (0x0)
2022-03-01 21:17:53.053841+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputActuatorDevice:0x1000002fb open by AppleActuatorHIDEventDriver 0x10000031c (0x0)
2022-03-01 21:18:03.339031+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:10.108584+0100  localhost kernel[0]: (VoodooI2CELAN) <private>
2022-03-01 21:18:12.324923+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.325099+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.325215+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.340307+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
[...]
2022-03-01 21:18:12.388003+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.388007+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.398673+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.399104+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.399208+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.399214+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:12.399284+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
[...]
2022-03-01 21:18:17.525602+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:18.764956+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 new user client
2022-03-01 21:18:18.766088+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 open by IOHIDLibUserClient 0x10000050d (0x0)
2022-03-01 21:18:18.766178+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 close by IOHIDLibUserClient 0x10000050d (0x0)
2022-03-01 21:18:19.264641+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:19.264662+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:19.269446+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:19.269462+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
[...]
2022-03-01 21:18:27.678638+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:27.934545+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:27.965004+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:18:31.239312+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
[...]
[... cut ...]
2022-03-01 21:24:04.143191+0100  localhost kernel[0]: (IOHIDFamily) VoodooInputSimulatorDevice:0x1000002f9 close by IOHIDLibUserClient 0x100000677 (0x0)
2022-03-01 21:24:13.242387+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:24:13.242392+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:24:24.722214+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
2022-03-01 21:24:24.722226+0100  localhost kernel[0]: (VoodooI2CELAN) <compose failure [UUID]>
```

Sorry 😢 Still flooding...

kprinssu, 2022-03-01:
Darn, I am not sure what could be causing this.

mackonsti, 2022-03-01:
If you later think of any other parameter for OpenCore, anything that you believe could be quickly tested as parameter, let me know @kprinssu. Cheers mate.

mackonsti, 2022-03-02:
Hi @kprinssu please kindly help me out here, I am not sure what I should do.

@vit9696 wrote in that message:

> Is Lilu debugging enabled when this happens? I do not have this with debugging disabled and believe the reason for this is oslog log parsing attempting to resolve binary UUID to a file ondisk. Since the file is not t[here](https://github.com/acidanthera/OpenCorePkg/blob/0.6.1/Platform/OpenCore/OpenCoreKernel.c#L608-L612), it shows <compose failure [UUID]>. The issue is caused by LoggingSupport.framework. All in all it is pretty much normal.
> 
> If you feel like experimenting, you could try changing /Library/Extensions prefix here and rebuilding OpenCore. Use something obviously weird like /Volumes/Extensions or maybe the real path (/Volumes/EFI/EFI/OC/Kexts).

For starters, I am **not** using Lilu or any kext in debug version, mode or build. Only RELEASE ones. 😞

Secondly, I am not really in a position or knowledge to build Lilu nor do I clearly understand this change of path.

What do you have in mind? A problem with Lilu's debugging mechanism still being used in the RELEASE version?
Could there be a parameter to use as a `boot-arg` instead?

This <compose failure UUID> could be due to Lilu eventually... but is VoodooI2C using Lilu as a dependence? I mean, I can try booting Big Sur with FakeSMC and VoodooI2C only, does it make sense? **Thanks**

kprinssu, 2022-03-02:
I really do not know what the issue could be. It looks to be some issue with the macOS and something else. There are numerous other reports of other Hackintosh related kexts generating the same warnings. 

I do not think it's a VoodooI2C, rather something else is causing it. I do not have any recommendations nor any suggestions to try out at this time.