VoodooI2C/VoodooI2C #467: Multitouch gesture not working on the touchscreen with the stylus pen.

astro-lee-k, 2021-08-06:
**Describe the bug**
Multitouch gesture not working on the touchscreen with the stylus pen on the secondary screen in ASUS ZenBook Pro Duo 15 (has the main 4k and secondary touchscreens. Only the cursor and click functions work. (multi touch, scroll, and hold to drag/right click functions are not available.) The secondary touchscreen comes with stylus pen and support multi touch in Windows 10 and Ubuntu). Had tried disconnected the main touchscreen (unplug the eDP connector from MB), and had only the secondary screen in macOS 10.15.7. However, the issue reamins the same. (only the cursor and click functions work.)

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**
Please answer:
1. What is your operating system?  macOS 10.15.7
2. What version of the drivers (including drivers) are you using? Lasest (2.6.5)
3. What is your hardware configuration? (ASUS ZenBook Pro Duo 15, ELAN device id: 9008 and 9008)

kprinssu, 2021-08-06:
Please use the VoodooI2C Dev channel for support. GitHub is not the right place for support with the project.

In general, VoodooI2C does support stylus but implementation of has been spotty as macOS *does not* support natively.

closed 2021-08-06 by kprinssu

shiecldk, 2021-08-06:
@kprinssu I have asked help on the Dev channel on Gitter already  many time but on one was really answering and help there, so could you keep this topic opening? I don't see the reason why you close this topic; this problem has presented in other laptops too many time and I want to resolve this problem as it was never really been resolved before.

kprinssu, 2021-08-06:
The dev channel is https://gitter.im/alexandred/VoodooI2CDev

Please ask on there for further help with the code.