VoodooI2C/VoodooI2C #469: VoodooI2CHID: VoodooI2CKeyboardHIDEventDriver attaches to PS2K & PS2M and make keyboard unresponsive

stevezhengshiqi, 2021-08-22:
Hi VoodooI2C Team, thank you all for the great work all these years.

My machine: i5-8250U with ETD2303 and PS2 Keyboard
I used VoodooI2C + VoodooI2CHID to drive trackpad and VoodooPS2Keyboard to drive keyboard

By testing https://github.com/VoodooI2C/VoodooI2CHID/pull/54, my keyboard becomes unresponsive. Digging the ioreg, I found that `VoodooI2CKeyboardHIDEventDriver` accidentally attached to all my PS2 devices, which probably broke down my keyboard. In the ioreg, I can see `VoodooI2CServices Supported` property is injected, corresponds to [VoodooI2CHID/VoodooI2CKeyboardHIDEventDriver.cpp#L177](https://github.com/VoodooI2C/VoodooI2CHID/blob/f0c868db0742f0040b4d1d7f951d37b5b6be9a79/VoodooI2CHID/VoodooI2CKeyboardHIDEventDriver.cpp#L177), so I guess the keyboard event driver might think my PS2 device as a USB and attaches on it(?)

Here's HID related log, ioreg when keyboard is unresponsive, and ioreg when keyboard is normal. ioregs are extracted from IORegistryExplorer v3.0.2. Please have a check, @1Revenger1 @kprinssu
[20210821VI2C_log1.txt](https://github.com/VoodooI2C/VoodooI2C/files/7026531/20210821VI2C_log1.txt)
[VI2CKeyboardErr.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/7026532/VI2CKeyboardErr.ioreg.zip)
[NormalKeyBoard.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/7026533/NormalKeyBoard.ioreg.zip)

1Revenger1, 2021-08-22:
lol

Sorry about that. Probably can do something like below to check that we are a product of VoodooI2C. Or, double check that we have the right usage page/usage set. Though, why doesn't the default macOS driver work then for the surface devices? Hrmm...

https://github.com/apple-opensource/IOHIDFamily/blob/315.7.16/IOHIDFamily/IOHIDEventDriver.cpp#L138

I'll reply back with a version that at least checks the provider, to make sure that works. I'd be happier to hear of better solutions, I don't have an ioreg on hand for a surface device that needs the keyboard event driver.

1Revenger1, 2021-08-22:
[VoodooI2CHID.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/7026617/VoodooI2CHID.kext.zip)
Here is the new VoodooI2C, with the diff below:
```diff
diff --git a/VoodooI2CHID/VoodooI2CKeyboardHIDEventDriver.cpp b/VoodooI2CHID/VoodooI2CKeyboardHIDEventDriver.cpp
index fb1f295..dc85b3f 100644
--- a/VoodooI2CHID/VoodooI2CKeyboardHIDEventDriver.cpp
+++ b/VoodooI2CHID/VoodooI2CKeyboardHIDEventDriver.cpp
@@ -7,6 +7,7 @@
 //
 
 #include "VoodooI2CKeyboardHIDEventDriver.hpp"
+#include "VoodooI2CHIDDevice.hpp"
 #include <IOKit/IOCommandGate.h>
 #include <IOKit/hid/IOHIDInterface.h>
 #include <IOKit/usb/USBSpec.h>
@@ -138,6 +139,10 @@ void VoodooI2CKeyboardHIDEventDriver::handleInterruptReport(AbsoluteTime timesta
 }
 
 bool VoodooI2CKeyboardHIDEventDriver::start(IOService* provider) {
+    IOService *parent = provider->getProvider();
+    if (OSDynamicCast(VoodooI2CHIDDevice, parent) == nullptr)
+        return false;
+    
     if (!super::start(provider))
         return false;
```

stevezhengshiqi, 2021-08-22:
@1Revenger1 Thanks for the quick fix! It works like a charm. Good job.

kprinssu, 2021-08-23:
@1Revenger1 Good job with the quick fix, can you perhaps create a quick PR for the patch above?

stevezhengshiqi, 2021-09-03:
If @1Revenger1 does not reply before Monday, would you mind creating a pr? @kprinssu

kprinssu, 2021-09-07:
Created a quick PR for the fix, https://github.com/VoodooI2C/VoodooI2CHID/pull/55

@stevezhengshiqi Can you please test the above and let me know if it does indeed fix your issue?

kprinssu, 2021-09-07:
Committed in via https://github.com/VoodooI2C/VoodooI2CHID/commit/8182b1af40017645dd845935a912ecc465ee5057

ben9923, 2021-09-07:
@kprinssu Wouldn't a better alternative be using an `IOPropertyMatch` with `Transport=I2C` ([Like this](https://github.com/VoodooI2C/VoodooI2CHID/blob/f0c868db0742f0040b4d1d7f951d37b5b6be9a79/VoodooI2CHID/Info.plist#L316)).

Not 100% sure about it, but I think it should work too, in a cleaner way.

ben9923, 2021-09-07:
@1Revenger1 Uhm, if it's USB then the parent device wouldn't be a `VoodooI2CHIDDevice` lol
Did this change break keyboard for the surface? :/

I do wonder what happens with other USB keyboards connected.
I also still wonder why the built-in keyboard driver isn't enough. Did it work with VoodooI2CHID not present?

1Revenger1, 2021-09-07:
> @1Revenger1 Uhm, if it's USB then the parent device wouldn't be a `VoodooI2CHIDDevice` lol
> Did this change break keyboard for the surface? :/

Oh, VoodooI2CHIDDevice doesn't attach to those?
That probably would break keyboard than for the surface.

> I do wonder what happens with other USB keyboards connected.
> I also still wonder why the built-in keyboard driver isn't enough. Did it work with VoodooI2CHID not present?

Hrmm, it might work to just add a match category, that might be best option. No clue if that'd actually work, or if more is needed. Never thought about that until now.

ben9923, 2021-09-07:
> Oh, VoodooI2CHIDDevice doesn't attach to those?
> That probably would break keyboard then for the surface.

Haha it doesn't attach. For USB trackpads VoodooI2CHID's trackpad event driver attaches directly, the USB/HID stack handles everything else natively. The kext just parses the reports like it does for I2C devices.

> Hrmm, it might work to just add a match category, that might be best option. No clue if that'd actually work, or if more is needed. Never thought about that until now.
> 
> Edit: You'd think though that people would've stumbled on that by now though, hrmm. Be great if someone could test.

We definitely need a tester. Do you know anyone with the right hardware?

@kprinssu I think we should sort that keyboard mess prior to the new planned release :(

Xiashangning, 2021-09-10:
@ben9923 @kprinssu Well, let me make it clear. The situation for surface is like this: the keyboard and the trackpad are integrated together connected to surface by one usb connector. When i do not use voodooi2c kext, the keyboard works however macos cannot recognize my builtin trackpad. When I use voodooi2c kext, macos can recognize the trackpad however I lose the keyboard. It thinks that this device is merely a trackpad.(maybe in apple’s ecosystem there is not something like this) So to solve my problem, i created a driver that both serves as a keyboard and trackpad driver.
BTW, this change does break keyboard for surface. VoodooI2CHID's trackpad event driver attaches directly under apple’s usb driver

ben9923, 2021-09-10:
@Xiashangning Ideally we would drop the keyboard changes altogether and simply let the native keyboard driver work.

Please check out the discussion in #447 with a suggestion for resolution we would like you to try :)

ben9923, 2021-11-13:
To be fixed by VoodooI2C/VoodooI2CHID#56

kprinssu, 2022-02-27:
This has been fixed with 2.7, please re-open if that is not the case.

closed 2022-02-27 by kprinssu