VoodooI2C/VoodooI2C #476: (Request) Rebuild version 2.4.4

thefiredragon, 2021-09-14:
Dear all,
I have a little request, could it possible to rebuild voodooi2c 2.4.4 with never sdk?
This is the only version which works well on my dell xps 9500.
I tried to rebuild itself, but dependencies on the other repos like VoodooI2CHID has not tags,
 
best Regards,
Dan

ben9923, 2021-09-14:
Hey

Is any version in the 2.5+ range not working well for you?
Can you elaborate about the problem you're having? Attach troubleshooting files for both versions?
We'd prefer all users have the latest version as maintaining old releases is hard and inefficient.

Why would you want to build it with a newer SDK anyway?
Simply checking out the `2.4.4` tag and updating submodules per [build instructions](https://voodooi2c.github.io/#Build%20Environment/Cloning%20VoodooI2C) should be enough.

thefiredragon, 2021-09-14:
Thank you,
did not know excatly what changes after 2.4.4 was happen which break polling mode for 2 devices.

https://github.com/VoodooI2C/VoodooI2C/issues/396
was one thread which sounds like the same.

Till yet 2.4.4 works great without any probles. Touch and Touchpad works fine also under monterey. 
I'm at work later I can provide more information about my patches.

ben9923, 2021-09-15:
@thefiredragon My best suggestion would be trying to make GPIO interrupts work, please do attach more info.
They provide smoother input and less CPU usage.

I assume the problem you have iscaused by the new feature of getting GPIO interrupts from an alternative source (`_DSM`).
You may try the latest version with the `-vi2c-force-polling` boot-arg, or by adding the `force-polling` property to the relevant I2C controller.
It's detailed in the issue you linked.

As it's working under Monterey, why do you want to build it with a newer SDK?

LeeBinder, 2022-02-27:
always astonishing how many people die during an issue.. must be bad karma.

kprinssu, 2022-02-27:
@thefiredragon I will be closing this issue in 3 days, please reply if 2.7 does not work.

kprinssu, 2022-03-02:
I am closing this as this issue is now stale and I am considering it fixed within 2.7.

closed 2022-03-02 by kprinssu