VoodooI2C/VoodooI2C #478: Tiger Lake on HP Probook 450 G8 trackpad not work

gianlucainnocenti, 2021-09-15:
**Describe the bug**
I tray to use voodool2c on my laptop with support of Avery Black @1Revenger1 but my trackpad want work

I use Big Sur 11.5 latest version of voodool2c 
I attach dimes and screenshot....
[image]
[dmesg_now.zip](https://github.com/VoodooI2C/VoodooI2C/files/7168171/dmesg_now.zip)

gianlucainnocenti, 2021-09-15:
Thank you in Advance..

kprinssu, 2021-09-15:
Please use Gitter for support.

m0d16l14n1, 2021-09-15:
>  Please actually read the Github issue template and fill it in. @ben9923 mentions that Tiger Lake is not supported by VoodooI2C.
> 
>  
> 
>  Simply uploading logs and a screenshot is not enough to troubleshoot.

I am not sure that you really *need* to support Tiger Lake until it will get graphics support on macOS (acceleration). It's not supported at all by now

gianlucainnocenti, 2021-09-15:
I am been using my laptop with 11th Gen intel

ben9923, 2021-09-15:
@m0d16l14n1 I believe it will never be supported :/
Maybe for those with eGPU/built-in AMD graphics?

@gianlucainnocenti Your IOReg shows no graphics hardware acceleration for iGPU.

gianlucainnocenti, 2021-09-15:
Yes I know, infact I have only 7mb but this issue will solve:)

m0d16l14n1, 2021-09-15:
> @m0d16l14n1 I believe such graphics will never be supported :/
> 
> Maybe for those with eGPU/built-in AMD graphics?
> 
> 
> 
> @gianlucainnocenti Your IOReg shows no graphics hardware acceleration for iGPU.

You are right, maybe with AMD dGPU wired directly to internal screen / eGPU it might be possible. Might be possible that Tiger Lake needs same D3 patch as Comet/Ice have and to include new pci ids?

gianlucainnocenti, 2021-09-15:
@m0d16l14n1 Yes, could be, I am verifying some info about...I hope to find a solution..
Thank for your message:)

gianlucainnocenti, 2021-09-15:
@ben9923 Do you have some instruction about "D3 Patch" and "IDs should be added to match the ICL/CML+ class "

ben9923, 2021-09-15:
> @ben9923 Do you have some instruction about "D3 Patch" and "IDs should be added to match the ICL/CML+ class "
> 
> 

You already did it by adding the ID to the proper location in the Info.plist.

gianlucainnocenti, 2021-09-15:
@Graphic id you mean?!Or l2c id?

ben9923, 2021-09-15:
> @Graphic id you mean?!Or l2c id?

I2C...

gianlucainnocenti, 2021-09-15:
@ben9923 ok ...scuseme

ben9923, 2021-09-17:
> @ben9923 ok ...scuseme...File I uploaded are correctly? Let me know

An example would be the archive you've uploaded to Gitter. It has ACPI, IOReg, logs, etc.

ben9923, 2021-09-19:
Confirmed to work by adding PCI IDs to the ICL+ class in Info.plist
Requires `force-polling` as VoodooGPIO doesn't support Tiger Lake yet.

Once both changes will be merged to master (& tested) we can close this issue.
CC @kprinssu

kprinssu, 2022-02-27:
This should be added within 2.7, please test using the latest release. If not, please re-open this issue.

closed 2022-02-27 by kprinssu