VoodooI2C/VoodooI2C #485: IOPropertyMatch does not work correctly in VoodooI2CHID

gvkt, 2021-11-12:
The IOKitPersonalities section in the kext has multiple Event Driver personalities. Each of them, in theory, support a IOPropertyMatch property to specify a property and a value that should match a device property for that Event Driver to be considered for that device. 

This works correctly if the Property Match is in the VoodooI2CHIDDevice section to make the entire kext available for devices with matched name and not to another I2c device with a different name. See Fig 1 - Fig 3. where it works correctly

[image]

[image]

[image]

However, if the Property Match is used in any of the other personalities (a use case is to have different event drivers assigned to different devices - e.g., a touch screen and a trackpad, or to prevent a specific event driver from being considered for a device, e.g., avoid touch screen event driver from being used for an unsupported trackpad and let it be used as just a mouse instead), that personality becomes unavailable for ANY device even if there is a match. 

For example:

[image]

The above configuration should allow the Touch Screen device driver to be available to the device named ELAN9008 but not to another device with a different name which may use other event drivers in the kext.

What happens: The event driver property match appears to fail even if the name matches. Hence, that event driver is not assigned to the device or to any device since it appears to fail for any device regardless of name.

What should happen: The device whose property matches the condition should consider that section as if that property match didn't exist. So, in the above examples, the device named ELAN9008 should get the Touch Screen Event Driver in both scenarios which should be equivalent to each other as long as the name of the device is ELAN9008.

ben9923, 2021-11-13:
I don't think you understand how IOKit matching works. Those personalities are checked against the provider...

In the `VoodooI2CHIDDevice` case, provider is ACPI device nub, which VoodooI2C adds the `name` property to.
In the `XXEventDriver` case, provider is the HID interface (You can see it right above the event driver). It has no `name` property you can match against.

You can see that, per the HID spec & IOKit implementation, those event drivers are matched by usage pairs.
Each personality has a score. After finding all of the matches, the one with the highest score is chosen (If it fails, next one is used instead).

As for your specific case, is your device a "screenpad"? i.e. those funky trackpads with a screen?
See #474 for an `IOProbeScore` change that will prioritize trackpad over touchscreen event driver.
I think we'll add this change to VoodooI2CHID so you won't have to do it manually.

gvkt, 2021-11-13:
Thanks for the explanation. I was going by the name matching in the Synaptics personality included in the kext. The reason changing IOProbeScore wasn't a solution in this case is that it would prevent the Touch Screen Event Driver from attaching to the touch screen screwing up its handling.

But the explanation allowed me to differentiate between the touchscreen and trackpad by doing a property match on the productID like so (in my specific case, not a general solution for the kext)

[image]

That allows the different event drivers to attach to the different devices as should happen

[image]

As you can see, this fixes the problem I was trying to solve in this thread. So, this issue can be closed. Although, it would be nice to have a solution for the kext in general that would allow this disambiguation with having to manually change the Info.plist. Perhaps a boot flag if possible.

There is still a problem of the Precision Trackpad Event Driver not working right for the GDX1515 device - no multi-touch and no left and right mouse clicks, only one type of click event that seems to trigger some kind of preview on anything clicked (not sure what the equivalent of this is with a normal mouse as I have never encountered that before).

But this is a separate issue and can be investigated independent of this necessary disambiguation between the two HID devices on the laptop as solved here.

Thanks

closed 2021-11-25 by gvkt