VoodooI2C/VoodooI2C #487: Almost silent failure mode of GPIO pinning resulting in polling while IORegistry suggests success

gvkt, 2021-12-07:
There is an almost silent mode of pinning failure which makes the kexts resort to polling even when the GPIO pinning seems to have succeeded (as verified via just looking at the IORegistry after the change). This is only evident by looking at the elusive logs that most people won't find and they will think GPIO pinning has succeeded while it is doing polling. IORegistry will list the GPIO pin and interrupt even though polling is in use.

Since many incorrect pin numbers can cause this situation, people may think they have switched to interrupt based reports while it is doing polling underneath.

This is when an incorrect pin number leads to VoodooI2CHIDDevice::handleStart not finding an interrupt source for the specified pin and switches to polling. This failure is not reflected in the IORegistry properties set for the pin number and interrupt and looks as if GPIO pinning has succeeded.

It might be useful to set an explicit flag in the device properties for polling being true as default which is set to false only on successful interrupt set up (or vice versa) so a confirmation can be done via looking at IORegistry as to whether polling or interrupt is the actual mode in use.

ben9923, 2021-12-12:
Sounds good to me, especially when logs seem to be broken (Requiring users to look at `dmesg`, only possible by increasing `msgbuf` - manually or via DebugEnhancer).

shiecldk, 2021-12-30:
@gvkt I found sqlsec mentions how he derive his gpio pin here: https://www.sqlsec.com/2021/12/screenpad.html
You may want to use Google Translate to read the article.

kprinssu, 2022-02-04:
This has been addressed by @1Revenger1(https://github.com/VoodooI2C/VoodooI2C/pull/493) and there is now an IO Registry to show if you are using interrupts or polling.

closed 2022-02-04 by kprinssu