VoodooI2C/VoodooI2C #489: ATML3000 touchpad not supported by VoodooI2C

Aster-the-Med-Stu, 2021-12-31:
**Describe the bug**
ATML3000 (Atmel MaxTouch controller), widely used on newer Samsung laptops, are not supported by VoodooI2C.

**Did you read the common errors documentation?**
Yes.

**Did you read the troubleshooting documentation?**
Yes.

**Have you searched the issue on Github, Gitter, or Google?**
Yes. It is reported in #242 and asked by other people from time to time in Gitter but didn't catch devs' attention.

**System Environment**

1. What is your operating system? 
    macOS Big Sur 11.5.2
2. What version of the drivers (including drivers) are you using?
    
3. What is your hardware configuration?
    Samsung Galaxy Book Flex 13.3" (NP930QCG)

**Troubleshooting Archive**
System DSDT (There's no DSDT patch applied) and IOReg is attached. (Is it necessary to apply SSDT-I2CF.dsl in my sceneario?)
[归档.zip](https://github.com/VoodooI2C/VoodooI2C/files/7796404/default.zip)

**Additional context**
Tried loading ATML1000 satellite by modifying ```VoodooI2C.kext/Contents/PlugIns/VoodooI2CAtmelMXT.kext/Contents/Info.plist``` to match ATML3000 but obviously it didn't work. Maybe another satellite is needed, porting from working Linux implementation...?

kprinssu, 2022-01-02:
Did you try VoodooI2C HID as @ben9923 suggested in the original comment?

closed 2022-01-02 by kprinssu

Aster-the-Med-Stu, 2022-01-03:
I think I did load VoodooI2CHID...? @kprinssu

```bash
➜  ~ kextstat | grep -y Voodoo
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   53    1 0                  0xb000     0xb000     com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   66    1 0                  0x27000    0x27000    org.coolstar.VoodooGPIO (1.1) 7ED39A4D-2EFD-35AE-9BC6-6215C17D93F1 <13 6 5 3>
   70    1 0                  0x19000    0x19000    as.acidanthera.voodoo.driver.PS2Controller (2.2.7) 07FB2C37-40D6-37AF-965F-6351582195B7 <13 8 6 5 3 1>
   96    1 0                  0x3b000    0x3b000    com.alexandred.VoodooI2C (2.6.5) 748C2CC2-07FD-3FD5-8E86-874C73317356 <66 54 53 14 13 6 5 3 1>
  103    0 0                  0x34000    0x34000    com.alexandred.VoodooI2CHID (1) 4DB3F9FC-39D6-3AB7-8962-7BC3E137019E <96 54 13 6 5 3>
  104    0 0                  0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.2.7) A7EFC06D-0A5C-3277-9EC1-E182089690D9 <70 54 8 6 5 3 1>
```

Did I load those kexts in a wrong order...?

And IORegistry's device tree shows that it's under ```IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/I2C0@15/VoodooI2CPCILakeController/VoodooI2CControllerNub/VoodooI2CControllerDriver/SPTP```
[image]

Aster-the-Med-Stu, 2022-01-03:
@kprinssu I thought I did load VoodooI2CHID?

```bash
➜  ~ kextstat | grep -y Voodoo
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   53    1 0                  0xb000     0xb000     com.alexandred.VoodooI2CServices (1) DBFA5EBE-2783-30ED-8E39-5829C42D6832 <5 3>
   66    1 0                  0x27000    0x27000    org.coolstar.VoodooGPIO (1.1) 7ED39A4D-2EFD-35AE-9BC6-6215C17D93F1 <13 6 5 3>
   70    1 0                  0x19000    0x19000    as.acidanthera.voodoo.driver.PS2Controller (2.2.7) 07FB2C37-40D6-37AF-965F-6351582195B7 <13 8 6 5 3 1>
   96    1 0                  0x3b000    0x3b000    com.alexandred.VoodooI2C (2.6.5) 748C2CC2-07FD-3FD5-8E86-874C73317356 <66 54 53 14 13 6 5 3 1>
  103    0 0                  0x34000    0x34000    com.alexandred.VoodooI2CHID (1) 4DB3F9FC-39D6-3AB7-8962-7BC3E137019E <96 54 13 6 5 3>
  104    0 0                  0xd000     0xd000     as.acidanthera.voodoo.driver.PS2Keyboard (2.2.7) A7EFC06D-0A5C-3277-9EC1-E182089690D9 <70 54 8 6 5 3 1>
```

Did I load those kexts in a wrong order?

BTW I searched for some Samsung laptops' opencore config. Those using ATML3000, like Samsung 35X0AA, also reported they couldn't make touchpad working.

Aster-the-Med-Stu, 2022-01-03:
Maybe this will be useful...?

Linux didn't have atml3000 working before this touch due to extra 7 bits blocking touchpad entering multitouch mode:
https://patchwork.kernel.org/project/linux-input/patch/1398894520-30407-1-git-send-email-benjamin.tissoires@redhat.com/

kprinssu, 2022-01-04:
Simply posting a kextstat is useless. Your posted ZIP contains a IO Registry that is corrupted.

Please take the time to read the troubleshooting documentation and attach the proper and necessary files in the archive. Additionally, that patch seems to be device specific and *will* cause issues for other devices. You're best off creating forking and doing something similar to this PR in VoodooI2CHID:

https://github.com/VoodooI2C/VoodooI2CHID/pull/57/files

Fvm94, 2022-03-07:
Hello... my samsung laptop have the ATML 3000 touchpad and work with this kext [https://github.com/VoodooI2C/VoodooI2C/files/8027221/Archive.5.zip](url) for this post [https://github.com/VoodooI2C/VoodooI2C/pull/491#issuecomment-1033031136](url)
but if i update to the last version of oficial repository stop work