VoodooI2C/VoodooI2C #496: Right click using click on bottom right doesn't work anymore

tmbt78, 2022-02-28:
**Describe the bug**
Right click doesn't work anymore on latest VoodooI2C version

**Did you read the common errors documentation?**
yes

**Did you read the troubleshooting documentation?**
yes

**Have you searched the issue on Github, Gitter, or Google?**
yes

**System Environment**
Please answer:
1. What is your operating system?  Big Sur 11.6.4
2. What version of the drivers (including drivers) are you using? 2.7 latest
3. What is your hardware configuration? Intel i5 9th gen - 16GB RAM - 512GB SSD

I've upgraded today to the latest VodooI2C version but after reboot i discovered that right click using bottom right setting in trackpad doesn't work anymore. If i chose right click using tap with two fingers it works as expected. Before the update was working fine in both ways. Gesture are working ok.

Mattia

kprinssu, 2022-02-28:
Please attach a troubleshooting archive.

1Revenger1, 2022-02-28:
Don't really have time to look at it right now, but I'm guessing it may be related to this?
https://github.com/VoodooI2C/VoodooI2C/commit/9ed1dcdb6e375c54d61945fe9415ccb63627d2f2

Looks like VoodooInput sets the pressure to 120 to simulate a button press, which was being used before the linked PR above:
https://github.com/acidanthera/VoodooInput/blob/master/VoodooInput/VoodooInputSimulator/VoodooInputSimulatorDevice.cpp#L138

If Apple's MT2 stack relies on pressure for this, then we probably need to use that instead of passing buttons through the trackpoint device. This probably means that devices with a single button still need to either set pressure to 120 (if force touch isn't enabled?) or set "isPhysicalButtonPressed".

tmbt78, 2022-02-28:
> Don't really have time to look at it right now, but I'm guessing it may be related to this? [9ed1dcd](https://github.com/VoodooI2C/VoodooI2C/commit/9ed1dcdb6e375c54d61945fe9415ccb63627d2f2)
> 
> Looks like VoodooInput sets the pressure to 120 to simulate a button press, which was being used before the linked PR above: https://github.com/acidanthera/VoodooInput/blob/master/VoodooInput/VoodooInputSimulator/VoodooInputSimulatorDevice.cpp#L138
> 
> If Apple's MT2 stack relies on pressure for this, then we probably need to use that instead of passing buttons through the trackpoint device. This probably means that devices with a single button still need to either set pressure to 120 (if force touch isn't enabled?) or set "isPhysicalButtonPressed".

Thanks for the fast answer !
I can confirm that reverting back to the latest commit before the one you indicated and compiling again VoodooI2C worked perfectly.
Let me know if you need me to test something else, i will be using this selfcompiled version for the moment.

Thanks again
Mattia

usr-sse2, 2022-03-16:
I've reproduced and understood the problem.
Actually, if a touchpad has two separate physical buttons, then setting it to corner click makes no sense, because it would mean that _left_ button will generate a right click if a finger is in the bottom right corner of the touchpad. It should be used only with touchpads with one integrated button.
I suggest to use the old path (through pressure) if there is one button and the new path (through a separate device) if there are two buttons (with a possibility to override it in case a device reports more buttons than it really has).

There is another problem for left-handed users that have two physical buttons and want to use left button for context menu, but it may be solved after this because it's not a regression.

closed 2022-07-22 by kprinssu

kprinssu, 2022-07-22:
This is fixed with https://github.com/VoodooI2C/VoodooI2C/commit/9ab98319b6411c4a4822cc12f840df85cc684bfc

tmbt78, 2022-07-25:
Hi,
i've compiled VoodooI2C with the latest commits but sadly for me still doesn't work. If i right click is still considered a left click.
I've also deleted the whole VoodooI2C folder and executed the git clone command again from scratch. Same result.
Maybe i'm doing something wrong ?

kprinssu, 2022-07-25:
Are you using VoodooI2CHID?

tmbt78, 2022-07-25:
Yes VoodooI2C + VoodooI2CHID

kprinssu, 2022-07-25:
Try the changes from the CI build:
[VoodooI2C-9ab9831.zip](https://github.com/VoodooI2C/VoodooI2C/files/9183191/VoodooI2C-9ab9831.zip)

kprinssu, 2022-07-26:
@tmbt78 I will be locking this iossue and marking this as done if the above kexts work for you.

tmbt78, 2022-07-27:
@kprinssu Sorry i got busy and couldn't answer sooner. Sadly i can't use the pre compiled kext because my touchpad goes in timeout during loading if i don't increase timeout ms in VoodooI2C code (i took the idea from an old closed issue here on github). I tried the already compiled kext but i got timeout as i was expecting so i cannot say if that version is working or not.

tmbt78, 2022-09-13:
@kprinssu Hi again. Today i had some spare time and i tried again the latest commit. I sadly can confirm that the latest commit doesn't solve the right click problem. As i stated before i have to modify some timeout values in VoodooI2CControllerDriver and VoodooI2CHIDDevice to avoid timeout during trackpad initialisation  at boot so i cannot use your precompiled binary. But i'm positive i've done everything right using the latest commit to compile but with no luck. So i've self compiled the commit i want (e0de321) with my custom timeout settings and everything is working back to normal.
I'm quite sure i'm doing things right so maybe the latest commit doesn't solve the secondary button problem (at least for my trackpad).

Thanks anyway for the hard work !

commit 524235cb (2022-07-22): 1. Always pass buttons as pressure on single-button touchpads (fix #496)
2. Always disable Force Touch and pass buttons to buttons device on touchpads with two physical buttons (fixes inability to click without finger on the touchpad (#499) and inability to use the right button with default settings)
3. Disable Force Touch on all touchpads if tap to click is disabled (fixes inability to make a non-force click with default settings, especially in Recovery where there is no System Preferences application)
commit 9ab98319 (2022-07-22): Update VoodooI2C Satellites and support for physical buttons (#514)

* 1. Always pass buttons as pressure on single-button touchpads (fix #496)
2. Always disable Force Touch and pass buttons to buttons device on touchpads with two physical buttons (fixes inability to click without finger on the touchpad (#499) and inability to use the right button with default settings)
3. Disable Force Touch on all touchpads if tap to click is disabled (fixes inability to make a non-force click with default settings, especially in Recovery where there is no System Preferences application)

* NULL -> nullptr,  [... cut]