VoodooI2C/VoodooI2C #500: Pen Touch stops working however touchscreen continues to work.

dikamsiyoung, 2022-03-17:
Pen touch stops working in v2.7. Touchscreen works but not Wacom Active Pen.
I downgraded to previous version and pen works again.

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**
macOS 12.3
Voodooi2c v2.7
Lenovo Yoga 720-15iKB i7-7700HQ 4K Display with Touch and Pen Support

kprinssu, 2022-03-17:
Thanks for the report! Can you perhaps attach a copy of IO Registry with 2.6 and 2.7?

dikamsiyoung, 2022-03-18:
As requested, here is my ioreg. As I switched back to 2.7 to take this snapshot, the pen starts working again. Hopefully this helps pinpoint the issue.

[Uploading ioreg-yoga-720-15ikb.zip…]()

kprinssu, 2022-03-18:
I am unable to download the zip file, it's linked back to this Github issue. That pen issue seems to be a kext cache issue, can you double check the IO Registry to check version of VoodooI2C is loaded?

dikamsiyoung, 2022-03-18:
As requested, here is my ioreg. As I switched back to 2.7 to take this snapshot, the pen starts working again. Hopefully this helps pinpoint the issue.

[ioreg-yoga-720-15ikb.zip](https://github.com/VoodooI2C/VoodooI2C/files/8295420/ioreg-yoga-720-15ikb.zip)

dikamsiyoung, 2022-03-18:
> I am unable to download the zip file, it's linked back to this Github issue. That pen issue seems to be a kext cache issue, can you double check the IO Registry to check version of VoodooI2C is loaded?

Sorry about that, I was having poor connection. Yes VoodooI2C was loaded prior switching back to 2.6.5. I could use my touchscreen but just couldn't use my pen.

Does the kext separate both pen and touch inputs?

timothy-ha, 2022-07-06:
I can confirm this as well on my ThinkPad 370 Yoga.  The pen input works without any kexts loaded, and also works on the 2.6.5 I2C kext.  However, when loading the 2.7 kext, pen input is broken.

kprinssu, 2022-07-23:
@dikamsiyoung @timothy-ha @vnln Does the pen not work at all? Does the pointer move? Does left/right click not work?

vnln, 2022-07-23:
@kprinssu As far as I can remember, nothing seemed working at all. Although, using **VoodooI2C.kext** version 2.7 with **VoodooI2CHID.kext** version 2.6.5 works almost fine for me (hover cursor, touch, right-click button – everything except left-click button on a pen).

kprinssu, 2022-07-23:
Thanks @vnln. We may need to compare the changes between 2.6.5 to 2.7 commit by commit. I do not have the time nor hardware to do so.

sunbos, 2022-10-17:
Pen touch stops working in v2.7. Touchscreen works but not Wacom Active Pen.System Environment
macOS 13 Beta
Voodooi2c v2.7
Lenovo Yoga 9 15IMH5 4K Display with Touch and Pen Support

hapshin, 2023-01-16:
@1Revenger1 @gvkt
offending commit https://github.com/VoodooI2C/VoodooI2CHID/commit/7b8ee9f42b9fa1147ee425778d89f978c32e2e43
Tyvm

Baio1977, 2023-03-27:
I also have the usual problem with the Lenovo PEN, I switched to the 2.6.5 version of the Kext, the Pen is working again and on Ventura 13.2.1 I DO NOT have KP as with the 2.7 and 2.7.1 versions.
[](https://github.com/Baio1977/Lenovo-ThinkPad-X1-Tablet-Gen3-Kabylake-r)

NotMifa, 2024-06-01:
Hey, according to https://github.com/VoodooI2C/VoodooI2CHID/pull/72#issuecomment-1667658777 Zhen-Zen has fixed the issue "Fixes https://github.com/VoodooI2C/VoodooI2C/issues/500" which has also been added to the master of this repo. Any chance this could be compiled into the voodooI2CHID kext and be used with the latest 2.8? This would close this issue and solve an issue that many of us were having...