VoodooI2C/VoodooI2C #501: HP 240 G8 TrackPad not working at all

Disk8, 2022-04-02:
**Describe the bug**
A clear and concise description of what the bug is.
 trackPad on laptop not working  , I tried everything but no effect 

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
Yes
**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.
Yes
**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
Yes
**System Environment**
Please answer:
1. macOS Monterey 12.3.1
2.  v2.6.4--2.6.5--2.7 + 
3. laptop HP 240 G8 / i5-1035g1 /IntelUHD/16gb ram/ 512gb nvmeSSD/

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

**Additional context**

looks like I have two device
in DSDT I found >>>>> PS2Mouse device and I2C device under TPD0 I think its TouchPadDevice

Device (PS2M)
        {
            Method (_HID, 0, NotSerialized)  // _HID: Hardware ID
            {
                If ((SSID == HKGU))
                {
                    Return ("ELAN0724")
                }
                ElseIf (((SSID == SNPD) || (SSID == SNPU)))
                {
                    Return ("ELAN0723")
                }
                ElseIf (((SSID == HWGD) || (SSID == HWGU)))
                {
                    Return ("ELAN0703")
                }
                Else
                {
                    Return ("ETD074A")
                }
            }
            Name (_CID, Package (0x02)  // _CID: Compatible ID
            {
                EisaId ("ETD0000"), 
                EisaId ("PNP0F13") /* PS/2 Mouse */
-------------------------------------------------------------------------------------------------------

Scope (_SB.PCI0.I2C0)
    {
        Name (I2CN, Zero)
        Name (I2CX, Zero)
        I2CN = SDS0 /* \SDS0 */
        I2CX = Zero
        Device (TPD0)
        {
            Name (HID2, Zero)
            Name (SBFB, ResourceTemplate ()
            {
                I2cSerialBusV2 (0x0015, ControllerInitiated, 0x00061A80,
                    AddressingMode7Bit, "\\_SB.PCI0.I2C0",
                    0x00, ResourceConsumer, _Y43, Exclusive,
                    )
            })
            Name (SBFG, ResourceTemplate ()
            {
                GpioInt (Level, ActiveLow, ExclusiveAndWake, PullDefault, 0x0000,
                    "\\_SB.GPI0", 0x00, ResourceConsumer, ,
                    )
                    {   // Pin list
                        0x0000
                    }
            })
            Name (SBFI, ResourceTemplate ()
            {
                Interrupt (ResourceConsumer, Level, ActiveLow, ExclusiveAndWake, ,, _Y44)
                {
                    0x00000000,
                }
            })
            CreateWordField (SBFB, \_SB.PCI0.I2C0.TPD0._Y43._ADR, BADR)  // _ADR: Address
            CreateDWordField (SBFB, \_SB.PCI0.I2C0.TPD0._Y43._SPE, SPED)  // _SPE: Speed
            CreateWordField (SBFG, 0x17, INT1)
            CreateDWordField (SBFI, \_SB.PCI0.I2C0.TPD0._Y44._INT, INT2)  // _INT: Interrupts
            Method (_INI, 0, NotSerialized)  // _INI: Initialize
            {
                If ((OSYS < 0x07DC))
                {
                    SRXO (GPDI, One)
                }

                INT1 = GNUM (GPDI)
                INT2 = INUM (GPDI)
                If ((TPDM == Zero))
                {
                    SHPO (GPDI, One)
                }

                If ((SSID == HKGU))
                {
                    _HID = "ELAN0724"
                }
                ElseIf (((SSID == SNPD) || (SSID == SNPU)))
                {
                    _HID = "ELAN0723"
                }
                Else
                {
                    _HID = "ELAN0703"
                }

                HID2 = One
                BADR = 0x15
                SPED = 0x00061A80
                Return (Zero)
            }

Disk8, 2022-04-02:
[DSDT-NoError Hp240.aml.zip](https://github.com/VoodooI2C/VoodooI2C/files/8403369/DSDT-NoError.Hp240.aml.zip)

kprinssu, 2022-04-02:
Please use Gitter for support with your device configuration.

closed 2022-04-02 by kprinssu