VoodooI2C/VoodooI2C #506: No Trackpad functionality what so ever: HP Pavillion x360 ELAN0722

Babyyoda777, 2022-06-02:
I have tried all the VoodooI2C, VoodooPS2, VooodooSMbus plugins but none seem to work.
CPU: i3 1005G1
Kexts used are all the latest, I am using Voodooi2c with the HID plugin
I have had no luck whatsoever and am now stuck on what to try next.
[image]
It seems I have correctly pinned my GPI0, but here is a screen shot incase.
I followed the Dortiana guide and checked all he trouble shooting issues.

Babyyoda777, 2022-06-02:
[Azlan’s MacBook Air.zip](https://github.com/VoodooI2C/VoodooI2C/files/8823508/Azlan.s.MacBook.Air.zip)
Here is an IOReg of the mackintosh

kprinssu, 2022-06-02:
Please use Gitter for support with configuration.

closed 2022-06-02 by kprinssu