VoodooI2C/VoodooI2C #507: kp on shutdown with USB connected touchpad - macOS 13 Beta

whatnameisit, 2022-06-19:
**Describe the bug**
KP on shutdown

**Did you read the common errors documentation?**
Yes

**Did you read the troubleshooting documentation?**
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Yes

**System Environment**
Please answer:
1. What is your operating system? macOS 13 Beta
2. What version of the drivers (including drivers) are you using? VoodooI2C 2.7, VoodooI2CHID 2.7 (not needed with the touchpad, only for the touchscreen), and AlpsHID 1.2
3. What is your hardware configuration? HP Elite X2 1012 G2 has detachable keyboard that is connected through USB.

**Troubleshooting Archive**
[아카이브 2.zip](https://github.com/VoodooI2C/VoodooI2C/files/8935589/2.zip)

**Additional context**
I have uploaded a clean DSDT.aml because no patching is necessary for USB connected touchpads to function.
The kp log is grabbed without VoodooI2CHID to narrow down the cause.
The touchpad works natively with limited gestures (up to two fingers) on macOS 13 Beta. I don't remember it working before this OS. IOReg: 
[jinmo의 MacBook Pro alpshid disabled.zip](https://github.com/VoodooI2C/VoodooI2C/files/8935604/jinmo.MacBook.Pro.alpshid.disabled.zip)

@Lorys89, https://github.com/blankmac/AlpsHID/issues/17#issuecomment-1159639465
> hello, on macos beta 13 it is a problem of all i2c that are alps or elan or others that work on USB protocol

Maybe they can provide more information.

kprinssu, 2022-06-21:
Please attach a copy of the kernel panic logs or a screenshot of the trace.

Lorys89, 2022-06-22:
[IOHIDFamily macos big sur.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/8954899/IOHIDFamily.macos.big.sur.kext.zip)
[IOHIDFamily macos ventura.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/8954901/IOHIDFamily.macos.ventura.kext.zip)

whatnameisit, 2022-06-22:
@kprinssu, I had attached the log in the first comment.
> **Troubleshooting Archive** [아카이브 2.zip](https://github.com/VoodooI2C/VoodooI2C/files/8935589/2.zip)

Unless you meant a different log?

crystall1nedev, 2022-07-12:
@kprinssu Since OP didn't post a log but I have the same issue (as well as random kernel panics during use), I can provide one.

[Kernel-2022-07-12-162318.txt](https://github.com/VoodooI2C/VoodooI2C/files/9096602/Kernel-2022-07-12-162318.txt)

whatnameisit, 2022-07-13:
Thank you for ignoring my log upload even when I mentioned right before your comment. In case nobody has seen it, I'll quote here again.

> @kprinssu, I had attached the log in the first comment.
> 
> > **Troubleshooting Archive** [아카이브 2.zip](https://github.com/VoodooI2C/VoodooI2C/files/8935589/2.zip)
> 
> Unless you meant a different log?

kprinssu, 2022-07-22:
@Lorys89 Are you using a modified VoodooI2CHID kext from https://github.com/blankmac/AlpsHID

If so, we will not be providing support for it.

Lorys89, 2022-07-22:
> @Lorys89 @crystall1nedev Are you using a modified VoodooI2CHID kext from https://github.com/blankmac/AlpsHID
> 
> If so, we will not be providing support for it.

In the pc Dell xps 9350 I do not use alpshid, I have on bus I2c I trackpad and on usb the touchscreen

Lorys89, 2022-07-22:
The problem is if you have touchscreen or trackpad on usb

kprinssu, 2022-07-23:
@Lorys89 Can you attach a valid IO Registry file? Please use https://www.tonymacx86.com/threads/guide-how-to-make-a-copy-of-ioreg.58368/

crystall1nedev, 2022-07-23:
> @Lorys89 @crystall1nedev Are you using a modified VoodooI2CHID kext from https://github.com/blankmac/AlpsHID
> 
> If so, we will not be providing support for it.

Nope, just the VoodooI2CHID provided on this repository. I don't currently have access to the machine in question but I'll get the ioreg once I can.

Vorshim92, 2022-07-28:
[xps9350 usb touch.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/9213146/xps9350.usb.touch.ioreg.zip)
[kp xps 9350.txt](https://github.com/VoodooI2C/VoodooI2C/files/9213155/kp.xps.9350.txt)
here the ioreg from @Lorys89 's xps9350
if you need anything else let me know

kprinssu, 2022-07-28:
@Vorshim92 I really need a valid IO Registry dump, please use the one here https://www.tonymacx86.com/threads/guide-how-to-make-a-copy-of-ioreg.58368/

Vorshim92, 2022-07-28:
> @Vorshim92 I really need a valid IO Registry dump, please use the one here https://www.tonymacx86.com/threads/guide-how-to-make-a-copy-of-ioreg.58368/

I attached the ioreg compressed... i did it with this last https://github.com/khronokernel/IORegistryClone/raw/master/ioreg-302.zip

kprinssu, 2022-07-28:
Thanks. It looks it an issue with `VoodooI2CTouchscreenHIDEventDriver`. I'll look into this shortly.

Vorshim92, 2022-07-28:
if you need I have also kp and ioreg from lenovo miix720 with touchpad on usb with the same problem.
here:
[MacBook Pro admin.ioreg.zip](https://github.com/VoodooI2C/VoodooI2C/files/9214218/MacBook.Pro.admin.ioreg.zip)
[kp lenovo.txt](https://github.com/VoodooI2C/VoodooI2C/files/9214222/kp.lenovo.txt)

SukkaW, 2022-08-04:
I am also having the same issue. The macOS 13 B4 KP after exact 2 minutes after the login.

I have managed to get a full backtrace:

```

I am having the same issue on macOS 13 B4, same kernel panic. And here is the backtrace of the panic (using keepsyms):

IMG_20220804_165551

panic(cpu 12 caller 0xffffff8003153797): Double fault at 0xffffff80056cb6c4, registers:
CR0: 0x0000000080010033, CR2: 0xffffffa0ba4e7ff8, CR3: 0x0000000879bb2049, CR4: 0x00000000003626e0
RAX: 0xffffff80074013f8, RBX: 0xffffff906b0f4560, RCX: 0x0000000000000021, RDX: 0x0000000000000002
RSP: 0xffffffa0ba4e8000, RBP: 0xffffffa0ba4e8030, RSI: 0xffffff8b9ec7d8c0, RDI: 0xffffff906b0f4560
R8:  0x0000000000000000, R9:  0x0000000000000000, R10: 0xffffff8002e5eb90, R11: 0xffffff8003fe7dc2
R12: 0x00000000e00002d7, R13: 0xffffff8b9ec7d8c0, R14: 0x0000000000000021, R15: 0x0000000000000021
RFL: 0x0000000000010246, RIP: 0xffffff80056cb6c4, CS:  0x0000000000000008, SS:  0x0000000000000010
Error code: 0x0000000000000000
 @trap_native.c:168
Panicked task 0xffffff906adc0df8: 15 threads: pid 182: WindowServer
Backtrace (CPU 12), panicked thread: 0xffffff906b3dd598, Frame : Return Address
0xffffff8002d12270 : 0xffffff8002ff6a1d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffff8002d122c0 : 0xffffff8003163316 mach_kernel : _kdp_i386_trap + 0x116
0xffffff8002d12300 : 0xffffff80031525a0 mach_kernel : _kernel_trap + 0x3e0
0xffffff8002d12350 : 0xffffff8002f91951 mach_kernel : _return_from_trap + 0xc1
0xffffff8002d12370 : 0xffffff8002ff6cfd mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffff8002d12460 : 0xffffff8002ff63a9 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffff8002d124c0 : 0xffffff80037e15fb mach_kernel : _panic + 0x84
0xffffff8002d125b0 : 0xffffff8003153797 mach_kernel : _sync_iss_to_iks_unconditionally + 0x167
0xffffff8002d126c0 : 0xffffff80037e93c8 mach_kernel : _panic_double_fault64 + 0x27
0xffffff8002d126d0 : 0xffffff8002f91e4f mach_kernel : _hndl_double_fault + 0xf
0xffffffa0ba4e8030 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8070 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e80b0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e80f0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8130 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8170 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e81b0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e81f0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8230 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8270 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e82b0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e82f0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8330 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8370 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e83b0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e83f0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8430 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8470 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e84b0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e84f0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8530 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8570 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e85b0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e85f0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
[... cut ...]
0xffffffa0ba4e8b70 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8bb0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8bf0 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHIDReportTypej + 0x26
0xffffffa0ba4e8c30 : 0xffffff80056cb6ca com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice9getReportEP18IOMemoryDescriptor15IOHID
```

Lorys89, 2022-08-29:
inject these 2 kext into config, setting min kernel 22.0.0 and see if it fixes.
I await news
[AppleUSBMultitouch.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/9442173/AppleUSBMultitouch.kext.zip)
[AppleUSBTopCase.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/9442174/AppleUSBTopCase.kext.zip)

yurhett, 2022-10-25:
Same issue in Release Version of macOS Ventura, please fix, please 🥺

flp4, 2022-10-29:
I am having the same issue. Anything new on this issue?

kprinssu, 2022-10-31:
@SukkaW Please avoid pinging me. That looks to be a different panic than what was originally reported.

As I previously mentioned, I don't have the hardware to debug and fix this panic. PRs are always welcome.

SukkaW, 2022-10-31:
> @SukkaW Please avoid pinging me. 

I am sorry.

> That looks to be a different panic than what was originally reported.

From others' reports (their attached kernel panic diag), I suspect the cause of the issue is the same. I will open another issue if you believe the panic is unrelated to others.

kprinssu, 2022-10-31:
> > @SukkaW Please avoid pinging me.
> 
> I am sorry.
> 
> > That looks to be a different panic than what was originally reported.
> 
> From others' reports (their attached kernel panic diag), I suspect the cause of the issue is the same. I will open another issue if you believe the panic is unrelated to others.

No worries, I am hoping Apple releases the new sources for IOKit so we can find the changes on Ventura or look at VoodooI2C's `getReport`.

Erictoby, 2022-11-03:
Peoples experiencing this issue may disable the USB port connected to the touch screen as a **temporary** fix. For example, edit the USBPorts.kext or USBMap.kext or similar things that you use. It will let you circumvent the issue until it's fixed but you won't be able to use the touch screen.

Jon602, 2022-11-10:
Dell Latitude 7390 with touchscreen - temporary fix was leaving all kext as was but disabling touchscreen in BIOS

VGerris, 2022-11-27:
I have the same issue on 13.0.1 on a Dell Precision 5520 with touch screen. No issues in Monterey but in Ventura on wake it reboots and shutdown also reboots. Disabling touch screen in the BIOS solves the issue. On a Inspiron 5482 laptop with touchscreen I do not have this issue. I'm not too familiar with the lower lever workings of this but let me know if I can gather more info that can be useful. Also happy to test on driver update ofcourse.

backdash756, 2022-11-29:
Microsoft Surface Pro 7 with connected Pro Type Cover has the same issue..

https://github.com/Xiashangning/BigSurface/issues/48

dkoluris, 2022-12-07:
I have a Dell Inspiron 7773, and experience this problem in Ventura. Let me know how I can help by providing whatever logs you wish to attempt to solve this problem.

kprinssu, 2022-12-07:
Please note this is a known issue, there are numerous issues with https://github.com/VoodooI2C/VoodooI2CHID/blob/master/VoodooI2CHID/VoodooI2CTouchscreenHIDEventDriver.cpp. I will try work on a fix during the holidays but will not guarantee that it will fix this panic. PRs are always welcome.

Furthermore, to avoid more people replying and spamming this issue, I have locked this issue to just contributors and collaborators.

kprinssu, 2022-12-09:
@SukkaW @EricToby @Jon602 @Vorshim92 @Lorys89 @crystall1nedev Please try the attached beta build, it has some minor fixes and logs. I cannot say this will fix the panic but will track it down.
[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/10198113/Archive.zip)

alessaba, 2022-12-10:
I tried on my T480s with Raydium touch and the panic persists. How can i help in debugging this?

SukkaW, 2022-12-10:
> Please try the attached beta build, it has some minor fixes and logs. I cannot say if this will fix the panic but will help track it down.

FYI, the panic I originally reported is indeed "a different panic than what was originally reported" and not related to others.

@williambj1 has created a custom VoodooI2C build (with the source code in the current `master` branch), using macOS 13 SDK (instead of the `mackernelsdk`). And he and I are able to use the touchpad without any KP. So the panic I reported is likely an issue with `acidanthera/mackernelsdk`.

alessaba, 2022-12-10:
> > Please try the attached beta build, it has some minor fixes and logs. I cannot say if this will fix the panic but will help track it down.
> 
> FYI, the panic I originally reported is indeed "a different panic than what was originally reported" and not related to others.
> 
> @williambj1 has created a custom VoodooI2C build (with the source code in the current `master` branch), using macOS 13 SDK (instead of the `mackernelsdk`). And he and I are able to use the touchpad without any KP. So the panic I reported is likely an issue with `acidanthera/mackernelsdk`, not VoodooI2C.
> 
> I just hide my original comment, as it is not related anymore.

Where can i find such build? His gothub does not appear to have any fork

andreszerocross, 2022-12-10:
> @SukkaW @Erictoby @Jon602 @Vorshim92 @Lorys89 @crystall1nedev Please try the attached beta build, it has some minor fixes and logs. I cannot say if this will fix the panic but will help track it down. [Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/10198113/Archive.zip)
> 
> The changes are on this branch here: https://github.com/VoodooI2C/VoodooI2CHID/tree/issue-507

I am installing in in Lenovo Thinkpad X280 with Touchscreen in XHC Bus (ELAN Chipset) and got kernel panic when reboot and shutdown with this kexts>

And system will freeze too when touchscreen is touched, with release version there is no freeze when touch but KP when reboot and shutdown with different KP logs from below

[image]

kprinssu, 2022-12-11:
@alessaba Please do not ask for support if you are unable to generate system logs. I am very sorry but I really do not have the time to help you with that.

@SukkaW That is interesting and would imply that the binary that we build against (libkmod) would need to be updated.

@andreszerocross Please revert back to the latest build of VoodooI2C. 

I am rather out of ideas as to what could be causing this panic. I will most likely need actual hardware to quickly track where the kernel panic is raised. Unfortunately, I do not have a device with a USB based touchscreen.

alessaba, 2022-12-11:
Very rude. I tried helping you and also discovered a regression in your "beta" build. No wonder you are taking so long. Flagging as spam and "off-topic" people who want to improve the situation is very wrong, but whatever, don’t worry, i guess I won’t bother you anymore, sorry for trying to be somewhat helpful. I just said that i could not find where the iologs were going, and that the panics caused from your kext were not saved.

kprinssu, 2022-12-11:
@alessaba I was hoping more technically savvy users would help with tracing the panic. This was to avoid the back and forth of instructing you on how to debug a beta/dev build. From past experiences, remote troubleshooting is very time consuming, especially in a thread style Github issue ticket. This is why I was a bit hesitant to even do a such a build. You are welcome to use the release builds of VoodooI2C.

As previously mentioned, just commenting "it doesn't work" is neither constructive nor helpful in tracking the issue down. It is spam at best and unnecessarily clutters things.

crystall1nedev, 2022-12-13:
> @SukkaW @EricToby @Jon602 @Vorshim92 @Lorys89 @crystall1nedev Please try the attached beta build, it has some minor fixes and logs. I cannot say if this will fix the panic but will help track it [...]

Alright, sorry for a late response (been slammed with a lot of personal stuff and haven't had the time :P) I'll test it when I get back home tonight, see what logs I can gather for you.

kprinssu, 2022-12-13:
@crystall1nedev That build is regrettably broken. I am not sure what is causing this panic and further work will be guesses.

kprinssu, 2022-12-30:
This is not an issue with VoodooI2C and any satellite kexts. On a test device, I had to modify the Alps kext to not rely on the VoodooI2CHID and just load VoodooInput.

The panic still occurs without any VoodooI2C kexts are loaded. I am now digging into VoodooInput to figure out where the issue is.

kprinssu, 2023-01-01:
@crystall1nedev Can you please try this build of [VoodooI2C.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/10329269/VoodooI2C.kext.zip)? It contains the latets build of 2.7 with a custom build of VoodooInput where the trackpoint classes are disabled.

zerostar-git, 2023-01-02:
> @Erictoby @Jon602 @Vorshim92 @Lorys89 @crystall1nedev Can you please try this build of [VoodooI2C.kext.zip](https://github.com/VoodooI2C/VoodooI2C/files/10329269/VoodooI2C.kext.zip)? It contains the latets build of 2.7 with a custom build of VoodooInput where the trackpoint classes are disabled.

Dear Mr. kprinssu. Thank you for your build. 
I tested on my device and issue still exist. 
Just more information: My device have PS2 keyboard and touchpad so I just use VoodooI2C for touchscreen. If I remove VoodooI2C (live without touchscreen), issue is gone. With lastest VoodooInput inside VoodooPS2Controller I have no issue when it shutdown. 
One more thing, if I disable VoodooHID satellite in config (or remove it) I have no touchscreen ofcouse, but no issue when I shutdown.

kprinssu, 2023-01-02:
@zerostar-git As previously mentioned, the issue is with VoodooInput. You are using the build from VoodooPS2 which *does not* contain the fix. You **must** use VoodooInput build from VoodooI2C attaching below. This is a release candidate for 2.7.1.

[VoodooI2C-330a454.zip](https://github.com/VoodooI2C/VoodooI2C/files/10332088/VoodooI2C-330a454.zip)

zerostar-git, 2023-01-02:
> @zerostar-git As previously mentioned, the issue is with VoodooInput. You are using the build from VoodooPS2 which _does not_ contain the fix. You **must** use VoodooInput build from VoodooI2C [...]
@kprinssu Thank you. I tested VoodooI2C 2.7.1 you posted, issue still there. 
Even have VoodooPS2, I still choose VoodooInput build from VoodooI2C in config.plist. What I mean on last comment is even I have VoodooInput (VoodooPS2) with out VoodooHID, I have no issue. So problem really come from VoodooInput you think?

kprinssu, 2023-01-02:
Can you please upload your EFI folder? I want to verify that you are using the correct VoodooInput build.

zerostar-git, 2023-01-02:
@kprinssu 
Sure!
https://file.io/6C9hS6mLdjzN

kprinssu, 2023-01-02:
> @kprinssu Sure! https://file.io/6C9hS6mLdjzN

Thanks @zerostar-git, you're using the right build of VoodooInput. I believe there's probably more to this. Would you be able to try a new build?

zerostar-git, 2023-01-02:
@kprinssu Sure! Just upload to me :) Thank you

kprinssu, 2023-01-02:
@zerostar-git Please give this a go, 
[Release.zip](https://github.com/VoodooI2C/VoodooI2C/files/10332445/Release.zip). This contains similar changes from VoodooInput by adding a willTerminate method to `VoodooI2CMultitouchHIDEventDriver`. The only changes here are within VoodooI2CHID.

zerostar-git, 2023-01-02:
@kprinssu Yes still panic, here is log

> panic(cpu 0 caller 0xffffff801d348953): Kernel trap at 0xffffff801f8da704, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000001060, CR3: 0x0000000027302000, CR4: 0x00000000003626e0
RAX: 0x0000000000001000, RBX: 0xffffff96ec2e5cc0, RCX: 0x0000000000000001, RDX: 0x0000000000000001
RSP: 0xffffffe589487e10, RBP: 0xffffffe589487e30, RSI: 0xffffffa5515de400, RDI: 0xffffff96ec2e5cc0
R8:    0x000000000000ffff, R9:    0x00000000ffffffff, R10: 0x0000000000000001, R11: 0xffffff9bb5cd50c8
R12: 0x0000000000000000, R13: 0xffffff9bb80ae130, R14: 0x0000000000000001, R15: 0xffffffa5515de400
RFL: 0x0000000000010246, RIP: 0xffffff801f8da704, CS:    0x0000000000000008, SS:    0x0000000000000010
Fault CR2: 0x0000000000001060, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Panicked task 0xffffff9bb82d7208: 181 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff9bb8141598, Frame : Return Address
0xffffffe5894876f0 : 0xffffff801d1eb31d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffe589487740 : 0xffffff801d359236 mach_kernel : _kdp_i386_trap + 0x116
0xffffffe589487780 : 0xffffff801d348480 mach_kernel : _kernel_trap + 0x3e0
0xffffffe5894877d0 : 0xffffff80220015ca as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x47a
0xffffffe589487850 : 0xffffff801d185951 mach_kernel : _return_from_trap + 0xc1
0xffffffe589487870 : 0xffffff801d1eb5fd mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffe589487960 : 0xffffff801d1eaca9 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffffe5894879c0 : 0xffffff801d9e08cb mach_kernel : _panic + 0x84
0xffffffe589487ab0 : 0xffffff801d348953 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffe589487c30 : 0xffffff801d348642 mach_kernel : _kernel_trap + 0x5a2
0xffffffe589487c80 : 0xffffff80220015ca as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x47a
0xffffffe589487d00 : 0xffffff801d185951 mach_kernel : _return_from_trap + 0xc1
0xffffffe589487d20 : 0xffffff801f8da704 com.apple.iokit.IOHIDFamily : __ZN11IOHIDDevice13willTerminateEP9IOServicej + 0x1a
0xffffffe589487e30 : 0xffffff801d8e40e3 mach_kernel : __ZN9IOService19actionWillTerminateEPS_jP7OSArraybPv + 0xa3
0xffffffe589487ea0 : 0xffffff801d912f0e mach_kernel : __ZN10IOWorkLoop9runActionEPFiP8OSObjectPvS2_S2_S2_ES1_S2_S2_S2_S2_ + 0x3e
0xffffffe589487ee0 : 0xffffff801d8e333f mach_kernel : __ZN9IOService15terminateWorkerEj + 0x52f
0xffffffe589487f60 : 0xffffff801d8efd97 mach_kernel : __ZN9IOService15terminateThreadEPvi + 0x97
0xffffffe589487fa0 : 0xffffff801d18519e mach_kernel : _call_continuation + 0x2e
            Kernel Extensions in backtrace:
                  com.apple.iokit.IOHIDFamily(2.0)[56F4AF08-8819-3F6D-B6B2-E35996B954D3]@0xffffff801f8cc000->0xffffff801f92cfff
                        dependency: com.apple.iokit.IOReportFamily(47)[EF0F8F2D-D40B-3B22-A3BC-8BB0145DA2F7]@0xffffff801fc41000->0xffffff801fc43fff
                  as.vit9696.VirtualSMC(1.3)[180B2207-900E-3469-B233-C81E9C1F3E84]@0xffffff8021ff2000->0xffffff8022018fff
                        dependency: as.vit9696.Lilu(1.6.2)[8DEDAFB9-5305-3D7B-AC3A-99A039007AFC]@0xffffff8021014000->0xffffff802109bfff
                        dependency: com.apple.iokit.IOACPIFamily(1.4)[7EF4F5CB-37B6-3929-B1E2-A0E960118509]@0xffffff801f791000->0xffffff801f792fff

Process name corresponding to current thread (0xffffff9bb8141598): kernel_task
Boot args: -v -wegnoegpu keepsyms=1 debug=0x100 -no_compat_check alcid=21

Mac OS version:
22C65

Kernel version:
Darwin Kernel Version 22.2.0: Fri Nov 11 02:08:47 PST 2022; root:xnu-8792.61.2~4/RELEASE_X86_64
Kernel UUID: 6B3374E9-7FD7-354D-886F-E995FAAC2CA3
roots installed: 0
KernelCache slide: 0x000000001ce00000
KernelCache base:    0xffffff801d000000
Kernel slide:            0x000000001cedc000
Kernel text base:    0xffffff801d0dc000
__HIB    text base: 0xffffff801cf00000
System model name: MacBookPro15,4 (Mac-53FDB3D8DB8CA971)
System shutdown begun: YES
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 64612053184
Last Sleep:                      absolute                      base_tsc                    base_nano
    Uptime    : 0x0000000f0b6bba77
    Sleep      : 0x0000000000000000 0x0000000000000000 0x0000000000000000
    Wake        : 0x0000000000000000 0x0000000c7a6202c2 0x0000000000000000
Compressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space
Zone info:
    Zone map: 0xffffff9083a5c000 - 0xffffffb083a5c000
    . PGZ      : 0xffffff9083a5c000 - 0xffffff9085a5d000
    . VM        : 0xffffff9085a5d000 - 0xffffff955225c000
    . RO        : 0xffffff955225c000 - 0xffffff96eba5c000
    . GEN0    : 0xffffff96eba5c000 - 0xffffff9bb825c000
    . GEN1    : 0xffffff9bb825c000 - 0xffffffa084a5c000
    . GEN2    : 0xffffffa084a5c000 - 0xffffffa55125c000
    . GEN3    : 0xffffffa55125c000 - 0xffffffaa1da5c000
    . DATA    : 0xffffffaa1da5c000 - 0xffffffb083a5c000
    Metadata: 0xffffff905da4c000 - 0xffffff907da4c000
    Bitmaps : 0xffffff907da4c000 - 0xffffff9083a4c000

kprinssu, 2023-01-02:
That is not related to VoodooInput nor VoodooI2C related, rather with VirtualSMC. I am going to ping vit9696 and get their thoughts on this.

Thanks for testing @zerostar-git .

Erictoby, 2023-01-05:
@zerostar-git  I get exactly the same log, except that I don't have the line of "as.vit9696.VirtualSMC(1.3)".  (using the 2.7.1 and the new VoodooInput.kext)

```
...
0xffffffd5a7913fa0 : 0xffffff801278819e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IOHIDFamily(2.0)[96D09D0F-DCBC-317F-A955-2BAAEB46FCFF]@0xffffff8014ecb000->0xffffff8014f2bfff
            dependency: com.apple.iokit.IOReportFamily(47)[3115840D-90F4-383F-8991-D3CF8891C24C]@0xffffff8015240000->0xffffff8015242fff

Process name corresponding to current thread (0xffffff958203d598): kernel_task
Boot args: keepsyms=1 debug=0x100 darkwake=0 brcmfx-country=#a
...
```

VGerris, 2023-01-07:
Hi, I still have the issue of a reboot with the touch screen enabled and the error popping up.
This is the stack trace, although I am not sure it was from the actual crash, might have been successive boot:
panic(cpu 0 caller 0xffffff8009748953): Kernel trap at 0xffffff8009301082, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff8bd0ffc000, CR3: 0x0000000017949000, CR4: 0x00000000003626e0
RAX: 0x00000004cc3ba600, RBX: 0xffffff909d3b6000, RCX: 0x0000000000000200, RDX: 0x0000000000000800
RSP: 0xffffffaba5c8f808, RBP: 0xffffffaba5c8f820, RSI: 0xffffff8bd0ffc000, RDI: 0xffffff909d3b6600
R8:  0x0000000000000004, R9:  0x0000000000000002, R10: 0x0000000018103000, R11: 0x0000000000000000
R12: 0xffffff909d3b6000, R13: 0x0000000000000800, R14: 0xffffff8bd0ffba00, R15: 0x0000000000000000
RFL: 0x0000000000010206, RIP: 0xffffff8009301082, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff8bd0ffc000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 10

Panicked task 0xffffff909d0b6208: 119 threads: pid 0: kernel_task
Backtrace (CPU 0), panicked thread: 0xffffff9568f090c8, Frame : Return Address
0xffffffaba5c8f1e0 : 0xffffff80095eb31d mach_kernel : _handle_debugger_trap + 0x4ad
0xffffffaba5c8f230 : 0xffffff8009759236 mach_kernel : _kdp_i386_trap + 0x116
0xffffffaba5c8f270 : 0xffffff8009748480 mach_kernel : _kernel_trap + 0x3e0
0xffffffaba5c8f2c0 : 0xffffff8009585951 mach_kernel : _return_from_trap + 0xc1
0xffffffaba5c8f2e0 : 0xffffff80095eb5fd mach_kernel : _DebuggerTrapWithState + 0x5d
0xffffffaba5c8f3d0 : 0xffffff80095eaca9 mach_kernel : _panic_trap_to_debugger + 0x1a9
0xffffffaba5c8f430 : 0xffffff8009de08cb mach_kernel : _panic + 0x84
0xffffffaba5c8f520 : 0xffffff8009748953 mach_kernel : _sync_iss_to_iks + 0x2c3
0xffffffaba5c8f6a0 : 0xffffff8009748642 mach_kernel : _kernel_trap + 0x5a2
0xffffffaba5c8f6f0 : 0xffffff8009585951 mach_kernel : _return_from_trap + 0xc1
0xffffffaba5c8f710 : 0xffffff8009301082 
0xffffffaba5c8f820 : 0xffffff800b6af441 com.apple.driver.AirPort.BrcmNIC : _otp_read_pci + 0xbd
0xffffffaba5c8f870 : 0xffffff800b6ae420 com.apple.driver.AirPort.BrcmNIC : _initvars_srom_pci + 0x19c
0xffffffaba5c8f960 : 0xffffff800b68f2ea com.apple.driver.AirPort.BrcmNIC : _si_doattach + 0x9c0
0xffffffaba5c8fa00 : 0xffffff800b68e8a4 com.apple.driver.AirPort.BrcmNIC : _si_attach + 0x5c
0xffffffaba5c8fa60 : 0xffffff800b4b9eb3 com.apple.driver.AirPort.BrcmNIC : _wlc_attach + 0xaf
0xffffffaba5c8fb10 : 0xffffff800b3ce622 com.apple.driver.AirPort.BrcmNIC : __ZN15AirPort_BrcmNIC8wlcStartEv + 0x19a
0xffffffaba5c8fbe0 : 0xffffff800b3ccba6 com.apple.driver.AirPort.BrcmNIC : __ZN15AirPort_BrcmNIC10startGatedEP9IOService + 0x444
0xffffffaba5c8fc50 : 0xffffff800b3cc74f com.apple.driver.AirPort.BrcmNIC : __ZN15AirPort_BrcmNIC10_doCommandEP8OSObjectPvS2_S2_S2_ + 0x4d
0xffffffaba5c8fc80 : 0xffffff8009d15d54 mach_kernel : __ZN13IOCommandGate9runActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb4
0xffffffaba5c8fce0 : 0xffffff800b3cd88e com.apple.driver.AirPort.BrcmNIC : __ZN15AirPort_BrcmNIC5startEP9IOService + 0x3dc
0xffffffaba5c8fd50 : 0xffffff800d7e88c9 as.lvs1974.AirportBrcmFixup : __ZN6BRCMFX5startEP9IOServiceS1_ + 0x329
0xffffffaba5c8fdd0 : 0xffffff8009ce8a62 mach_kernel : __ZN9IOService14startCandidateEPS_ + 0xd2
0xffffffaba5c8fe30 : 0xffffff8009ce85da mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0xe0a
0xffffffaba5c8fef0 : 0xffffff8009ce7640 mach_kernel : __ZN9IOService14doServiceMatchEj + 0x3b0
0xffffffaba5c8ff50 : 0xffffff8009cea747 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x157
0xffffffaba5c8ffa0 : 0xffffff800958519e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         as.lvs1974.AirportBrcmFixup(2.1.6)[E6C2F921-ADCD-33DE-892C-51BF01BB7A15]@0xffffff800d7e5000->0xffffff800d7f8fff
            dependency: as.vit9696.Lilu(1.6.3)[97D2D6D6-A48D-3B6E-BD78-1BC201A6E107]@0xffffff800d412000->0xffffff800d499fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[752CD69F-309E-31FF-881C-2A22ACB6C0B6]@0xffffff800c000000->0xffffff800c02ffff
         com.apple.driver.AirPort.BrcmNIC(1400.1.1)[D35E9C22-17DD-30E6-BF29-1F3C10BCB958]@0xffffff800b3a2000->0xffffff800b8fafff
            dependency: com.apple.driver.corecapture(1.0.4)[3A3001CC-D289-35D6-B101-7B16A7F8AC9E]@0xffffff800c7f0000->0xffffff800c812fff
            dependency: com.apple.driver.mDNSOffloadUserClient(1.0.1b8)[A560C3FC-5D7E-3B9B-8B2D-425FB3ED908E]@0xffffff800bff9000->0xffffff800bffdfff
            dependency: com.apple.iokit.IO80211FamilyLegacy(1200.12.2b1)[63565270-42B3-31A8-80FB-16DA0EBBFA4C]@0xffffff800b240000->0xffffff800b386fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[196DFD8C-1D79-39E6-AEFA-FCF645C51977]@0xffffff800bd9f000->0xffffff800bdb5fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[752CD69F-309E-31FF-881C-2A22ACB6C0B6]@0xffffff800c000000->0xffffff800c02ffff
            dependency: com.apple.iokit.IOSkywalkFamily(1.0)[F96DD1E4-1285-31A2-8AE8-4C6884D59E11]@0xffffff800c0e9000->0xffffff800c116fff

Process name corresponding to current thread (0xffffff9568f090c8): kernel_task
Boot args: -lilubetaall -liluforce vsmcgen=1 darkwake=1 keepsyms=1 rtsx_timeout_shift=2 agdpmod=vit9696 brcmfx-country=#a brcmfx-driver=2 bpr_probedelay=100 bpr_initialdelay=300 bpr_postresetdelay=300 -disablegfxfirmware -igfxmpc alcverbs=1

Mac OS version:
Not yet set

Kernel version:
[... cut ...]
  . GEN1  : 0xffffff8bd1834000 - 0xffffff909d1cd000
  . GEN2  : 0xffffff909d1cd000 - 0xffffff9568b66000
  . GEN3  : 0xffffff9568b66000 - 0xffffff9a34500000
  . DATA  : 0xffffff9a34500000 - 0xffffffa0991ce000
  Metadata: 0xffffffa0b91de000 - 0xffffffa0d91de000
  Bitmaps : 0xffffffa0d91de000 - 0xffffffa0f11de000

Note that like I wrote before, the issue does not occur on a Dell Inspiron 5482 with touch screen enabled.
If I can do anything to help troubleshoot, please share specific instructions, happy to help.
Thanks for your efforts!

kprinssu, 2023-01-09:
@Erictoby @zerostar-git The newest build of VoodooI2C incoporates the upstream VoodooInput support, however, this will most likely not the panics you have.

@VGerris That panic log does not have anything to related to VoodooI2c nor VoodooInput. Please get support for that panic elsewhere.

zerostar-git, 2023-01-10:
@kprinssu Should do I back to Monterey to use touchscreen? I mean, do you guy have plan to release new update to fix this soon? I will be happy to test next beta version to fix this.

kprinssu, 2023-01-10:
@zerostar-git Unforunatley, without the hardware on hand, I am not able to fix it. I also don't have much more free time as I did during the holidays so this is probably going not be addressed for a while.

As always, PRs are welcome.

VGerris, 2023-01-12:
@kprinssu thanks for trying and the feedback. I noticed something interesting. I installed Ventura on a Precision 5550 with 4k touch screen which worked without issues with crashes or shutdown. But after I used migration assistant to move data from the 5520, the crash message at boot occurred on the 5550, but no reboot on shutdown issue, nor an effect after disabling touch: like on the Inspiron 5482, this all works fine. So I am beginning to suspect at least part of the issue is in one of the applications, which I will try to troubleshoot. On the 5520 the issue does not occur with the touchscreen disabled, perhaps it's some usb driver conflict.

Regarding you not having the hardware, if you can tell me what to do to troubleshoot in detail I am happy to. I have no experience in C and drivers but have a fullstack developer background so should be able to understand some stuff. I would need a really elaborate kickstart from someone, or just instructions in detail on how to look at stuff.
I hope this helps anyone else, thanks again for your work and efforts!

andreszerocross, 2023-01-12:
> @kprinssu thanks for trying and the feedback. I noticed something interesting. I installed Ventura on a Precision 5550 with 4k touch screen which worked without issues with crashes or shutdown. But [...]

I assume your touchscreen is connected to I2C Controller not USB (XHC Controller).

AFAIK, crash on shutdown or reboot is experienced by USB / XHC Touchscreen

smwasi98, 2023-01-21:
I'm encountering this same issue, this might be considered spam in which case perfectly okay to mark it as such, here's my hack build, I'll be taking a crack at this since I got the hardware on hand, any good ways to understand or get started on debugging this? @kprinssu

__**Fast's Dell Inspiron 5379:**__

Laptop: Dell Inspiron 5379 2-in-1
CPU: Intel Core i7-8550U
iGPU: Intel UHD Graphics 620
Wifi + BT: DW1560 (BCM9435Z)

kprinssu, 2023-02-12:
I have backported some changes from the BigSurface repo into VoodooI2C. Can you please test these sets of kexts? 

[Archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/10716885/Archive.zip)

smwasi98, 2023-02-13:
This is working quite well initially and does appear to fix the issue of kernel panics on reboot, I'll get back with any nuances or changes if I see any but so far so good!

[10 more replies]

closed 2023-02-18 by kprinssu