VoodooI2C/VoodooI2C #509: Touchpad loses response after about 5-15 minutes from cold startup.

TomHeaven, 2022-06-24:
**Describe the bug**

Touchpad loses response after about 5-15 minutes from cold startup. 
The no response issue cannot be fixed from hot startup most time. 

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors

Yes.

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

Yes.

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.

Yes. But I didn't find similar bug reports.

**System Environment**
Please answer:
1. What is your operating system?  macOS 12.3.1
2. What version of the drivers (including drivers) are you using?  VoodooI2C 2.6.5 & 2.7, both tried.
3. What is your hardware configuration? Huawei Matebook X Pro 2020 laptop with touchpad TPD0（HQTL2393）and touch screen TPL1（ SYNA2393）.

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

Here is also a IOReg screenshot:
[image]

Please download the bug report files here:
https://drive.google.com/drive/folders/1AUETLfbulWeMDTUbrg-uO5vEuPaJjQCI?usp=sharing
which includes three files:
1. Original DSDT file. 
2. Used SSDT (No DSDT used.)3. 
3. kernel log file.

**Additional context**
Add any other context about the problem here.

The issue seems to be related to the following error messages (I'm not sure):
```
2022-06-24 10:00:32.626432+0800 0x20f      Error       0x0                  0      0    kernel: (IOHIDFamily) IOHIDInterface:0x100000432 Large amount of outstanding buffers: 1297 1296
2022-06-24 10:00:32.813340+0800 0x20f      Error       0x0                  0      0    kernel: (IOHIDFamily) IOHIDInterface:0x100000432 Large amount of outstanding buffers: 1299 1298
```
The no response occurs about 5 minutes since I observe the error messages.

The touch screen works perfectly. The issue only exists for the touchpad. I try disabling the touch screen using SSDT, but it does not help solve the issue.

kprinssu, 2022-07-25:
Please attach a proper troubleshooting archive with a valid IO Registry dump and your EFI folder. Your logs do not mentioned VoodooI1C whatsoever and are not helpful.

kprinssu, 2022-07-27:
Please attach a proper troubleshooting archive and re-open the issue after doing so.

closed 2022-07-27 by kprinssu

TomHeaven, 2022-11-29:
Merely forcing polling mode may still meet the no response issue from the trackpad. In summary, there are three steps to fix the issue:
1. Force polling mode by adding boot argument ` -vi2c-force-polling`  using VodooI2C 2.7.0.
2. Boot to Windows and use device manager to uninstall the trackpad, scan PnP devices and update the driver locally. Of course, the step requires you have Windows installed.
3. Shutdown the laptop, unplug power cable, and wait for above 30 seconds.
Now, start the laptop and the trackpad should work fine.