VoodooI2C/VoodooI2C #510: Is this support macOS Ventura 13?

chris1111, 2022-06-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.

**System Environment**
Please answer:
1. What is your operating system? 
2. What version of the drivers (including drivers) are you using?
3. What is your hardware configuration?

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

**Additional context**

Working perfectly on Monterey 12.4 but not booting Ventura 13 beta 2 OC 0.8.2

kprinssu, 2022-07-01:
Ventura is still in beta and currently we do not support beta builds. There are few issues with panics with USB devices devices.

closed 2022-07-01 by kprinssu

chris1111, 2022-07-01:
I fix the problem thanks anyway