VoodooI2C/VoodooI2C #533: Latest updates breaks trackpad

SavageAUS, 2023-06-30:
**Describe the bug**
Up until a recent update my touchpad has been working 100%
Now it is not working at all unless I rollback VoodooI2CHID.kext & VoodooI2C.kext

**Did you read the common errors documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Common%20Errors/Common%20Errors
Yes

**Did you read the troubleshooting documentation?**
Please answer yes or no after reading https://voodooi2c.github.io/#Troubleshooting/Troubleshooting.
Yes

**Have you searched the issue on Github, Gitter, or Google?**
Please answer yes or no, this is to avoid creating duplicate Github issues.
Yes

**System Environment**
Please answer:
1. What is your operating system? Ventura
2. What version of the drivers (including drivers) are you using? VoodooI2CHID.kext  1.0 VoodooI2C.kext 2.8
3. What is your hardware configuration? Dell Latitude 2410, i5-10210, UHD Graphics, 16Gb RAM.

**Troubleshooting Archive**
Please attach the troubleshooting archive (as per https://voodooi2c.github.io/#Troubleshooting/Troubleshooting) and any additional logs or videos demonstrating the problem. Please attach any extra files to Github and *not* anywhere else like Google Drive.

**Additional context**
Add any other context about the problem here.
[image]
[image]
[MacBook Pro (2).zip](https://github.com/VoodooI2C/VoodooI2C/files/11915577/MacBook.Pro.2.zip)

![Screenshot 2023-06-30 at 8 13 59 pm](https://github.com/VoodooI2C
[image]
/VoodooI2C/assets/17326832/dd767617-dd44-4a80-a782-d93a9809c175)

SavageAUS, 2023-06-30:
This is Jorge after rolling back kexts and track pad working.
[MacBook Pro (2) older kexts.zip](https://github.com/VoodooI2C/VoodooI2C/files/11915626/MacBook.Pro.2.older.kexts.zip)

kprinssu, 2023-06-30:
I am closing this as:

- No logs attached
- Ventura is not supported (as well as all beta versions of macOS)

closed 2023-06-30 by kprinssu