VoodooI2C/VoodooI2C #534: HP 15-ey0xxx, touchpad and touchscreen work well but no pen input (ELAN2513)

big-mak, 2023-07-15:
I have been able to get trackpad and touch working, but not pen input.

I have searched and this seems to be the same as issue #513. Couldn't test with controllermate since it hasn't been updated to work with catalina and up, though I did notice a similar behaviour as the OP of that issue in that if modify events is enabled in karabiner-elements the pen acts like a mouse and the cursor moves very quickly around the screen.

I have seen in the posts below that it is possible to get pen working, with the pen providing absolute position input similar to when you use the touchscreen. [post](https://www.tonymacx86.com/threads/hp-spectre-x360-15-2017-i2c-touchscreen-and-pen-working-in-osx.246052/), [repo](https://github.com/acai66/lenovo-miix-520-hackintosh)

Monterey 12.6.7, VoodooI2C release 2.9 (couldn't get anything to work with 2.8) 
HP 15-ey0xxx, Ryzen 7 5825U

OC folder, ioreg, DSDT, VoodooI2C build, modify events video.
[archive.zip](https://github.com/VoodooI2C/VoodooI2C/files/12061958/archive.zip)

kprinssu, 2023-07-15:
Pen support is not officially supported by macOS. VoodooI2C emulates a magic touchpad to get native trackpad support.

Even touch screens are hacked in. As such, there are no plans to support pens nor are they officially supported. Pens and touchscreen are supported on a best effort and per device by the community.

As things are heading with Apple Silicon, I do not think full support for pens and any other specialised hardware will ever be added to VoodooI2C. However, there are numerous efforts by other contributors and I suggest you follow those efforts.

closed 2023-07-15 by kprinssu

big-mak, 2023-07-15:
Would you be able to link me some of these efforts? I've been unable to find anything.
Thanks

kprinssu, 2023-07-15:
My apologies but I mistook the touchscreen work for the pen. Pen support for the time being will be an open invitation for the community to help support.